zephyrproject-rtos/zephyr #31704: tests/bluetooth/init/bluetooth.init.test_ctlr_tiny Fails to build on nrf52dk_nrf52832

galak, 2021-01-27:
`./scripts/twister -p nrf52dk_nrf52832 -s tests/bluetooth/init/bluetooth.init.test_ctlr_tiny`

Fails

```
/home/galak/git/zephyr/subsys/bluetooth/controller/ll_sw/ull_conn.c: In function 'event_conn_upd_prep':
/home/galak/git/zephyr/subsys/bluetooth/controller/ll_sw/ull_conn.c:2473:15: error: 'struct <anonymous>' has no member named 'ticks_to_offset'
 2473 |    conn->slave.ticks_to_offset = 0U;
      |               ^
gmake[2]: *** [zephyr/subsys/bluetooth/controller/CMakeFiles/subsys__bluetooth__controller.dir/build.make:251: zephyr/subsys/bluetooth/controller/CMakeFiles/subsys__bluetooth__controller.dir/ll_sw/ull_conn.c.obj] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:3395: zephyr/subsys/bluetooth/controller/CMakeFiles/subsys__bluetooth__controller.dir/all] Error 2
```

galak, 2021-01-27:
Introduced in commit:

commit d2415fcfd014ce10e97725b818356019aad8ac00
Author: Vinayak Kariappa Chettimada <vich@nordicsemi.no>
Date:   Sun Jan 24 13:40:49 2021 +0530

    Bluetooth: controller: Add peripheral/central conditional compilations

galak, 2021-01-27:
See this also on:

`rv32m1_vega_ri5cy` `samples/bluetooth/peripheral_hr/sample.bluetooth.peripheral_hr_rv32m1_vega_ri5cy`

cvinayak, 2021-01-27:
Should be fixed by this: https://github.com/zephyrproject-rtos/zephyr/pull/31477/commits/2b2b044309c8e8dfdafc1bfd3030ae410fb9b5a2 in https://github.com/zephyrproject-rtos/zephyr/pull/31477

PR #31477 (2021-01-27): Bluetooth: controller: Fix deferred Conn Update offset population
Fix deferred Connection Update offset population by
introduction of explicit states waiting for the offset
calculation to complete in the ULL_LOW context.

Fixes #29636.

The problem was, in an encrypted connection the enqueued PDU
to be transmitted is encrypt in the prepare callback by the
hardware and swapped to a different buffer for transmission;
the deferred offset population did not reflect in the
transmitted PDU as it was filled in the cleartext buffer
while encryption completed into the encrypted buffer.

~Draft PR, pend [... cut]
closed 2021-01-27 by nashif