zephyrproject-rtos/zephyr #31711: UART failure with CONFIG_UART_ASYNC_API

edesta-be, 2021-01-27:
Hello!

So i started a very simple project with a NUCLEO_F746ZG. It essentially is responsible for connecting to a serial device and parsing its information. I wanted to use the buffer system in order to wait until the buffer was filled prior to throwing an event. With that I saw the async api configuration and decided to use it

In my prj.conf i have CONFIG_SERIAL = y and CONFIG_UART_ASYNC_API = y.
My actual main simply:
1. Gets the device binding
2. Ensures it was found
3. disables RX
4. calls uart_callback_set(device, callback, NULL)
And this is where the failure occurs. This function returns a negative number, which only occurs if it is not supported. 
I took a look src code for uart_callback_set and it has a ifdef around a set of code that needs to run in order to do this. however, it fails to know that it is defined. 

Not sure what to do next as it jumps to returning negative due to this failure.

Please let me know if there is something missing.

erwango, 2021-01-27:
@shlomow would you have time to have a look ?

shlomow, 2021-01-27:
@edesta-be did you set the dts bindings of the dma controller to `okay`? Please make sure that the async api is actually enabled using `ninja menuconfig`. In addition, the async api works only with dma. Please let me know if this solves the issus

erwango, 2021-02-03:
@edesta-be Could you have a check to the answer from @shlomow and provide feedback please ?

edesta-be, 2021-02-03:
I apologise! I did not see this! I will respond asap when I get to my setup! 

my apologies and thank you for your time!

erwango, 2021-02-08:
@edesta-be any news ? We're in the late stage of V2.5.0, and as a consequence we are tracking bug count.
So some feedback would be welcome

edesta-be, 2021-02-08:
Not yet I apologise, my team is backlogged on getting the parts back to me after having to do some retesting. So I have been unable to test.

erwango, 2021-03-03:
@edesta-be can you have a try with https://github.com/zephyrproject-rtos/zephyr/pull/32607 ?

FRASTM, 2021-03-03:
It seems that a similar approach based on the #27911  could help to clean and invalidate the data Cache when using the DMA for the UART async.
This test fails on Both NUCLEO_F746ZG  and NUCLEO_F767ZI   in the same manner.

edesta-be, 2021-03-03:
Hi @erwango! 
Should I configure the dma as previously suggested or try again with my original setup of only enabling the a sync api via the configs?

erwango, 2021-03-03:
@edesta-be First, please follow indications provided here: https://github.com/zephyrproject-rtos/zephyr/issues/31711#issuecomment-768549727
Then also, you can add #32607 on top.

FRASTM, 2021-03-03:
I confirm that using a flush command on Tx and invalidate one on Rx, improves the  test a lot
```
void uart_stm32_flush_cache(const void *ptr, uint32_t size)
{
	if ((SCB->CCR & SCB_CCR_DC_Msk) != 0U && size > 0) {
		SCB_CleanDCache_by_Addr((uint32_t *)ptr, size);
	}
}

void uart_stm32_invalidate_cache(void *ptr, uint32_t size)
{
	if ((SCB->CCR & SCB_CCR_DC_Msk) != 0U && size > 0) {
		SCB_InvalidateDCache_by_Addr((uint32_t *)ptr, size);
	}
}
```

erwango, 2021-03-31:
@eden-desta Should we keep this ticket open ?

erwango, 2021-05-10:
No feedback. Closing

closed 2021-05-10 by erwango