zephyrproject-rtos/zephyr #31726: RISC-V MIV SoC clock rate is specified 100x too slow

andyross, 2021-01-28:
The hardware clock rate on the RISC-V MIV platforms is off by a factor of 100.  This simple tool simply writes two bytes (".\n") every 100ms:

```c
#include <zephyr.h>
#include <sys/printk.h>

/* 100ms in hardware cycles */
#define CYC_100MS (CONFIG_SYS_CLOCK_HW_CYCLES_PER_SEC / 10)h

void main(void)
{
	/* Spin forever printing a "." line at 10 Hz */
	while(1) {
		uint32_t t0 = k_cycle_get_32();

		while((k_cycle_get_32() - t0) < CYC_100MS) {
		}

		printk(".\n");
	}
}
```

When run on renode using current Zephyr, the output looks like this:

```
6:40:23.0691 [INFO] uart: [+0.32s host +0s virt 0s virt from start] *** Booting Zephyr OS build v2.5.0-rc1-104-g4ab6dd003ed0  ***
16:40:23.1144 [INFO] uart: [+45.31ms host +1ms virt 1ms virt from start]   .
16:40:23.1609 [INFO] uart: [+46.45ms host +1ms virt 2ms virt from start]   .
16:40:23.2007 [INFO] uart: [+39.86ms host +1ms virt 3ms virt from start]   .
16:40:23.2435 [INFO] uart: [+42.81ms host +1ms virt 4ms virt from start]   .
...
```

Note that the virtual/simulated time is 1ms between messages, not the expected 100ms.

The root cause is the CONFIG_SYS_CLOCK_HW_CYCLES_PER_SEC value, which is specified as 660000 (660 kHz) in the SoC config when clearly the emulator is configured to be a 66 MHz board.

This patch fixes the problem and results in the correct output below:

```patch
--- a/soc/riscv/riscv-privilege/miv/Kconfig.defconfig.series
+++ b/soc/riscv/riscv-privilege/miv/Kconfig.defconfig.series
@@ -6,7 +6,7 @@ config SOC_SERIES
        default "miv"
 
 config SYS_CLOCK_HW_CYCLES_PER_SEC
-       default 660000
+       default 66000000
 
 config RISCV_SOC_INTERRUPT_INIT
        default y
```

```
16:45:29.5103 [INFO] uart: [+0.32s host +0s virt 0s virt from start] *** Booting Zephyr OS build v2.5.0-rc1-104-g4ab6dd003ed0  ***
16:45:33.4725 [INFO] uart: [+3.96s host +0.1s virt 0.1s virt from start]   .
16:45:37.2405 [INFO] uart: [+3.77s host +0.1s virt 0.2s virt from start]   .
16:45:40.9539 [INFO] uart: [+3.71s host +0.1s virt 0.3s virt from start]   .
16:45:45.1195 [INFO] uart: [+4.17s host +0.1s virt 0.4s virt from start]   .
...
```

This turns out to be the root cause to the failures we see on this platform in CI.  The schedule_api and lifo_usage tests simply have too much work to complete for a 660 kHz CPU.

Unfortunately, making that change in isolation results in **more** test failures, as the emulating the 66 MHz rate results in tests taking multiple minutes to complete (e.g. my development machine is about 40x slower than real time when spinning).

Someone needs to find a middle ground.  I tried a 6.6 MHz rate as an experiment and got passing results from the two tests I was looking at.  That seems like it should be acceptable, but it should be matched with a change to the emulator configuration that produces a correct clock rate.

PiotrZierhoffer, 2021-01-28:
@nashif @andyross  we're looking into this, will try to address it soon

galak, 2021-01-28:
I was curious if there's any implication of the renode version we have in CI.  We currently have 1.8.2.

PiotrZierhoffer, 2021-01-28:
@galak we should be able to adjust things without modifying the version (I'm about 90% sure).
Just switching to the most recent one (1.11) would not affect this situation.

andyross, 2021-01-28:
FWIW: playing with this everything passes with a 6.6MHz setting (even though the VM output timestamps are technically wrong) **except** the tickless_concept test, which oddly works at 660 kHz but not at any frequencies higher than 703999 Hz (yes, exactly!  704000 fails).

This actually looks like it's exercising an edge case in @ceolin's timeslicing fix, as reverting that recovers this one test (but obviously introduces the old bug elsewhere).  Somehow it's ending up in a situation where the new code is suppressing a timeout from being set when no timer interrupt is actually scheduled.  I'm working this part separately.

But beyond that, an ideal fix here would change the simulator clock rate to somewhere around 4-8 MHz and adjust the kconfig to match.

ajelinski, 2021-01-30:
@andyross Changing the frequency in the kconfig fixes the time because in Renode the frequency in the prepared MI-V platform is 66 MHz indeed. I'm not sure however whether you're right about these testing failures after using 66 Mhz in Kconfig.

Can you please write down what real vs virtual time ratio you have for the 6.6 MHz setting (for 66 MHz you wrote "40x slower")?

From what I've tested it seems that it's quite similar to the one for 66 MHz. At least on my machine both of these settings needed about 3.1s to write each subsequent dot by the binary built using your code. The current frequency of 660 kHz had the ratio about 33x on the other hand so it's not a linear relationship. If the ratio for 6.6 MHz and 66 MHz almost equals, then it shouldn't cause any big rise of failures in testing.

andyross, 2021-02-01:
There is a workaround now in #31841 that works to make all tests on this platform pass.  It uses 4 MHz as the clock rate, but as mentioned changes only the Zephyr kconfig because I don't know how to change the clock rate in renode.

@ajelinski You can set the value to 66000000 yourself and run the tests using twister.  Lots and lots fail with timeouts, they simply take too long.  The reason is that lots of our tests do busy waiting or other spinning tasks that take up some fixed (not dependent on clock rate) amount of simulated time.  And at the full clock rate that's simply too slow.

galak, 2021-02-02:
> @ajelinski You can set the value to 66000000 yourself and run the tests using twister. Lots and lots fail with timeouts, they simply take too long. The reason is that lots of our tests do busy [...]

Is there not some mode or option to have renode deal with simulated time differently?  Feels like most simulators have something along those lines, so you can have the software set the time to match real hardware, but have simulated time be something sane.

ajelinski, 2021-02-04:
Sorry for the delay. We've been discussing your problems internally in the Renode team. From what we understand it seems that the test timeouts set require having faster simulation, especially when there's any busy waiting involved.

There is such a speed up currently because of using different frequency – 660 kHz vs 66 MHz set in the Renode's MI-V platform description (`renode/platforms/cpus/miv.repl` loaded by `boards/riscv/m2gl025_miv/support/m2gl025_miv.resc`).

This is the root cause of the problem from this GH issue description (1ms wait in virtual time vs expected 100ms).

It can be solved simply by using matching frequency in both places. We think it makes sense to do so. If it's really necessary to use some other frequency than 66 MHz (like the proposed 4 MHz) then we can put a custom platform description alongside the `m2gl025_miv.resc` file.

To boost the waiting speed it's best to manipulate the `PerformanceInMips` value from the RESC file: https://github.com/zephyrproject-rtos/zephyr/blob/76ebe71fcbf40fd876edcc0cd3eb9d3d6855a754/boards/riscv/m2gl025_miv/support/m2gl025_miv.resc#L11 . Setting it, e.g., to 4 or 8 can give quite a significant boost. This seems to be the proper solution. I'll create a respective WIP PR to find settings that'll solve this issue without causing Twister failures.

ajelinski, 2021-02-05:
That's curious, all tests passed with 66 MHz in MI-V's `Kconfig.defconfig.series` without any more changes: https://github.com/zephyrproject-rtos/zephyr/pull/32036

Are all tests for `m2gl025_miv` run by buildkite in CI?

nashif, 2021-02-05:
> Are all tests for `m2gl025_miv` run by buildkite in CI @andyross?

based on your change, no. Add this change to force testing on the platform:

```diff
diff --git a/boards/riscv/m2gl025_miv/m2gl025_miv.yaml b/boards/riscv/m2gl025_miv/m2gl025_miv.yaml
index ff41948899..00e67f6450 100644
--- a/boards/riscv/m2gl025_miv/m2gl025_miv.yaml
+++ b/boards/riscv/m2gl025_miv/m2gl025_miv.yaml
@@ -7,6 +7,7 @@ toolchain:
 ram: 64
 simulation: renode
 testing:
+  default: true
   ignore_tags:
     - net
     - bluetooth

```

ajelinski, 2021-02-05:
Thanks @nashif, I'll try that.

PR #32036 (2021-02-10): Fix virtual clock in RISC-V
The first commit fixes #31726 and the second one's goal is to compensate timeouts caused by extended busy waiting.
commit 86ca5ff5 (2021-02-10): m2gl025_miv: Adjust frequencies and performance

The default frequency for this board in Renode is 66 MHz. This needs to
be even with `SYS_CLOCK_HW_CYCLES_PER_SEC` to avoid such problems as in
the #31726 issue.

Unfortunately, the difference in these values was helpful for some tests
that are failing with 66 MHz set in both places. It created artificial
boost in certain circumstances.

The frequencies in the default Renode platform description (`.repl`)
file for MI-V were overridden with 4 MHz value that seems to be better
tolerated based on testing. The `SYS_CLOCK_HW_CYCLES_PER_SEC` was
adjus [... cut]
closed 2021-02-10 by nashif