zephyrproject-rtos/zephyr #31729: test: build fatal related testcase failed on qemu_cortex_m0 and run failed on qemu_nios2

yingmingx, 2021-01-28:
**Describe the bug**
Run twister failed on some fatal related testcase.

**To Reproduce**
Steps to reproduce the behavior:
1. twister -T tests/kernel/fatal/exception/  or twister -T tests/ztest/error_hook/
2. See error

**Expected behavior**
The test passes

**Impact**
Not clear

**Logs and console output**
Renaming output directory to /home/yingming/zephyrproject/zephyr/twister-out.14
INFO    - Zephyr version: zephyr-v2.4.0-3716-g2712d74cdcbf
INFO    - JOBS: 8
INFO    - Selecting default platforms per test case
INFO    - Building initial testcase list...
INFO    - 2 test suites (48 configurations) selected, 10 configurations discarded due to filters.
INFO    - Adding tasks to the queue...
INFO    - Added initial list of jobs to queue
ERROR   - Cmake build failure: /home/yingming/zephyrproject/zephyr/tests/ztest/error_hook for qemu_cortex_m0

ERROR   - qemu_cortex_m0            tests/ztest/error_hook/testing.ztest.error_hook.no_userspace FAILED: Cmake build failure
ERROR   - see: /home/yingming/zephyrproject/zephyr/twister-out/qemu_cortex_m0/tests/ztest/error_hook/testing.ztest.error_hook.no_userspace/build.log
INFO    - Total complete:   10/  38  26%  skipped:   17, failed:    1
ERROR   - qemu_nios2                tests/ztest/error_hook/testing.ztest.error_hook.no_userspace FAILED: Exited with 2
ERROR   - see: /home/yingming/zephyrproject/zephyr/twister-out/qemu_nios2/tests/ztest/error_hook/testing.ztest.error_hook.no_userspace/handler.log
INFO    - Total complete:   27/  38  71%  skipped:   21, failed:    2ERROR   - Cmake build failure: /home/yingming/zephyrproject/zephyr/tests/ztest/error_hook for qemu_cortex_m0

ERROR   - qemu_cortex_m0            tests/ztest/error_hook/testing.ztest.error_hook    FAILED: Cmake build failure
ERROR   - see: /home/yingming/zephyrproject/zephyr/twister-out/qemu_cortex_m0/tests/ztest/error_hook/testing.ztest.error_hook/build.log
INFO    - Total complete:   38/  38  100%  skipped:   26, failed:    3
INFO    - 19 of 22 test configurations passed (86.36%), 3 failed, 26 skipped with 0 warnings in 37.74 seconds
INFO    - In total 88 test cases were executed, 104 skipped on 24 out of total 329 platforms (7.29%)
INFO    - 16 test configurations executed on platforms, 6 test configurations were only built.
INFO    - Saving reports...
INFO    - Writing xunit report /home/yingming/zephyrproject/zephyr/twister-out/twister.xml...
INFO    - Writing xunit report /home/yingming/zephyrproject/zephyr/twister-out/twister_report.xml...
INFO    - Run completed

Including boilerplate (Zephyr base): /home/yingming/zephyrproject/zephyr/cmake/app/boilerplate.cmake
-- Application: /home/yingming/zephyrproject/zephyr/tests/ztest/error_hook
-- Zephyr version: 2.5.0-rc1 (/home/yingming/zephyrproject/zephyr)
-- Found Python3: /usr/bin/python3.6 (found suitable exact version "3.6.9") found components: Interpreter
-- Found west (found suitable version "0.7.2", minimum required is "0.7.1")
-- Board: qemu_cortex_m0
-- Cache files will be written to: /home/yingming/.cache/zephyr
-- Found toolchain: zephyr (/home/yingming/zephyr-sdk-0.11.3)
-- Found dtc: /home/yingming/zephyr-sdk-0.11.3/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found BOARD.dts: /home/yingming/zephyrproject/zephyr/boards/arm/qemu_cortex_m0/qemu_cortex_m0.dts
-- Generated zephyr.dts: /home/yingming/zephyrproject/zephyr/twister-out/qemu_cortex_m0/tests/ztest/error_hook/testing.ztest.error_hook.no_userspace/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /home/yingming/zephyrproject/zephyr/twister-out/qemu_cortex_m0/tests/ztest/error_hook/testing.ztest.error_hook.no_userspace/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: /home/yingming/zephyrproject/zephyr/twister-out/qemu_cortex_m0/tests/ztest/error_hook/testing.ztest.error_hook.no_userspace/zephyr/include/generated/device_extern.h

**Environment (please complete the following information):**
 - OS: ubuntu1~18.04
 - Toolchain zephyr-sdk-0.11.3
 - Commit id zephyr-v2.4.0-3716-g2712d74cdcbf

**Additional context**

warning: HAS_NRFX (defined at modules/hal_nordic/nrfx/Kconfig:4) has direct dependencies 0 with value n, but is currently being y-selected by the following symbols:
 - SOC_SERIES_NRF51X (defined at soc/arm/nordic_nrf/nrf51/Kconfig.series:7), with value y, direct dependencies <choice> (value: y), and select condition <choice> (value: y)

warning: NRFX_CLOCK (defined at modules/hal_nordic/nrfx/Kconfig:14) has direct dependencies HAS_HW_NRF_CLOCK && HAS_NRFX && 0 with value n, but is currently being y-selected by the following symbols:
 - CLOCK_CONTROL_NRF (defined at drivers/clock_control/Kconfig.nrf:13), with value y, direct dependencies SOC_COMPATIBLE_NRF && !CLOCK_CONTROL_NRF_FORCE_ALT && CLOCK_CONTROL (value: y), and select condition SOC_COMPATIBLE_NRF && !CLOCK_CONTROL_NRF_FORCE_ALT && CLOCK_CONTROL (value: y)
Parsing /home/yingming/zephyrproject/zephyr/Kconfig
Loaded configuration '/home/yingming/zephyrproject/zephyr/boards/arm/qemu_cortex_m0/qemu_cortex_m0_defconfig'
Merged configuration 'prj_no_userspace.conf'

enjiamai, 2021-01-30:
@yingmingx  I can't reproduce this in zephyr-sdk version 0.11.3 / 0.11.4 / 0.12.1 , I think it has something to do with your toolchain version, could you please:
1. Do "west update" before you run the test cases again.
2. Use the newer version of SDK to check if this still happens.

Thanks!

yingmingx, 2021-02-01:
I do what you said according the steps you have given. All these testcases have passed. 
It is indeed my environment problem.
Thanks!

closed 2021-02-01 by yingmingx