zephyrproject-rtos/zephyr #31742: Bluetooth: BLE 5 data throughput to Linux host

Oxymoron79, 2021-01-28:
**Describe the bug**

I'm trying to maximize the data throughput of a Bluetooth LE 5 Peripheral to a Bluetooth Dongle connected to a
Linux host running BlueZ using two [Nordic nRF52840 Dongles](https://docs.zephyrproject.org/latest/boards/arm/nrf52840dongle_nrf52840/doc/index.html).

The firmware of the Bluetooth LE 5 Peripheral and the Dongle are built with zephyr.

The data is transferred using GATT notifications using the Bluetooth LE 5 Data Length Extension and the 2MBit PHY.

I found 2 issues when I increase the GATT notifications rate to about 250 notifications per second (or higher):
1. The data transmission gets stuck after a few notifications have been transferred. This may be related to the issue
#25917
2. After the data transmission has finished, bluez is unable to disconnect the peripheral. In this case, the bluez service
has to be restarted and the device has to be removed from the bluez cache.

I am new to using the Zephyr RTOS, so these issues may be caused by some wrong kernel configuration parameters. I tried
different settings but was not successful to resolve the issues.

**To Reproduce**

To reproduce the behavior I have uploaded the zephyr projects for the peripheral and the central devices to my
[GitHub repository](https://github.com/Oxymoron79/zephyr-ble-peripheral) with a python script to test the data throughput.

The README describes the build steps and the usage of the test script.

Run the `throughput_test.py` with these parameters for a notification rate of about 250 notifications per second and
a notification payload size of 240 bytes:
`./throughput_test.py -i 4 -l 240 -n 1000`

**Expected behavior**

The test script should finish without exceptions.

**Logs and console output**

Peripheral console logs some occurrences of `bt_gatt_notify` returning a error
and the kernel logs a `Unable to allocate TX context` error when the issue occurs:

```
data_work_handler: bt_gatt_notify returned: -57
[14:07:44.410,400] <err> bt_conn: Unable to allocate TX context
```

I also captured the HCI traffic with `btmon` for the 2 issues and attached the files:
1. `throughput_stuck_i4_l240_n250.btsnoop` when the notification transmission was stuck.
2. `throughput_disconnect_i4_l240_n200.btsnoop` when the disconnect failed.

**Environment (please complete the following information):**
 - OS: Arch Linux, Bluez 5.55, Linux Kernel 5.10.10-arch1-1  SMP PREEMPT
 - Toolchain: Zephyr SDK 0.11.4
 - v2.5.0-rc1

Oxymoron79, 2021-01-28:
file:///home/martin/Projects/Zephyr/zephyrproject/throughput_stuck_i4_l240_n250.btsnoop

Oxymoron79, 2021-01-28:
[throughput_stuck_i4_l240_n250.log](https://github.com/zephyrproject-rtos/zephyr/files/5886572/throughput_stuck_i4_l240_n250.log)
[throughput_disconnect_i4_l240_n200.log](https://github.com/zephyrproject-rtos/zephyr/files/5886575/throughput_disconnect_i4_l240_n200.log)

carlescufi, 2021-01-28:
> The firmware of the Bluetooth LE 5 Peripheral and the Dongle are built with zephyr.

So one dongle runs a standalone Zephyr application and the other runs `hci_usb`?
In fact, can you please give us steps to reproduce the issue?

Oxymoron79, 2021-01-28:
> So one dongle runs a standalone Zephyr application and the other runs hci_usb?

Yes, exactly.

carlescufi, 2021-01-28:
Could  you please try setting:
`CONFIG_BT_RX_BUF_LEN=258`
on both apps? Note that this is 258, not 255.

carlescufi, 2021-01-28:
Also, in `hci_usb`'s `prj.conf`, could you uncomment:
`#CONFIG_BT_CTLR_TX_BUFFER_SIZE=251`

Oxymoron79, 2021-01-28:
@carlescufi I applied your suggestions but unfortunately it didn't help. The test got stuck after 71 notifications:
```
./throughput_test.py -i 4 -l 240 -n 1000
Using Adapter (/org/bluez/hci0, org.bluez.Adapter1)
Discover device hosting the throughput service abcdef00-f5bf-58d5-9d17-172177d1316a for 10 seconds.
Found.
Connect to Device (/org/bluez/hci0/dev_E3_4D_2A_E8_66_06, org.bluez.Device1)
Done.
Check for throughput service abcdef00-f5bf-58d5-9d17-172177d1316a
Found.
Read config characteristic abcdef01-f5bf-58d5-9d17-172177d1316a parameters:
- interval: 100 ms
- data_len: 10 bytes
Set config characteristic abcdef01-f5bf-58d5-9d17-172177d1316a parameters:
- interval: 4 ms
- data_len: 240 bytes
Done.
Receive 1000 notifications from data characteristic abcdef02-f5bf-58d5-9d17-172177d1316a
Notification    1: 240 bytes, timestamp: 0.00132 s, dt: 0.00132 s
Notification    2: 240 bytes, timestamp: 0.00465 s, dt: 0.00333 s
...
Notification   70: 240 bytes, timestamp:   0.328 s, dt: 0.00338 s
Notification   71: 240 bytes, timestamp:   0.331 s, dt: 0.00342 s
<it's stuck here>
```
Here's the peripheral console log:
```
Connected                                                                                                 
Connection Information:                                                                                   
* Role: Slave.                                                                                            
* Connection Type: LE                                                                                     
  PHY: TX: 0x01, RX: 0x01.                                                                                
  Data Length: TX Max length: 27, TX Max time: 328us, RX Max length: 27, RX Max time: 328us.              
Update Data Length.                                                                                       
Update 2MBit PHY.                                                                                         
Update Connection Parameters: Interval min: 7ms, Interval max: 400ms, Latency: 0, timeout: 400ms.         
Data Length updated: TX Max length: 251, TX Max time: 1064us, RX Max length: 251, RX Max time: 1064us     
PHY updated: TX: 0x01, RX: 0x01                                                                           
Wrote config:                                                                                             
- interval_ms: 4                                                                                          
- data_length: 240                                                                                        
"Data" Characteristic Notifications got enabled                                                           
[00:00:22.615,661] <err> bt_conn: Unable to allocate TX context 
```

carlescufi, 2021-01-28:
>[00:00:22.615,661] <err> bt_conn: Unable to allocate TX context 

@joerchan the number of contexts are not configurable, so I think this is either a problem in the way the application uses the APIs or an actual bug.

carlescufi, 2021-02-04:
@Oxymoron79 a few fixes have been merged to master since you last reported here. Could you please try to reproduce the issue again and let us know?

Oxymoron79, 2021-02-05:
@carlescufi I tested it with the commit [b6a4db3](https://github.com/zephyrproject-rtos/zephyr/commit/b6a4db3766936d738ed84f9f6a4ee19f586ce062) on the master branch.

Unfortunately I could reproduce the same issues.

**BUT**: I started to measure the time delta between the notifications received and noticed that it sometimes easily exceeds 100 milliseconds. I wondered whether submitting the `bt_gatt_notify` call to a system work item in the timer handler is a good idea.

So I refactored the peripheral to enqueue a notification request into a FIFO in the timer and feed the `bt_gatt_notify` calls from that FIFO. I pushed this implementation to the `notification_fifo` [branch](https://github.com/Oxymoron79/zephyr-ble-peripheral/tree/notification_fifo) of my repository, in case you're interested.

With this implementation I couldn't reproduce the issues and was able to get the throughput up to about 640 kbit/second:
```
$ ./throughput_test.py -i 3 -l 243 -n 10000
...
Notification 9998: 243 bytes, timestamp:    30.2 s, dt: 0.00133 s
Notification 9999: 243 bytes, timestamp:    30.2 s, dt: 0.00139 s
Notification 10000: 243 bytes, timestamp:    30.2 s, dt: 0.00146 s
Summary: Received 2430000 bytes in 10000 notifications during 30.2 seconds: 643.536 kbits/sec.
min dt:   0.133 ms, max dt: 1.03e+02 ms.
```
Also the error `bt_conn: Unable to allocate TX context` never occurred with this implementation.

So I think that the issues I encountered were caused by my usage of the API - but the latest commits to master certainly also helped.

I think this issue can be closed.

Thank you for your support with this issue.

joerchan, 2021-02-05:
@Oxymoron79 Thank you for the update.

I'm gonna re-open this issue just so that we can agree on the facts and close it once we all agree that there is nothing that should be fixed here.

The reason for the `Unable to allocate TX context` is because bt_gatt_notify is called from the system work queue. We cannot block from this thread, since this is the thread that would release them, and blocking would cause a deadlock.

```
	/* The TX context always get freed in the system workqueue,
	 * so if we're in the same workqueue but there are no immediate
	 * contexts available, there's no chance we'll get one by waiting.
	 */
	if (k_current_get() == &k_sys_work_q.thread) {
		return k_fifo_get(&free_tx, K_NO_WAIT);
	}
```

@jhedberg @carlescufi Do we agree that this is the behavior, or is the number of TX contexts incorrect? 

@Oxymoron79 Can you elaborate on what you meant by "stuck", I think there is some confusion about whether you managed to get the Stack stuck, or if your application just didn't handle the returned error code from bt_gatt_notify?

jfischer-no, 2021-02-05:
@Oxymoron79 Can you retest with the changes in https://github.com/zephyrproject-rtos/zephyr/pull/32009, see also bug report https://github.com/zephyrproject-rtos/zephyr/issues/31922, please?

Oxymoron79, 2021-02-07:
@carlescufi With "stuck" I meant that the receiver of the notifications (the application accessing the BlueZ API) no longer receives notifications.

Here's how I will refer to the involved components:

```
 ------------------       -----------        -----------------------------------
| Peripheral       |     | hci_usb   |      | Kernelspace   | Userspace         |
| App -> BT stack -|-RF->| BT stack -|-USB->| btusb -|-HCI->| BlueZ -> Receiver |
 ------------------       -----------       |           ^---- btmon             |
                                             -----------------------------------
```

The Peripheral App sets up a [timer](https://docs.zephyrproject.org/latest/reference/kernel/timing/timers.html) whose expiry function adds a work item to the [system workqueue](https://docs.zephyrproject.org/latest/reference/kernel/threads/workqueue.html).
The enqueued work item calls `bt_gatt_notify` of the BT stack API, logging the error to the console if the function fails:

```
static void data_work_handler(struct k_work *work)
{
    int err = 0;
    err = bt_gatt_notify(NULL, &service.attrs[3], data, config.data_length);
    if (err != 0)
        printk("data_work_handler: bt_gatt_notify returned: %i\n", err);
}
```

On the PC side, the Receiver application is receiving the notifications from the BlueZ userspace daemon. To check whether
the BlueZ daemon is working properly, the HCI communcation with the Kernel Bluetooth subsystem is captured with `btmon`.

When the notification transfer is "stuck", this is what I observed:
* The Peripheral App doesn't log anything to the console. Some time before the transmission gets "stuck" the BT stack
  of the Peripheral logged the `bt_conn: Unable to allocate TX context` error:

```
...
Wrote config:
- interval_ms: 3
- data_length: 243
"Data" Characteristic Notifications got enabled
[00:05:07.898,895] <err> bt_conn: Unable to allocate TX context
```

* The Receiver is waiting for the notification from the BlueZ API. There is no timeout or error received from the
  BlueZ daemon:

```
...
Notification  240: 243 bytes, timestamp:   0.756 s, dt:  0.0014 s
Notification  241: 243 bytes, timestamp:   0.758 s, dt: 0.00142 s
Notification  242: 243 bytes, timestamp:   0.759 s, dt: 0.00134 s
Notification  243: 243 bytes, timestamp:   0.761 s, dt: 0.00143 s
<stuck here>
```

* The `btmon` is no longer capturing any communication:

```
...
> ACL Data RX: Handle 0 flags 0x02 dlen 250                                         #310 [hci0] 17.212098
      ATT: Handle Value Notification (0x1b) len 245
        Handle: 0x0014
          Data: 000102030405060708090a0b0c0d0e0f101112131415161718191a1b1c1d1e1f202122232425262728292a2b2c2d2e2f303132333435363738393a3b3c3d3e3f404142434445464748494a4b4c4d4e4f505152535455565758595a5b5c5d5e5f606162636465666768696a6b6c6d6e6f707172737475767778797a7b7c7d7e7f808182838485868788898a8b8c8d8e8f909192939495969798999a9b9c9d9e9fa0a1a2a3a4a5a6a7a8a9aaabacadaeafb0b1b2b3b4b5b6b7b8b9babbbcbdbebfc0c1c2c3c4c5c6c7c8c9cacbcccdcecfd0d1d2d3d4d5d6d7d8d9dadbdcdddedfe0e1e2e3e4e5e6e7e8e9eaebecedeeeff0f1f2
> ACL Data RX: Handle 0 flags 0x02 dlen 250                                         #311 [hci0] 17.213525
      ATT: Handle Value Notification (0x1b) len 245
        Handle: 0x0014
          Data: 000102030405060708090a0b0c0d0e0f101112131415161718191a1b1c1d1e1f202122232425262728292a2b2c2d2e2f303132333435363738393a3b3c3d3e3f404142434445464748494a4b4c4d4e4f505152535455565758595a5b5c5d5e5f606162636465666768696a6b6c6d6e6f707172737475767778797a7b7c7d7e7f808182838485868788898a8b8c8d8e8f909192939495969798999a9b9c9d9e9fa0a1a2a3a4a5a6a7a8a9aaabacadaeafb0b1b2b3b4b5b6b7b8b9babbbcbdbebfc0c1c2c3c4c5c6c7c8c9cacbcccdcecfd0d1d2d3d4d5d6d7d8d9dadbdcdddedfe0e1e2e3e4e5e6e7e8e9eaebecedeeeff0f1f2
<stuck here>
```

What is interessting is that when the system is stuck and I send a read request to another characteristic of the
Peripheral (here: the configuration characteristic with the UUID `abcdef01-f5bf-58d5-9d17-172177d1316a`),
[... 89 lines cut ...]
        Reason: Connection terminated by local host (0x02)
```

After the peripeheral has been disconnected
1. it can be discovered again with `bluetoothctl`
2. a connection can be established, but the peripheral doesn't call the
   [connected callback](https://github.com/Oxymoron79/zephyr-ble-peripheral/blob/master/src/main.c#L286).
3. a few seconds later, `bluetoothctl` reports that the services were resolved (without actually creating the DBus
   objects for the services and characteristics).
4. after that, the device is disconnected again at which time the peripheral logs the error: `bt_hci_core: Unable to look up conn with handle 1`

The system can only be recovered by restarting the Peripheral.

I can't deduce with certainty which component gets "stuck" in this setup:
* The Peripheral seems to be the component as it is the only one that needs to be reset to get the system working again.
  IMO, it can get stuck due to:
  * a bug in the BT stack.
  * the application overloading the system workqueue with work items from the timer. This seems to be the case as I am
    able to work around the issue by using a normal FIFO instead of the system workqueue.
* The Bluetooth dongle running the `hci_usb` sample application may be the cause.
* The `btusb` kernel module may be the cause as well.
* The BlueZ daemon is unlikely to be the cause as its communication with the kernel driver seems to be fine according to
  the captured messages from `btmon`.

Sorry for this long post, but I wanted to explain my findings in detail. I hope it helps.

Oxymoron79, 2021-02-07:
@jfischer-no I built the Peripheral and the Dongle using your [pr-fix-hci_usb](https://github.com/jfischer-no/zephyr/tree/pr-fix-hci_usb) branch.

Unfortunately this issue still occurs. Here are the logs:

Peripheral:

```
Connected                                                                                                 
Connection Information:                                                                                   
* Role: Slave.                                                                                            
* Connection Type: LE                                                                                     
  PHY: TX: 0x01, RX: 0x01.                                                                                
  Data Length: TX Max length: 27, TX Max time: 328us, RX Max length: 27, RX Max time: 328us.              
Update Data Length.                                                                                       
Update 2MBit PHY.                                                                                         
Update Connection Parameters: Interval min: 7ms, Interval max: 400ms, Latency: 0, timeout: 400ms.         
Data Length updated: TX Max length: 251, TX Max time: 1064us, RX Max length: 251, RX Max time: 1064us     
PHY updated: TX: 0x02, RX: 0x02                                                                           
Wrote config:                                                                                             
- interval_ms: 3                                                                                          
- data_length: 243                                                                                        
"Data" Characteristic Notifications got enabled                                                           
[00:00:31.398,315] <err> bt_conn: Unable to allocate TX context 
```

Receiver application:

```
$ ./throughput_test.py -i 3 -l 243 -n 1000
Using Adapter (/org/bluez/hci0, org.bluez.Adapter1)
Discover device hosting the throughput service abcdef00-f5bf-58d5-9d17-172177d1316a for 10 seconds.
Found.
Connect to Device (/org/bluez/hci0/dev_E3_4D_2A_E8_66_06, org.bluez.Device1)
Done.
Check for throughput service abcdef00-f5bf-58d5-9d17-172177d1316a
Found.
Read config characteristic abcdef01-f5bf-58d5-9d17-172177d1316a parameters:
- interval: 100 ms
- data_len: 10 bytes
Set config characteristic abcdef01-f5bf-58d5-9d17-172177d1316a parameters:
- interval: 3 ms
- data_len: 243 bytes
Done.
Receive 1000 notifications from data characteristic abcdef02-f5bf-58d5-9d17-172177d1316a
Notification    1: 243 bytes, timestamp: 0.000306 s, dt: 0.000306 s
Notification    2: 243 bytes, timestamp: 0.00166 s, dt: 0.00136 s
Notification    3: 243 bytes, timestamp: 0.00298 s, dt: 0.00132 s
....
Notification  310: 243 bytes, timestamp:   0.959 s, dt: 0.00134 s
Notification  311: 243 bytes, timestamp:   0.961 s, dt: 0.00146 s
Notification  312: 243 bytes, timestamp:   0.962 s, dt: 0.00129 s
<stuck here>
```

`btmon` capture:

```
Bluetooth monitor ver 5.55
= Note: Linux version 5.10.13-arch1-1 (x86_64)                                                   0.482684
= Note: Bluetooth subsystem version 2.22                                                         0.482689
= New Index: 00:00:00:00:00:00 (Primary,USB,hci0)                                         [hci0] 0.482692
= Open Index: 00:00:00:00:00:00                                                           [hci0] 0.482693
= Index Info: 00:00:00:00:00:00 (The Linux Foundation)                                    [hci0] 0.482695
@ MGMT Open: bluetoothd (privileged) version 1.18                                       {0x0001} 0.482698
@ MGMT Command: Start Discovery (0x0023) plen 1                                 {0x0001} [hci0] 15.919294
        Address type: 0x06
          LE Public
          LE Random
< HCI Command: LE Set Random Address (0x08|0x0005) plen 6                             #1 [hci0] 15.919395
        Address: 19:82:E1:FB:AD:71 (Non-Resolvable)
[... 409 lines cut ...]
          Data: 000102030405060708090a0b0c0d0e0f101112131415161718191a1b1c1d1e1f202122232425262728292a2b2c2d2e2f303132333435363738393a3b3c3d3e3f404142434445464748494a4b4c4d4e4f505152535455565758595a5b5c5d5e5f606162636465666768696a6b6c6d6e6f707172737475767778797a7b7c7d7e7f808182838485868788898a8b8c8d8e8f909192939495969798999a9b9c9d9e9fa0a1a2a3a4a5a6a7a8a9aaabacadaeafb0b1b2b3b4b5b6b7b8b9babbbcbdbebfc0c1c2c3c4c5c6c7c8c9cacbcccdcecfd0d1d2d3d4d5d6d7d8d9dadbdcdddedfe0e1e2e3e4e5e6e7e8e9eaebecedeeeff0f1f2
> ACL Data RX: Handle 0 flags 0x02 dlen 250                                          #70 [hci0] 17.456279
      ATT: Handle Value Notification (0x1b) len 245
        Handle: 0x0014
          Data: 000102030405060708090a0b0c0d0e0f101112131415161718191a1b1c1d1e1f202122232425262728292a2b2c2d2e2f303132333435363738393a3b3c3d3e3f404142434445464748494a4b4c4d4e4f505152535455565758595a5b5c5d5e5f606162636465666768696a6b6c6d6e6f707172737475767778797a7b7c7d7e7f808182838485868788898a8b8c8d8e8f909192939495969798999a9b9c9d9e9fa0a1a2a3a4a5a6a7a8a9aaabacadaeafb0b1b2b3b4b5b6b7b8b9babbbcbdbebfc0c1c2c3c4c5c6c7c8c9cacbcccdcecfd0d1d2d3d4d5d6d7d8d9dadbdcdddedfe0e1e2e3e4e5e6e7e8e9eaebecedeeeff0f1f2
> ACL Data RX: Handle 0 flags 0x02 dlen 250                                          #71 [hci0] 17.457622
      ATT: Handle Value Notification (0x1b) len 245
        Handle: 0x0014
[... cut ...]
          Data: 000102030405060708090a0b0c0d0e0f101112131415161718191a1b1c1d1e1f202122232425262728292a2b2c2d2e2f303132333435363738393a3b3c3d3e3f404142434445464748494a4b4c4d4e4f505152535455565758595a5b5c5d5e5f606162636465666768696a6b6c6d6e6f707172737475767778797a7b7c7d7e7f808182838485868788898a8b8c8d8e8f909192939495969798999a9b9c9d9e9fa0a1a2a3a4a5a6a7a8a9aaabacadaeafb0b1b2b3b4b5b6b7b8b9babbbcbdbebfc0c1c2c3c4c5c6c7c8c9cacbcccdcecfd0d1d2d3d4d5d6d7d8d9dadbdcdddedfe0e1e2e3e4e5e6e7e8e9eaebecedeeeff0f1f2
```

carlescufi, 2021-02-11:
> The reason for the `Unable to allocate TX context` is because bt_gatt_notify is called from the system work queue. We cannot block from this thread, since this is the thread that would release them, and blocking would cause a deadlock.

@Oxymoron79 could you please move the call to `bt_gatt_notify()` out of the system workqueue and then see if this still reproduces the issue?

Oxymoron79, 2021-02-11:
> @Oxymoron79 could you please move the call to `bt_gatt_notify()` out of the system workqueue and then see if this still reproduces the issue?

@carlescufi You mean to use a separate workqueue to call `bt_gatt_notify()` in, right? I did that and it indeed solves the issue.

I put that version in the [notification_workqueue](https://github.com/Oxymoron79/zephyr-ble-peripheral/tree/notification_workqueue) branch of my repository.

carlescufi, 2021-02-18:
> @carlescufi You mean to use a separate workqueue to call `bt_gatt_notify()` in, right? I did that and it indeed solves the issue.

Right, so this probably requires a fix in the documentation. @joerchan will send a PR.
 
> I put that version in the [notification_workqueue](https://github.com/Oxymoron79/zephyr-ble-peripheral/tree/notification_workqueue) branch of my repository.

Can we close this GitHub issue now?

joerchan, 2021-03-01:
> Can we close this GitHub issue now?

Assuming that the "stuck" was because the error code returned (`data_work_handler: bt_gatt_notify returned: -57`) wasn't handled then I think this can be closed with the documentation update.

closed 2021-03-03 by carlescufi

rajithavk, 2021-06-02:
Hi, I am testing the throughput for hci_uart on 1M PHY. I've used the same source from @Oxymoron79 . Thanks for that work.
My configuration is
Kernel : 5.19+
Bluez : 5.58

UART baud rate : 460800
Notification data length : 20Bytes.

I'm using 2 Adafruit_feather_nrf52840 for this purpose and a custom linux host based on BBBW. Somehow my observations are abnormal from what  I see. So i'm posting below plots for different n-number of notifications and i-interval between notifications.

[image]

[image]

[image]

[image]

I created the issue here since it's more relatable to the original issue here, otherwise please accept my apologies.

So what could be the reason for the periodic jitter, especially visible in the 2nd and 4th configuration?

PR #32740 (2021-03-03): Bluetooth: host: Document not waiting for TX context from sys workqueue
Document the GATT APIs not waiting for TX context when being called
from the system workqueue context. This is because the TX contexts
are freed by the system workqueue so blocking would cause a deadlock.

The number of TX contexts are by default equal to the number of
TX buffers in the system. Since TX contexts is allocated after a buffer,
but freed from a lower priority than the buffer then there can be more
allocated TX contexts than TX buffers.

Remove the incorrect documentation note about the indication callback
bein [... cut]