zephyrproject-rtos/zephyr #31745: Conan package manager support (using cross building)

RonAmihai, 2021-01-28:
**Is your feature request related to a problem? Please describe.**
[Conan package manager](https://conan.io/) is a clever way of managing C/C++ project's 3rd party packages/dependencies. 
It can also be used for embedded development using cross building, which requires to manage Conan profiles according to the desired host/target arch platforms and their corresponding toolchains. 

There are many Conan packages which can be compiled successfully with Zephyr and make the development far better.

Configuring Conan manually, especially in case of embedded development & cross building, can be cumbersome. 
Using Zephyr-SDK alongside Conan can be even harder - because of the need to manually configure many profiles for each toolchain & run Conan with the right profile before building for the desired arch.

**Describe the solution you'd like**
Adding Conan support to Zephyr, including appropriate config option, CMake support and internal-predefined conan profiles for every supported Zephyr toolchain. 

The goal is to be able to integrate Zephyr application with Conan, using nothing but a `conanfile.txt` (for listing packages/dependencies for the application) and enable some Conan configs (like `CONFIG_USE_CONAN`). 
It shouldn't even require the user to configure Conan in the project's `CMakeLists.txt`, just add `conanfile.txt` and enable Conan through a Zephyr config.

Zephyr's tree will take care of running Conan onto the project's build directory using the right Conan profile (according to the current Zephyr toolchain, whether it is SDK or specific) & configure/build the Zephyr application afterwards.

**Describe alternatives you've considered**
Configure Conan manually (including modification of the Zephyr app project's `CMakeLists.txt`, adding multiple profiles to support each toolchain/arch, and running Conan from scratch before building the project for different boards)

carlescufi, 2021-01-29:
@RonAmihai Thanks for the suggestion. Since, AFAIK, we don't have active Conan users so far, could you perhaps send a Draft PR with the files and changes required to achieve that support? Would the SDK also require changes?

RonAmihai, 2021-01-29:
> @RonAmihai Thanks for the suggestion. Since, AFAIK, we don't have active Conan users so far, could you perhaps send a Draft PR with the files and changes required to achieve that support? Would the SDK also require changes?

The SDK won't require any change in my opinion. 

The only required thing to modify is actually Zephyr's CMake tree, with some side configuration files (mostly Conan profiles for each supported toolchain/arch & maybe some Conan scripts). 

Maybe the Conan install command can be separated into some `west` logic, but this can complicate regular/clean CMake usage with Zephyr, so I think It's better to "force" `conan install --build=missing` as part of the CMake tree by default if Conan support is enabled for the project.

I would like to try helping to Darft PR Conan support soon

tim-hilt, 2023-11-15:
I‘m also interested in this. However, wouldn’t this PR be better suited for the conan-center repository? It shouldn’t be Zephyrs concern, but rather Conan’s!

Ju1He1, 2024-09-23:
usually the conan center only hosts 3rd party recipes (i.e. a recipe to build spdlog on any platform).
The conan integration must be located in the consumer project (in this case zephyr).

Our non zephyr projects use [cmake-conan](https://github.com/conan-io/cmake-conan) to have an easy to use conan integration.
Maybe you guys can integrate conan using cmake-conan  :)

jasal82, 2025-01-20:
+1 interest