zephyrproject-rtos/zephyr #31748: boards:lpcxpresso55s69: Manual toggling of CS required with ETH Click shield

microbuilder, 2021-01-28:
When using the SPI interface on the LPC55S69 and the Mikroe ETH Click shield (likely other shields as well), the SPI CS line must be manually toggled as a GPIO pin, requiring the following change to [dts/arm/nxp/nxp_lpc55S6x_common.dtsi](https://github.com/zephyrproject-rtos/zephyr/blob/78c77da578987d7b4774509347d14f9c04eb07da/dts/arm/nxp/nxp_lpc55S6x_common.dtsi#L237-L240):

```
	hs_lspi: spi@9f000 {
		compatible = "nxp,lpc-spi";
		/* Enabling cs-gpios below will allow using GPIO CS,
		 rather than Flexcomm SS */
		cs-gpios = <&gpio0 20 GPIO_ACTIVE_LOW>,
			<&gpio1 1 GPIO_ACTIVE_LOW>,
			<&gpio1 12 GPIO_ACTIVE_LOW>,
			<&gpio1 26 GPIO_ACTIVE_LOW>;
		reg = <0x9f000 0x1000>;
		interrupts = <59 0>;
		clocks = <&syscon MCUX_HS_SPI_CLK>;
		label = "HS_LSPI";
		status = "disabled";
		#address-cells = <1>;
		#size-cells = <0>;
	};
```

**To Reproduce**

Run and flash with the LPCXpresso55S69 (output on USB serial) with and without the .dtsi file changes:

```bash
$ west build -p auto -b lpcxpresso55s69_cpu0 zephyr/samples/net/dhcpv4_client -- -DSHIELD=mikroe_eth_click
$ pyocd flash -t lpc55s69 build/zephyr/zephyr.elf 
```

**Expected behavior**

The manual GPIO selection and file change in this .dtsi file should not be required here.

microbuilder, 2021-02-08:
> @microbuilder Is this a regression after #26371?

@MaureenHelm As I understand it, the CS line always had to be manually selected as a GPIO pin, but @agansari is the person to ping.

nashif, 2021-02-09:
@agansari ping

elcritch, 2021-08-11:
This still appears to be an issue as of v2.6. I'm using boards:lpcxpresso55s69 and mikroe_eth_click shield with Zephyr `v2.6-branch` (currently at 837ab4a9). The underlying issue appears to be a conflict between using the NXP "flexcomm" SPI config which appears to utilize memory addresses, not gpio indexes/offsets. 

The main branch as of now appears to have refactored this code in PR #37442, hopefully resolving this for future release(s). I haven't tested this. 

For others potentially running across this issue, the fix I'm using is to provide my own `lpcxpresso55s69` board override (`my_proj/boards/lpcxpresso55s69_cpu0.overlay`) that incorporates the Flexcomm/CS changes noted above: 

```dts
/delete-node/ &eth_click;

&peripheral {

	hs_lspi: spi@9f000 {
		compatible = "nxp,lpc-spi";
		/* Enabling cs-gpios below will allow using GPIO CS,
		 rather than Flexcomm SS */
		cs-gpios = <&gpio0 20 GPIO_ACTIVE_LOW>,
			<&gpio1 1 GPIO_ACTIVE_LOW>,
			<&gpio1 12 GPIO_ACTIVE_LOW>,
			<&gpio1 26 GPIO_ACTIVE_LOW>;
		reg = <0x9f000 0x1000>;
		interrupts = <59 0>;
		clocks = <&syscon MCUX_HS_SPI_CLK>;
		label = "HS_LSPI";
		status = "disabled";
		#address-cells = <1>;
		#size-cells = <0>;
	};
};

&mikrobus_spi {
	status = "okay";

	eth_click: eth_click@1 {
		compatible = "microchip,enc28j60";
		reg = <1>;
		local-mac-address = [01 01 01 01 02 03];
		/* Errata B7/1 specifies min 8Mhz, 20MHz max according to RM */
		spi-max-frequency = <14000000>;
		int-gpios = <&mikrobus_header 7 GPIO_ACTIVE_LOW>;	/* INT */
		// cs-gpios = <&mikrobus_header 2 GPIO_ACTIVE_LOW>;	/* CS */
		label = "ETH_CLICK";
	};
};

```

microbuilder, 2021-08-11:
@elcritch Thanks for the update. Re-opening the issue. If you have any bandwidth for a PR, I'm sure it would be appreciated!

dleach02, 2021-09-14:
@elcritch, would your changes work if applied to boards/shields/mikroe_eth_click/boards/lpcxpresso55s69_cpu0.overlay?

Also, I note that you have a different value for spi-max-frequency then what is found in the current overlay file. Is there a problem with the existing value in the overlay?

elcritch, 2021-09-14:
My changes should work with `boards/shields/mikroe_eth_click/boards/lpcxpresso55s69_cpu0.overlay`. It also works with Adafruit nRF boards as well so probably nice to add them. The SPI change was just a tweak to make it a little more robust on the different Mikroe Click expansion boards (I'm using a Mikroe shield for the Arduino-Uno header to provide 2-MikroeClick boards). 

I'll be looking into creating some PR's next week (hopefully). I've got a few other boards tweaks I've run across as well.

dleach02, 2021-09-15:
A PR would fall under "bug fix" so it would be good if you could get one in soon for the 2.7 release being worked now.

dleach02, 2021-10-04:
@elcritch, did you get time to look at doing a PR?

elcritch, 2021-10-05:
No, but it's a really simple patch. Mainly I'm not too familiar with what's required for a PR for Zephyr. However, I'll take a look into it tomorrow.

PR #41129 (2021-12-17): shields: mikroe_eth_click: Fix CS pins for LPC55S69
Fix CS pins for Mikroe Click Ethernet shield overlay on LPC55S69's. Should resolve #31748.
closed 2021-12-17 by cfriedt