zephyrproject-rtos/zephyr #31757: GCC compiler option should include '-mcpu=hs38' for HSDK

jacoba-at, 2021-01-28:
**Describe the bug**
When building Zephyr for the HSDK board, it passed to gcc the option '-mcpu=hs'. This creates code that misses some processor functionality. For example if you use modulo C operator (a % b), it calls a function to perform this operation, instead of using the 'REM' assembly instruction.

**To Reproduce**
Steps to reproduce the behavior:
1. west -v build -b hsdk ....
2. Observe the compiler flags passed to gcc.

**Expected behavior**
I would expect to see '-mcpu=hs38' and not 'mcpu=hs'

**Impact**
I ran a test that uses heavily the modulo operator and it ran x4 slower without the -mcpu=hs38 flag.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain : Zephyr sdk-0.11.4
 - Commit Zypher 2.4.0

**Additional context**
How can I add more flags or override some to the zephyr compilation? I tries adding zephyr_cc_option() to my application CMakeLists.txt , but didn't effected to compilation flags.

nashif, 2021-01-29:
@abrodkin FYI

nashif, 2021-02-09:
@abrodkin  @evgeniy-paltsev  ping

abrodkin, 2021-02-09:
We do plan to address that, but the idea was to do it for the v2.6 to not add more troubles in yet stabilized v2.5.
Or @nashif you prefer us to do that last second fix?
@jacoba-at how critical is this one for you? Please note that in case of MW tools we use right set of options (https://github.com/zephyrproject-rtos/zephyr/blob/master/soc/arc/snps_arc_hsdk/CMakeLists.txt#L14) which matches HSDK's SoC .tcf.

jacoba-at, 2021-02-10:
I do have a WA for this by patching the GCC cmake file, so no critical. As for the support of HSDK with MW toolchain (#30162) that you mentioned, would this be available for 2.5., or only in 2.6?

abrodkin, 2021-02-10:
@jacoba-at since we're talking about an issue in libs bundled with MW toolchain it doesn't have any direct connection with a Zephyr release, instead it has to do with MW toolchain release.

That said if this atomics problem gets fixed in the next release 2021.03 of MW tools, then you'll be able to use that newer MW version with Zephyr 2.5 as well as with 2.6 etc.

In the meantime I'd strongly suggest you to file a formal customer case via Synopsys Solvnet (https://solvnet.synopsys.com) to add more visibility to our engineering team referring to https://github.com/zephyrproject-rtos/zephyr/issues/30162.

jacoba-at, 2021-04-12:
Please remove the stale label.

abrodkin, 2021-04-12:
@jacoba-at just FYI we do remember about adding `-mcpu` [and probably more compiler options] to match real SoC configuration. But now we're about to submit support of ARCv3 64-bit processors (see https://www.synopsys.com/dw/ipdir.php?ds=arc-HS6x-processors) and to not add an extra turbulence in the CI and verification results (as a part of the aforementioned port we have quite some re-work of the existing code for better co-existence of ARCv2 & ARcv3 ports) we decided to postpone a bit this change of compiler options.

So stay tuned and I hope to get this one addresses soon as well and do that right before v2.6 code freeze later this May.

jacoba-at, 2021-05-26:
> @jacoba-at <https://github.com/jacoba-at> please see #35415
> <https://github.com/zephyrproject-rtos/zephyr/pull/35415>, instead of
> -mcpu=hs38 that uses -mcpu=archs, which has the same effect.
> ok to close this once that PR gets merged?

Looks good to me, thanks.
-Jacob

closed 2021-05-27 by galak