zephyrproject-rtos/zephyr #31761: logging:buffer_write with len=0 causes kernel panic

eanderlind, 2021-01-28:
https://github.com/zephyrproject-rtos/zephyr/blob/66d1b3ce10635dd9988898479ce8b8f6516a98df/subsys/logging/log_output.c#L134
buffer_write() doesn't handle being called with len=0. Simplest fix would be to convert to while(len>0) {} loop.
Also unclear if use of `log_output->control_block->offset` is thread-safe.

See below gdb snippet where call occurs -we exit from an ISR and resume processing RTT logging. Unclear why len has been reset to zero before final "\n" in string has been output.(Below output is for v2.3.0-rc1-ncs1, so line numbering is off, however this function hasn't changed.) 
```
(gdb) bt
#0  0x00029d1c in assert_post_action (file=0x4 <z_finalize_fd+4> "\265r", file@entry=0x3ce03 "WEST_TOPDIR/zephyr/kernel/include/ksched.h", 
    line=line@entry=267) at zephyr/lib/os/assert.c:45
#1  0x000263aa in z_sched_lock () at xx/zephyr/kernel/include/ksched.h:267
#2  z_impl_k_mutex_unlock (mutex=mutex@entry=0x20002278 <rtt_term_mutex>) at xx/zephyr/kernel/mutex.c:233
#3  0x00003fb0 in k_mutex_unlock (mutex=<optimized out>) at zephyr/include/generated/syscalls/kernel.h:728
#4  data_out_block_mode (data=0x20004a45 <char_buf> "\resh initialized\001", length=0, ctx=<optimized out>)
    at zephyr/subsys/logging/log_backend_rtt.c:214
#5  0x0002a066 in buffer_write (outf=0x3f71 <data_out_block_mode>, buf=0x20004a45 <char_buf> "\resh initialized\001", len=0, 
    ctx=0x0 <z_finalize_fd>) at xx/zephyr/subsys/logging/log_output.c:149
#6  0x0002a33a in log_output_flush (log_output=log_output@entry=0x33b20 <log_output>)
    at xx/zephyr/subsys/logging/log_output.c:158
#7  0x00003f14 in log_backend_std_panic (log_output=0x33b20 <log_output>)
    at xx/zephyr/subsys/logging/log_backend_std.h:51
(removed rest of long call stack - multiple ISRs and threads also panic, since they can't log faults )
```

nordic-krch, 2021-02-05:
Can you elaborate more about the scenario where you hit that issue? len=0 shouldn't be the issue there. I've run a scenario where it was 0 and seen no fault. I guess that it might have something to do with panic occurring while logging is processing.

eanderlind, 2021-02-05:
Yes, I am not sure, but it looks like RTT logger (#3-#4) in the  RTT data_out_block() is still holding its mutex lock and the kernel is doing a priority inversion which causes an ISR to run and (#0-#2) the zephyr k_mutex_unlock() implementation to 
assert because it is an ISR. My interpretation is this is caused by the RTT logger already holding a lock and requested (by an ISR) to 
log some more data or flush the log. buffer_write always calls the function at least once even if len=0) 
In the (long) backtrace, logging is invoked multiple times (with len=0) - I suspect this is when tries to flush the log.. 

Simplest would be to only call log function if len>0, since logs in general will hold locks on their logging buffer.

Here is earlier output showing how the panic was started (it is possible it was triggered because running in gdb and ISR was running late). Nevertheless, IMHO kernel logging should still work so that can debug.
```
#28 z_impl_log_panic () at xx/zephyr/subsys/logging/log_core.c:576
#29 0x0002364a in log_panic () at zephyr/include/generated/syscalls/log_ctrl.h:34
#30 k_sys_fatal_error_handler (reason=<optimized out>, esf=<optimized out>)
    at xx/nrf/lib/fatal_error/fatal_error.c:22
#31 0x00024706 in z_fatal_error (reason=3, esf=0x200070b0 <z_interrupt_stacks+2032>)
    at xx/zephyr/kernel/fatal.c:123
#32 0x000061d6 in _oops () at xx/zephyr/arch/arm/core/aarch32/swap_helper.S:465
#33 <signal handler called>
#34 lll_done (param=<optimized out>)
    at xx/zephyr/subsys/bluetooth/controller/ll_sw/nordic/lll/lll.c:313
#35 0x00030428 in radio_nrf5_isr_body ()
    at xx/zephyr/subsys/bluetooth/controller/ll_sw/nordic/lll/lll.c:76
```

closed 2021-04-21 by github-actions[bot]