zephyrproject-rtos/zephyr #31769: Twister:  AttributeError: 'NoneType' object has no attribute 'serial_pty'

LeiW000, 2021-01-29:
This is an intermittent issue. please see below.

./scripts/twister --device-testing --device-serial /dev/ttyACM0 -p reel_board -T tests/kernel
Renaming output directory to /home/leiw/Documents/src/zephyrproject/zephyr/twister-out.1
INFO    - Zephyr version: v2.5.0-rc1-136-g30de4b5dd952
INFO    - JOBS: 8
INFO    - Building initial testcase list...
INFO    - 109 test suites (109 configurations) selected, 19 configurations discarded due to filters.

Device testing on:

| Platform   | ID   | Serial device   |
|------------|------|-----------------|
| reel_board |      | /dev/ttyACM0    |

INFO    - Adding tasks to the queue...
INFO    - Added initial list of jobs to queue
INFO    - Total complete:   36/  90  40%  skipped:   26, failed:    0
ERROR   - reel_board                tests/kernel/mem_protect/sys_sem/kernel.memory_protection.sys_sem FAILED: Timeout
ERROR   - see: /home/leiw/Documents/src/zephyrproject/zephyr/twister-out/reel_board/tests/kernel/mem_protect/sys_sem/kernel.memory_protection.sys_sem/handler.log
INFO    - Total complete:   40/  90  44%  skipped:   27, failed:    1
ERROR   - reel_board                tests/kernel/mem_protect/futex/kernel.futex        FAILED: Timeout
ERROR   - see: /home/leiw/Documents/src/zephyrproject/zephyr/twister-out/reel_board/tests/kernel/mem_protect/futex/kernel.futex/handler.log
INFO    - Total complete:   78/  90  86%  skipped:   28, failed:    2Process Process-3:
Traceback (most recent call last):
  File "/usr/lib/python3.8/multiprocessing/process.py", line 315, in _bootstrap
    self.run()
  File "/usr/lib/python3.8/multiprocessing/process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "/home/leiw/Documents/src/zephyrproject/zephyr/scripts/pylib/twister/twisterlib.py", line 3204, in pipeline_mgr
    pb.process(pipeline, done_queue, task, lock, results)
  File "/home/leiw/Documents/src/zephyrproject/zephyr/scripts/pylib/twister/twisterlib.py", line 2272, in process
    self.run()
  File "/home/leiw/Documents/src/zephyrproject/zephyr/scripts/pylib/twister/twisterlib.py", line 2482, in run
    instance.handler.handle()
  File "/home/leiw/Documents/src/zephyrproject/zephyr/scripts/pylib/twister/twisterlib.py", line 696, in handle
    serial_pty = hardware.serial_pty
AttributeError: 'NoneType' object has no attribute 'serial_pty'
INFO    - Total complete:   89/  90  98%  skipped:   29, failed:    2
INFO    - 77 of 80 test configurations passed (96.25%), 2 failed, 29 skipped with 0 warnings in 705.09 seconds
INFO    - In total 796 test cases were executed, 283 skipped on 1 out of total 329 platforms (0.30%)
INFO    - 79 test configurations executed on platforms, 1 test configurations were only built.

Hardware distribution summary:

| Board      | ID   |   Counter |
|------------|------|-----------|
| reel_board |      |        79 |
INFO    - Saving reports...
INFO    - Writing xunit report /home/leiw/Documents/src/zephyrproject/zephyr/twister-out/twister.xml...
ERROR   - reel_board/tests/kernel/threads/thread_init/kernel.threads.init: No status
INFO    - Writing xunit report /home/leiw/Documents/src/zephyrproject/zephyr/twister-out/twister_report.xml...
INFO    - Run completed

nashif, 2021-02-04:
@LeiW000  can you reliably reproduce this?

LeiW000, 2021-02-04:
> @LeiW000 can you reliably reproduce this?
To reproduce it, need to try several times. Also, Peng told me that it was seen in log of daily testing.

PR #31992 (2021-02-05): twister: fix possible race and  hardware map parsing fix
Fix possible race when determining available devices. Find available
device and return it in the same operation to avoid the race.

Also fix issue where we did not parse fixtures from hardware map.

Fixes #31769
Fixes #31948
commit 2a740535 (2021-02-05): twister: fix possible race

Fix possible race when determining available devices. Find available
device and return it in the same operation to avoid the race.

Fixes #31769
closed 2021-02-05 by nashif