zephyrproject-rtos/zephyr #31789: samples/scheduler/metairq_dispatch: Regression after 6b17f2fb2a75f2dab7dc8ef19c67555df742fb22

erwango, 2021-01-29:
**Describe the bug**
Following merge of 6b17f2fb2a75f2dab7dc8ef19c67555df742fb22, sample `samples/scheduler/metairq_dispatch` is now failed on 2 platforms: nucleo_wb55rg and nucleo_f091rc.
Sample works correctly on other boards. Likely the common characteristic is they are the slowest in our bench.

**To Reproduce**
twister

**Expected behavior**
Sample should run w/o error

**Logs and console output**
```
*** Booting Zephyr OS build v2.5.0-rc1-106-g2712d74cdcbf  ***
I: Starting Thread0 at priority -2
I: Starting Thread1 at priority -1
I: Starting Thread2 at priority 0
I: Starting Thread3 at priority 1
ERROR: Queue full, event dropped!
ERROR: Queue full, event dropped!
ERROR: Queue full, event dropped!
ERROR: Queue full, event dropped!

```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK
 - Commit SHA or Version used: From 6b17f2fb2a75f2dab7dc8ef19c67555df742fb22

erwango, 2021-01-29:
I'm seeing the same effect with `tests/kernel/semaphore/semaphore (only on nucleo_f091rc this time)

```
 : START - test_sem_measure_timeouts

    Assertion failed at ../../src/main.c:990: test_sem_measure_timeouts: ((en)
time missmatch - expected 1, got 1
 FAIL - test_sem_measure_timeouts
```

erwango, 2021-01-29:
@ceolin I'm available for testing fixes if needed (well, starting monday :-))

nashif, 2021-02-02:
@andyross FYI

andyross, 2021-02-02:
Can you try #31841 ? That fixes an edge case where a pre-existing bug with slice timeouts got exposed by the fix you bisected to. I was seeing that fail with non-standard clock rates on a renode riscv platform and the tickless_context test.  But it's not impossible it's the root cause here too.

PR #31891 (2021-02-02): Timeslice fixups
Fix an edge case that snuck in with the recent fix: if timeslicing is
enabled, the CPU's slice_ticks will be zero, and thus match a timeout
object's dticks value of zero, and thus get suppressed (because "we
already have a timeout scheduled for that") incorrectly.

Fixes #31789
commit 544475d8 (2021-02-02): kernel/timeout: Schedule zero-time timeouts

Fix an edge case that snuck in with the recent fix: if timeslicing is
enabled, the CPU's slice_ticks will be zero, and thus match a timeout
object's dticks value of zero, and thus get suppressed (because "we
already have a timeout scheduled for that") incorrectly.

Fixes #31789
closed 2021-02-02 by nashif