zephyrproject-rtos/zephyr #31797: need 2.5 release notes on switch to k_heap from mem_pool

pabigot, 2021-01-29:
First, IMO the release notes should mention that these APIs are now officially deprecated, and repeat the text from the 2.4 release notes saying that they're going away in 2.6.

Second, as described in https://github.com/zephyrproject-rtos/zephyr/issues/31524#issuecomment-770070946, the replacement of using `k_heap` has a behavior difference that can result in applications that formerly allocated from mem_pools running out of memory with the replacement heap data structure.  I don't think that's avoidable, but it too should be mentioned in the release notes as it's likely something other than littlefs broke as a result.

I'm tagging this as a bug because it's a behavior change that can break existing applications that isn't documented, but if it should be something else please recategorize it.

pabigot, 2021-01-29:
@nashif @andyross FYI

nashif, 2021-02-01:
are we going to start filing bugs against release notes that are WIP and part of the release process? Most of the sections of document are still empty and now is the time to start filling those, if the kernel section is not ready by rc3, file a bug please or reopen this.

closed 2021-02-01 by nashif