zephyrproject-rtos/zephyr #31807: USB DFU Broken for a number of ARM boards

alfonsotames, 2021-01-31:
**Describe the bug**

Building the USB DFU subsys sample fails with a file "bootutil_public.h" not found error. Where is even bootutil_public.h supposed to be??? Tested with some boards (STM32 and SAM) preconfigured with mcuboot partitions like stm32f4_disco, sam_e70_xplained, sam_v71_xult and  blackpill_f401ce. Same results for all.

**To Reproduce**
Steps to reproduce the behavior:

$ cd zephyr/samples/subsys/usb/dfu
$ west build -b blackpill_f401ce

**Expected behavior**
A clean build.

**Impact**
USB DFU is broken.

**Logs and console output**
west build outputs:

-- west build: building application
[1/160] Preparing syscall dependency handling

[145/160] Building C object zephyr/subsys/dfu/boot/CMakeFiles/subsys__dfu__boot.dir/mcuboot.c.obj
FAILED: zephyr/subsys/dfu/boot/CMakeFiles/subsys__dfu__boot.dir/mcuboot.c.obj 
ccache /usr/local/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -DBUILD_VERSION=v2.5.0-rc1-172-gfd382b4c480b -DCORE_CM4 -DHSE_VALUE=25000000 -DKERNEL -DSTM32F401xE -DUSE_FULL_LL_DRIVER -DUSE_HAL_DRIVER -D_FORTIFY_SOURCE=2 -D__PROGRAM_START -D__ZEPHYR__=1 -I../../../../../include -Izephyr/include/generated -I../../../../../soc/arm/st_stm32/stm32f4 -I../../../../../drivers -I../../../../../soc/arm/st_stm32/common/. -I../../../../../soc/arm/st_stm32/common -I../../../../../subsys/usb -I/home/mufufu/zephyr/modules/hal/cmsis/CMSIS/Core/Include -I/home/mufufu/zephyr/modules/hal/stm32/stm32cube/stm32f4xx/soc -I/home/mufufu/zephyr/modules/hal/stm32/stm32cube/stm32f4xx/drivers/include -I/home/mufufu/zephyr/modules/hal/stm32/stm32cube/stm32f4xx/drivers/include/Legacy -I/home/mufufu/zephyr/modules/hal/stm32/stm32cube/common_ll/include -isystem ../../../../../lib/libc/minimal/include -isystem /usr/local/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/9.2.0/include -isystem /usr/local/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/9.2.0/include-fixed -Os -imacros /home/mufufu/zephyr/zephyr/samples/subsys/usb/dfu/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -mcpu=cortex-m4 -mthumb -mabi=aapcs -imacros /home/mufufu/zephyr/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wno-address-of-packed-member -Wno-unused-but-set-variable -Werror=implicit-int -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=/home/mufufu/zephyr/zephyr/samples/subsys/usb/dfu=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/mufufu/zephyr/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/home/mufufu/zephyr=WEST_TOPDIR -ffunction-sections -fdata-sections -std=c99 -nostdinc -MD -MT zephyr/subsys/dfu/boot/CMakeFiles/subsys__dfu__boot.dir/mcuboot.c.obj -MF zephyr/subsys/dfu/boot/CMakeFiles/subsys__dfu__boot.dir/mcuboot.c.obj.d -o zephyr/subsys/dfu/boot/CMakeFiles/subsys__dfu__boot.dir/mcuboot.c.obj   -c /home/mufufu/zephyr/zephyr/subsys/dfu/boot/mcuboot.c
/home/mufufu/zephyr/zephyr/subsys/dfu/boot/mcuboot.c:19:10: fatal error: bootutil/bootutil_public.h: No such file or directory
   19 | #include "bootutil/bootutil_public.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
[152/160] Building C object zephyr/kernel/CMakeFiles/kernel.dir/sched.c.obj
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /home/mufufu/zephyr/zephyr/samples/subsys/usb/dfu/build

**Environment (please complete the following information):**

- Zephyr OS build v2.5.0-rc1-172-gfd382b4c480b
- Host OS: Ubuntu Linux
-  Zephyr SDK

jfischer-no, 2021-01-31:
I can not reproduce it. Please do 'west update' and retry.

> Building the USB DFU subsys sample fails with a file "bootutil_public.h" not found error. Where is even bootutil_public.h supposed to be???

`bootloader/mcuboot/boot/bootutil/include/bootutil/bootutil_public.h`

alfonsotames, 2021-01-31:
That was it! I thought I did it. Stupid me. Thanks a lot.

alfonsotames, 2021-01-31:
So, it works for stm32f4_disco, sam_e70_xplained, sam_v71_xult and blackpill_f401ce but fails for the STM32L4:

mufufu@mufu:~/zephyr/zephyr/samples/subsys/usb/dfu$ west build -b nucleo_l432kc
-- west build: generating a build system
Including boilerplate (Zephyr base): /home/mufufu/zephyr/zephyr/cmake/app/boilerplate.cmake
-- Application: /home/mufufu/zephyr/zephyr/samples/subsys/usb/dfu
-- Zephyr version: 2.5.0-rc1 (/home/mufufu/zephyr/zephyr)
-- Found Python3: /usr/bin/python3.8 (found suitable exact version "3.8.6") found components: Interpreter 
-- Found west (found suitable version "0.8.0", minimum required is "0.7.1")
-- Board: nucleo_l432kc
-- Cache files will be written to: /home/mufufu/.cache/zephyr
ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found toolchain: zephyr (/usr/local/zephyr-sdk)
-- Found dtc: /usr/local/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found BOARD.dts: /home/mufufu/zephyr/zephyr/boards/arm/nucleo_l432kc/nucleo_l432kc.dts
-- Generated zephyr.dts: /home/mufufu/zephyr/zephyr/samples/subsys/usb/dfu/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /home/mufufu/zephyr/zephyr/samples/subsys/usb/dfu/build/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: /home/mufufu/zephyr/zephyr/samples/subsys/usb/dfu/build/zephyr/include/generated/device_extern.h
Parsing /home/mufufu/zephyr/zephyr/samples/subsys/usb/dfu/Kconfig
Loaded configuration '/home/mufufu/zephyr/zephyr/boards/arm/nucleo_l432kc/nucleo_l432kc_defconfig'
Merged configuration '/home/mufufu/zephyr/zephyr/samples/subsys/usb/dfu/prj.conf'
Configuration saved to '/home/mufufu/zephyr/zephyr/samples/subsys/usb/dfu/build/zephyr/.config'
Kconfig header saved to '/home/mufufu/zephyr/zephyr/samples/subsys/usb/dfu/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 9.2.0
-- The CXX compiler identification is GNU 9.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/local/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
CMake Warning at ../../../../subsys/usb/CMakeLists.txt:22 (message):
  CONFIG_USB_DEVICE_VID has default value 0x2FE3.

  This value is only for testing and MUST be configured for USB products.

-- Configuring done
-- Generating done
-- Build files have been written to: /home/mufufu/zephyr/zephyr/samples/subsys/usb/dfu/build
-- west build: building application
[1/162] Preparing syscall dependency handling

[91/162] Building C object zephyr/CMakeFiles/zephyr.dir/subsys/dfu/img_util/flash_img.c.obj
FAILED: zephyr/CMakeFiles/zephyr.dir/subsys/dfu/img_util/flash_img.c.obj 
ccache /usr/local/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -DBUILD_VERSION=v2.5.0-rc1-172-gfd382b4c480b -DCORE_CM4 -DKERNEL -DSTM32L432xx -DUSE_FULL_LL_DRIVER -DUSE_HAL_DRIVER -D_FORTIFY_SOURCE=2 -D__PROGRAM_START -D__ZEPHYR__=1 -I../../../../../kernel/include -I../../../../../arch/arm/include -I../../../../../include -Izephyr/include/generated -I../../../../../soc/arm/st_stm32/stm32l4 -I../../../../../drivers -I../../../../../soc/arm/st_stm32/common/. -I../../../../../soc/arm/st_stm32/common -I../../../../../subsys/usb -I/home/mufufu/zephyr/modules/hal/cmsis/CMSIS/Core/Include -I/home/mufufu/zephyr/modules/hal/stm32/stm32cube/stm32l4xx/soc -I/home/mufufu/zephyr/modules/hal/stm32/stm32cube/stm32l4xx/drivers/include -I/home/mufufu/zephyr/modules/hal/stm32/stm32cube/stm32l4xx/drivers/include/Legacy -I/home/mufufu/zephyr/modules/hal/stm32/stm32cube/common_ll/include -isystem ../../../../../lib/libc/minimal/include -isystem /usr/local/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/9.2.0/include -isystem /usr/local/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/9.2.0/include-fixed -Os -imacros /home/mufufu/zephyr/zephyr/samples/subsys/usb/dfu/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -mcpu=cortex-m4 -mthumb -mabi=aapcs -imacros /home/mufufu/zephyr/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wno-address-of-packed-member -Wno-unused-but-set-variable -Werror=implicit-int -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=/home/mufufu/zephyr/zephyr/samples/subsys/usb/dfu=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/mufufu/zephyr/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/home/mufufu/zephyr=WEST_TOPDIR -ffunction-sections -fdata-sections -std=c99 -nostdinc -MD -MT zephyr/CMakeFiles/zephyr.dir/subsys/dfu/img_util/flash_img.c.obj -MF zephyr/CMakeFiles/zephyr.dir/subsys/dfu/img_util/flash_img.c.obj.d -o zephyr/CMakeFiles/zephyr.dir/subsys/dfu/img_util/flash_img.c.obj   -c /home/mufufu/zephyr/zephyr/subsys/dfu/img_util/flash_img.c
In file included from ../../../../../include/arch/arm/aarch32/arch.h:20,
                 from ../../../../../include/arch/cpu.h:19,
                 from ../../../../../include/kernel_includes.h:33,
                 from ../../../../../include/kernel.h:17,
                 from ../../../../../include/init.h:11,
                 from ../../../../../include/device.h:22,
                 from ../../../../../include/drivers/flash.h:26,
                 from ../../../../../include/storage/stream_flash.h:24,
                 from ../../../../../include/dfu/flash_img.h:11,
[... cut ...]
 1950 |  DT_IRQ_BY_NAME(DT_DRV_INST(inst), name, cell)
      |                 ^~~~~~~~~~~
/home/mufufu/zephyr/zephyr/drivers/usb/device/usb_dc_stm32.c:83:19: note: in expansion of macro 'DT_INST_IRQ_BY_NAME'
   83 | #define USB_IRQ   DT_INST_IRQ_BY_NAME(0, USB_IRQ_NAME, irq)
      |                   ^~~~~~~~~~~~~~~~~~~
/home/mufufu/zephyr/zephyr/drivers/usb/device/usb_dc_stm32.c:808:15: note: in expansion of macro 'USB_IRQ'
  808 |   irq_disable(USB_IRQ);
      |               ^~~~~~~
ninja: build stopped: subcommand failed.

jfischer-no, 2021-01-31:
I guess it is not USB DFU implementation issue that Nucleo L432KC board does not have USB support.

closed 2021-01-31 by jfischer-no