zephyrproject-rtos/zephyr #31817: mec15xxevb_assy6853: tests/boards/mec15xxevb_assy6853/i2c_api/ failed

s-kelvin, 2021-02-01:
**Describe the bug**
test_i2c_pca95xx in tests/boards/mec15xxevb_assy6853/i2c_api/ has assertion failure. 

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b mec15xxevb_assy6853 -p auto tests/boards/mec15xxevb_assy6853/i2c_api/
2. west flash
3. See error

**Expected behavior**
Test suite should pass.

**Logs and console output**
```
*** Booting Zephyr OS build v2.5.0-rc1-133-gdfd8e7522d11  ***
Running test suite i2c_test
===================================================================
START - test_i2c_pca95xx
E: Write error: -60

    Assertion failed at WEST_TOPDIR/zephyr/tests/boards/mec15xxevb_assy6853/i2c_api/src/main.c:48: test_i2c_pca95xx: (ret == 0 is false)
Failed to write data to i2c device
 FAIL - test_i2c_pca95xx
===================================================================
Test suite i2c_test failed.
===================================================================
PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - Linux Ubuntu 18.04.4 LTS
 - zephyr-v2.5.0-rc1-133-gdfd8e7522d11
 - Toolchain zephyr-sdk-0.12.0

chen-png, 2021-02-02:
need to make some jumpers setting to change to use I2C01 port according to the README.txt file in tests/boards/mec15xxevb_assy6853/i2c_api/, this will be passed.

chen-png, 2021-02-02:
tested after jumper setting according to README.txt file, it was passed.

*** Booting Zephyr OS build v2.5.0-rc1-191-g8595a390a7b5  ***
Running test suite i2c_test
===================================================================
START - test_i2c_pca95xx
 PASS - test_i2c_pca95xx
===================================================================
Test suite i2c_test succeeded
===================================================================
PROJECT EXECUTION SUCCESSFUL

closed 2021-02-02 by chen-png