zephyrproject-rtos/zephyr #31819: intel_adsp_cavs15: signing not correct thus download firmware failed

KangJianX, 2021-02-01:
**Describe the bug**
When run testcases we **use west script(sign.py) to signing** with image (no error and show completed) then invoke download script to load firmware, it has some error log at firmware loaded step. By debug, found signing not correct in fact(even if show completed without error), if we **use command to signing by ourselves without west script(sign.py)** it will download successful.

**To Reproduce**
Steps to reproduce the behavior:
1.  west build -b intel_adsp_cavs15 tests/kernel/thread/thread_apis/
2. west sign -t rimage -p ../modules/audio/sof/zephyr/ext/rimage/build/rimage -D ../modules/audio/sof/zephyr/ext/rimage/config/ -- -k ../modules/audio/sof/keys/otc_private_key.pem
3. boards/xtensa/intel_adsp_cavs15/tools/fw_loader.py -f /home/ztest/work/zephyrproject/zephyr/build/zephyr/zephyr.ri
4. See error

1.  west build -b intel_adsp_cavs15 tests/kernel/thread/thread_apis/
2. ../modules/audio/sof/zephyr/ext/rimage/build/rimage -k ../modules/audio/sof/keys/otc_private_key.pem -o /home/ztest/work/zephyrproject/zephyr/build/zephyr/zephyr.ri -c ../modules/audio/sof/zephyr/ext/rimage/config/apl.toml -i 3 -e /home/ztest/work/zephyrproject/zephyr/build/zephyr/bootloader.elf.mod /home/ztest/work/zephyrproject/zephyr/build/zephyr/zephyr.elf.mod
3. boards/xtensa/intel_adsp_cavs15/tools/fw_loader.py -f /home/ztest/work/zephyrproject/zephyr/build/zephyr/zephyr.ri
4. load firmware successful

**Environment (please complete the following information):**
OS: Fedora28
Toolchain: Zephyr-sdk-0.12.1
Commit ID: 3f1d1f5773ddaf

nashif, 2021-02-01:
@lyakh  can you check what is going on?

lyakh, 2021-02-02:
> @lyakh can you check what is going on?

@nashif Signing using `west sign` didn't work whereas signing calling `rimage` manually worked, so, looks like scripts have to be fixed. I see @KangJianX has assigned this to himself, so I think he is of the same opinion.

KangJianX, 2021-02-02:
@lyakh the root cause is the xman will be wrote in zephyr.ri, as my understanding, we run kernel testing not use SOF's way to load image by kernel DSP driver, so it do not need the xman in zephyr.ri and it cause check the image failed from SOF. I will add a option in sign.py to control if we need xman or not, what do you think of?

lyakh, 2021-02-02:
> @lyakh the root cause is the xman will be wrote in zephyr.ri, as my understanding, we run kernel testing not use SOF's way to load image by kernel DSP driver, so it do not need the xman in zephyr.ri [...]

@KangJianX but you did use the "-e" option when you ran `rimage` manually, so you did force it to generate the extended manifest. Can you see what exactly the `rimage` command line is when run from the `west sign` command?

KangJianX, 2021-02-02:
> @KangJianX but you did use the "-e" option when you ran `rimage` manually, so you did force it to generate the extended manifest. Can you see what exactly the `rimage` command line is when run from the `west sign` command?

Yes, use 'west sign' will show a command, it is same with run by manually. The difference is script will has a additional process like this:
`filenames = [out_xman, out_bin]
with open(out_tmp, 'wb') as outfile:
...
...`
Those code will write xman to zephyr.ri.

lyakh, 2021-02-02:
@KangJianX ok, so you want to add a parameter to `west sign` to avoid that? Got it, thanks!

KangJianX, 2021-02-02:
@lyakh Yes, I want add a option to avoid this after signing with rimage. Do you think it is make sense?

lyakh, 2021-02-02:
> @lyakh Yes, I want add a option to avoid this after signing with rimage. Do you think it is make sense?

@KangJianX sure, if it fixes your problem and doesn't change the default behaviour I don't see a problem with that.

closed 2021-02-11 by nashif