zephyrproject-rtos/zephyr #31822: tests: drivers: timer: Test drivers.timer.nrf_rtc_timer.stress fails on nrf52 platforms

PerMac, 2021-02-01:
**Describe the bug**
The test drivers.timer.nrf_rtc_timer.stress from /tests/drivers/timer/nrf_rtc_timer/ fails on nrf52dk_nrf52832 and nrf52840dk_nrf52840.

**To Reproduce**
Steps to reproduce the behavior:
1. have nrf52 platform connected
2. go to zephyr dir
3. run `./scripts/twister -p nrf52dk_nrf52832 --device-testing --device-serial /dev/ttyACM0 -v -v --inline-logs -T tests/drivers/timer/nrf_rtc_timer`
4. See error

**Expected behavior**
The test passes
**Impact**
Not clear

**Logs and console output**
The script enters test_stress test case then hangs and is terminated by a timeout.

**Environment (please complete the following information):**
 - OS: Ubuntu 18.04
 - Toolchain ZEphyr SDK
 - Commit SHA or Version used: v2.5.0-rc1-173-g3f1d1f5773

pabigot, 2021-02-08:
With my nrf52dk and nrf52840dk I can't reproduce the failures: three runs on each board all passed.  Is this *very* intermittent?

Also look at https://github.com/zephyrproject-rtos/zephyr/issues/32072#issuecomment-775091078 and try running the clock skew program it references.  If LFCLK is slower than HFCLK on the test boards (negative skew) the test may be failing because of that.

pabigot, 2021-02-08:
I have now reproduced it at the reported commit 3f1d1f5773, but have not on current mainline a281d10cff8d after 10 runs on the same board that failed at the reported commit.

Interestingly it's the non-ZLI test variant that failed for me:
```
tests/drivers/timer/nrf_rtc_timer/drivers.timer.nrf_rtc_timer FAILED Timeout
```

PerMac, 2021-02-09:
It seems the issue was fixed in the meantime. I saw no fails in recent versions in any of my setups. Closing the issue.

closed 2021-02-09 by PerMac