zephyrproject-rtos/zephyr #31836: Correct values of _msg_len arg in BT_MESH_MODEL_PUB_DEFINE macro

RAlexeev, 2021-02-01:
I ask about some explanation for choosing the correct values for **_msg_len** arg in [BT_MESH_MODEL_PUB_DEFINE](https://developer.nordicsemi.com/nRF_Connect_SDK/doc/1.4.2/zephyr/reference/bluetooth/mesh/access.html#c.BT_MESH_MODEL_PUB_DEFINE) macro, which defines the length of the publication message. As I understand, it should be exactly the length of Access payload (_3.7.3 Access payload, Mesh Profile v1.0.1_) from Access layer, which includes an opcode length, 1, 2 or 3 bytes, and the length of parameters.

For example, in both these examples https://github.com/zephyrproject-rtos/zephyr/blob/ad2fd44d7ad915e66765ba17ce53d4c5829a0efb/samples/boards/nrf/mesh/onoff_level_lighting_vnd_app/src/mesh/device_composition.c#L23-L24 and https://github.com/zephyrproject-rtos/zephyr/blob/ad2fd44d7ad915e66765ba17ce53d4c5829a0efb/samples/boards/nrf/mesh/onoff-app/src/main.c#L106-L107 model publication contexts are defined with different length. But why does publication declarations define different length in those two examples for the same models? They define length as a sum of two numbers. It seems that the first number, 2, defines opcode in both cases, and the second number in those sums defines the length of parameters. Opcodes have the same length in both examples and all messages, but lengths of parameters differ.

According to the Bluetooth mesh documentation:

- Generic OnOff Server publishes (_3.3.1.2.3 Sending a Generic OnOff Status message, Mesh Model v1.0.1_) Generic OnOff Status message (_3.2.1.4 Generic OnOff Status, Mesh Model v1.0.1_), which has the minimum length - 1 byte, and the maximum length - 3 bytes.
- Generic OnOff Client pulishes (_3.4.1.2.2 Sending Generic OnOff Set / Generic OnOff Set Unacknowledged messages, Mesh Model v1.0.1_) Generic OnOff Set message (_3.2.1.2 Generic OnOff Set, Mesh Model v1.0.1_), which has the minimum length - 2 bytes, and the maximum length - 4 bytes.

https://github.com/zephyrproject-rtos/zephyr/blob/ad2fd44d7ad915e66765ba17ce53d4c5829a0efb/samples/boards/nrf/mesh/onoff-app/src/main.c#L107 The publication length for Generic OnOff Client is "2 + 2", which is enough to publish only mandatory fields of Generic OnOff Set message.
https://github.com/zephyrproject-rtos/zephyr/blob/ad2fd44d7ad915e66765ba17ce53d4c5829a0efb/samples/boards/nrf/mesh/onoff_level_lighting_vnd_app/src/mesh/device_composition.c#L24 The publication length for Generic OnOff Client is "2 + 4", which is enough to publish both mandatory and optional fields of Generic OnOff Set message.
Why do those examples have different length in such about the same definitions?

And another question is about the publication length for Generic OnOff Server for the same examples.
https://github.com/zephyrproject-rtos/zephyr/blob/ad2fd44d7ad915e66765ba17ce53d4c5829a0efb/samples/boards/nrf/mesh/onoff_level_lighting_vnd_app/src/mesh/device_composition.c#L23 The publication length for Generic OnOff Server is "2 + 3", which is enough to publish both mandatory and optional fields of Generic OnOff Status message.
https://github.com/zephyrproject-rtos/zephyr/blob/ad2fd44d7ad915e66765ba17ce53d4c5829a0efb/samples/boards/nrf/mesh/onoff-app/src/main.c#L106 The publication length for Generic OnOff Server is "2 + 2", which is enough to publish mandatory field of Generic OnOff Status message, but not both optional fields. Generic OnOff Status message may have only two length - 1 or 3 (_3.2.1.4 Generic OnOff Status, Mesh Model v1.0.1_), but not 2, as it's defined in the example. So, tell me, please, is it a mistake or I understand something wrongly? I would like to understand clearly which length **_msg_len** arg in [BT_MESH_MODEL_PUB_DEFINE](https://developer.nordicsemi.com/nRF_Connect_SDK/doc/1.4.2/zephyr/reference/bluetooth/mesh/access.html#c.BT_MESH_MODEL_PUB_DEFINE) macro really defines.

Thank you in advance for any explanation.

trond-snekvik, 2021-02-03:
Hi, generally, these samples only allocates the buffer space they need, and not the maximum theoretical length of the message. In the onoff-app sample, none of the models use the full message length as defined by spec. As you point out though, the onoff server's publish parameters could be reduced to `2 + 1`, since it only adds one byte of data in addition to the opcode.

In the lighting sample, we are using the full message length, which is reflected in the buffer lengths. 

For your own applications, you only need to define the message length you need, but for most valid implementations of the models, that's usually the full message length.

RAlexeev, 2021-02-03:
Hi, Trond!

Many thanks for your perfect explanation, as ever! Now it seems that I understand the meaning of this argument completely. One extra allocated byte confused me. I was also puzzled because of another Bluetooth mesh app implementation for ESP-IDF, where they allocated only "2 + 1" bytes for Generic OnOff Client https://github.com/espressif/esp-idf/blob/release/v4.2/examples/bluetooth/esp_ble_mesh/ble_mesh_node/onoff_client/main/main.c#L68, although the documentation says about the minimum length - 2 bytes, and the maximum length - 4 bytes. But it's related neither Zephyr RTOS nor nRF Connect SDK.

Thank you, Trond!

trond-snekvik, 2021-02-03:
Np! Closing as resolved as the question is answered. Please reopen if you need additional clarification on this topic.

closed 2021-02-03 by trond-snekvik