zephyrproject-rtos/zephyr #31854: undefined reference to `sys_arch_reboot'

Dream-math, 2021-02-02:
When I build my own application, it gets error during linking to generate `zephyr_prebuilt.elf`
```
/home/hye/bin/zephyr-sdk-0.12.2/riscv64-zephyr-elf/bin/../lib/gcc/riscv64-zephyr-elf/10.2.0/../../../../riscv64-zephyr-elf/bin/ld: libzephyr.a(reboot.c.obj): in function `sys_reboot':
/home/hye/tmp/zephyrproject/zephyr/subsys/power/reboot.c:25: undefined reference to `sys_arch_reboot'
```
but with the same prj.conf file it works in `sample/hello_world`, there are no errors. I try to find the differences of this two applications

my application:
```
/home/hye/bin/zephyr-sdk-0.12.2/riscv64-zephyr-elf/bin/riscv64-zephyr-elf-g++ CMakeFiles/zephyr_prebuilt.dir/misc/empty_file.c.obj -o zephyr_prebuilt.elf  -Wl,-T linker.cmd -Wl,-Map=/home/hye/tmp/tvm/tests/micro/qemu/test_zephyr-host-workspace/2021-02-01T17-04-41/build/runtime/zephyr/zephyr_prebuilt.map -Wl,--whole-archive libzephyr.a ../app/libapp.a arch/common/libarch__common.a arch/arch/riscv/core/libarch__riscv__core.a lib/libc/newlib/liblib__libc__newlib.a lib/posix/liblib__posix.a subsys/random/libsubsys__random.a drivers/serial/libdrivers__serial.a -Wl,--no-whole-archive kernel/libkernel.a CMakeFiles/offsets.dir/arch/riscv/core/offsets/offsets.c.obj -L"/home/hye/bin/zephyr-sdk-0.12.2/riscv64-zephyr-elf/bin/../lib/gcc/riscv64-zephyr-elf/10.2.0/rv32imafc/ilp32f" -L/home/hye/tmp/tvm/tests/micro/qemu/test_zephyr-host-workspace/2021-02-01T17-04-41/build/runtime/zephyr -lgcc -Wl,--print-memory-usage /home/hye/tmp/tvm/tests/micro/qemu/zephyr-runtime/libtvm__libutvm_rpc_server.a /home/hye/tmp/tvm/tests/micro/qemu/zephyr-runtime/libtvm__libutvm_rpc_common.a /home/hye/tmp/tvm/tests/micro/qemu/zephyr-runtime/libtvm__libcommon.a /home/hye/tmp/tvm/tests/micro/qemu/zephyr-runtime/libtvm__libsrc.a /home/hye/tmp/tvm/tests/micro/qemu/zephyr-runtime/libtvm__libmodule.a arch/common/libisr_tables.a -mabi=ilp32f -march=rv32imafc -Wl,--gc-sections -Wl,--build-id=none -Wl,--sort-common=descending -Wl,--sort-section=alignment -Wl,-u,_OffsetAbsSyms -Wl,-u,_ConfigAbsSyms -nostdlib -static -no-pie -Wl,-X -Wl,-N -Wl,--orphan-handling=warn -lm -Wl,-lc -L"/home/hye/bin/zephyr-sdk-0.12.2/riscv64-zephyr-elf/riscv64-zephyr-elf"/lib/rv32imafc/ilp32f -Wl,-lgcc -lc -specs=nano.specs
/home/hye/bin/zephyr-sdk-0.12.2/riscv64-zephyr-elf/bin/../lib/gcc/riscv64-zephyr-elf/10.2.0/../../../../riscv64-zephyr-elf/bin/ld: libzephyr.a(reboot.c.obj): in function `sys_reboot':
/home/hye/tmp/zephyrproject/zephyr/subsys/power/reboot.c:25: undefined reference to `sys_arch_reboot'
```
hello world 
```
/home/hye/bin/zephyr-sdk-0.12.2/riscv64-zephyr-elf/bin/riscv64-zephyr-elf-g++ CMakeFiles/zephyr_prebuilt.dir/misc/empty_file.c.obj -o zephyr_prebuilt.elf  -Wl,-T linker.cmd -Wl,-Map=/home/hye/tmp/zephyrproject/zephyr/samples/hello_world/build/zephyr/zephyr_prebuilt.map -Wl,--whole-archive ../app/libapp.a libzephyr.a arch/common/libarch__common.a arch/arch/riscv/core/libarch__riscv__core.a lib/libc/newlib/liblib__libc__newlib.a lib/posix/liblib__posix.a subsys/random/libsubsys__random.a drivers/serial/libdrivers__serial.a -Wl,--no-whole-archive kernel/libkernel.a CMakeFiles/offsets.dir/arch/riscv/core/offsets/offsets.c.obj -L"/home/hye/bin/zephyr-sdk-0.12.2/riscv64-zephyr-elf/bin/../lib/gcc/riscv64-zephyr-elf/10.2.0/rv32imafc/ilp32f" -L/home/hye/tmp/zephyrproject/zephyr/samples/hello_world/build/zephyr -lgcc -Wl,--print-memory-usage arch/common/libisr_tables.a -mabi=ilp32f -march=rv32imafc -Wl,--gc-sections -Wl,--build-id=none -Wl,--sort-common=descending -Wl,--sort-section=alignment -Wl,-u,_OffsetAbsSyms -Wl,-u,_ConfigAbsSyms -nostdlib -static -no-pie -Wl,-X -Wl,-N -Wl,--orphan-handling=warn -lm -Wl,-lc -L"/home/hye/bin/zephyr-sdk-0.12.2/riscv64-zephyr-elf/riscv64-zephyr-elf"/lib/rv32imafc/ilp32f -Wl,-lgcc -lc -specs=nano.specs
```
still can't solve it, can anybody help me, thanks!

Dream-math, 2021-02-02:
It seems the error gets with multiple qemu simulated boards, it works on qemu_x86. Does it mean `sys_arch_reboot` need to be supported by soc, for example qemu_riscv32?

katsuster, 2021-02-03:
Thanks for reporting. Additional information.

**Reproduce this issue**
Set CONFIG_REBOOT = y with qemu_riscv32/64 boards. If something got wrong please tell me.

katsuster, 2021-02-03:
I hope PR #31901 fixes this issue.

Dream-math, 2021-02-03:
@katsuster thanks for your help, it does help to make the build work, but when I run my application, it will not exit normally, it just hangs out, I can only quit by Ctr+C, I am using a third library which finishes running by call sys_reboot, and when the board is qemu_riscv32, it can not quit right now. Thanks anyway.

katsuster, 2021-02-03:
OK, QEMU riscv-virt pc machine mode can reboot by writing special register.
I can add it by another PR...

katsuster, 2021-02-03:
Add PR #31918. I hope this helps.

Dream-math, 2021-02-03:
DO I need to add this new PR based on the old one? or recover the old one firstly, use this new alone? Thanks!

katsuster, 2021-02-03:
I checked on adding #31918 after #31901.
But I think #31918 will work alone and fine.

katsuster, 2021-02-12:
@nashif @MaureenHelm 
I think it's better also merging #31901 to avoid build error around CONFIG_REBOOT features for RISC-V in other and future added boards.

katsuster, 2021-03-02:
@nashif @MaureenHelm
Ping... How about this?

MaureenHelm, 2021-03-03:
> @nashif @MaureenHelm
> Ping... How about this?

@katsuster I don't understand your question?

katsuster, 2021-03-04:
@MaureenHelm 
Sorry for cofusing.
In this issue, only #31918 is merged. But I think it's better also merge #31901.

MaureenHelm, 2021-03-04:
> Sorry for cofusing.
> In this issue, only #31918 is merged. But I think it's better also merge #31901.

No problem, thanks for clarifying!

katsuster, 2021-03-05:
Thanks for merging. Yes we can close.

PR #31918 (2021-02-04): soc/riscv: support reboot for QEMU RISC-V VirtIO
This patch adds reboot function for QEMU RISC-V VirtIO board.

Fixes #31854
closed 2021-03-05 by MaureenHelm