zephyrproject-rtos/zephyr #31857: UART de-initilisation and re-initilisation

stoberblog, 2021-02-02:
**Is your enhancement proposal related to a problem? Please describe.**
I would like to during run time de-initialize (de-configure) the uart, and later re-initilise (re-configure) the uart. Initilisation of uart seems to be done on startup and there does not seem to be any api to allow de-configuring and re-configuring the uart after boot.

My reason for this is that I am implementing SDI-12 and which mostly resembles standard serial however has specific periods where the line must be held in a specific state (eg logic low for 12ms). These special periods I'm hoping to deconfigure the uart, reconfigure as GPIO, manually set state, wait for a period of time, and then deconfigure GPIO and return to uart mode.

Though I am working on an nrf52840, I know from previous use of STM32 devices that some devices effectively disconnect the physical pin from the GPIO and attach to an "alternate" peripheral (the uart) and as such, the reconfiguring between the uart and gpio is required (as opposed to setting up as a uart and then trying to write gpio out while in uart mode).

**Describe the solution you'd like**
Functions to de-initialize (de-configure) the uart, and later re-initilise (re-configure) the uart. Also these should cope with changing out of other modes if in those (specifically the GPIO mode).

**Describe alternatives you've considered**
Bit banging... I've considered bit banging the SDI-12 interface (it's 1200 baud so quite slow), but it would be cleaner to do in hardware (specially if can use DMA).

**Additional context**
For SDI-12 specification, please see http://www.sdi-12.org/current_specification/SDI-12_version-1_4-Jan-10-2019.pdf. Section 4.0 has the details on the protocol including the "break" (the 12ms logic low). Section 4.1 has the uart specifics. I'm planning on implementing the start bit as the first bit of an 8 bit uart frame.

stoberblog, 2021-02-03:
I've found that sending breaks is actually a standard use of UART in some circumstances and so probably this could be implemented instead (modorator's may want to change the issue title?).

According to this post in Nordic's devzone (https://devzone.nordicsemi.com/f/nordic-q-a/51615/send-uart-break), one way to do this (without requiring a reconfigure of the pin), is to set the baud quite low (depending on break length), disable parity and stop bits and then send a byte of 0x00. The other option discussed is the implementation discussed in my initial post.

I'm not sure if the above is the neatest way of implementing a break, but may be the most universal method and doesn't require significant effort to provide de-initilisation and re-initilisation of peripherals.

dcpleung, 2021-02-03:
Does your UART controller allow you to hold the UART pins in certain state? If so, the current line control API might work for your use case in the mean time while we figure out how to do de-/re-init.