zephyrproject-rtos/zephyr #31868: twister: Raise error/warning when unexisting platform called

PerMac, 2021-02-02:
**Is your enhancement proposal related to a problem? Please describe.**
When twister is called for an unexisting platform it will just report, that 0 tests were executed. Apparently, this can cause issues in internal CI plans. If a platform name changed, it was removed or typo was made, it can result in not detecting such changes in the CI as currently no error/warning is raised.

**Describe the solution you'd like**
When a given platform is not existing in zephyr, twister should report it by raising a warning/error when a user tries to run twister for such platform.

PerMac, 2021-02-02:
@nashif @galak Is there any reason why no warning/error is raised in the above case? Any objection against adding such a warning?

nashif, 2021-02-02:
submit a PR please.

commit 3194f750 (2021-12-15): twister: log error for non-existing platform call

Changes will log error message in following three situations:

1. Platform name pass in --platform option does not exist.
2. During using --all option, platform from platform_allow list does not
exist.
3. During using --integration option, platform from
integration_platforms list does not exist.

Fixes #31868
closed 2021-12-15 by nashif