zephyrproject-rtos/zephyr #31900: Out of tree user API auto documenation support part of offering

manoj153, 2021-02-03:
I have discussed this feature in slack and people confirmed this is not part of the zephyr ecosystem. I think this is a useful feature that would enhance the development experience. The idea is to document all the user introduced API and possibly source codes in multiple directory levels. 

I am an intermediate embedded C programmer to set up doxygen, breathe and sphinx along with CMake wasn't easy and straightforward. In fact, I lost motivation while trying to bring this feature up.

My tree contains only user-level source code and separated from Zephyr tree to have a dedicated docs folder on a top directory.

gregshue, 2021-02-03:
@manoj153 I'm not sure what all you have in mind.  Consider:

1. Are you wanting to reuse/extend the structure/organization of the Zephyr documentation system?
2. Are you wanting to specify your own logo/versioning/platform name (e.g. "Breeze Platform" instead of "Zephyr Project")
3. Are you wanting to include Zephyr documentation as a distinct subset?
4. Are you wanting documentation from each discovered module to be a distinct subset in the documentation tree?
5. Are you wanting this to be available and maintained in an example template/boilerplate?

gregshue, 2021-02-03:
@carlescufi Does this issue touch on features similar to those needed for Zephyr to integrate stand-alone `west` documentation?

manoj153, 2021-02-04:
Hi @gregshue , my goal is always to document the **user introduced code**. Zephyr docs are available perfectly on their website and so accessible by anyone. Since user-introduced code is also written in C and similar, I think re-using the doc framework makes sense. What do you think?

Specifically doc on own project-specific tree. I hope I conveyed the message right and clear.
This can be useful when releasing the project-specific code as a library with an API doc.

gregshue, 2021-02-04:
@manoj153 After playing around with the documentation in my `oot_example_module` and in a couple private manifest+module repos, I am considering the partitioning model below. The focus being what is needed by new owners/contributors for (re)verification, integration, and support.

- Manifest repo documentation contains at least the following:
  - The manifest file specifies the source in the "workspace" (integrated collection of modules & applications).
  - The "workspace" is what is versioned and released, no matter how many projects are built out of it.
  - The versioned workspace needs release notes.
  - It is useful to have comprehensive documentation for _some(?)_ content in the workspace, e.g.:
    - Kconfig
    - APIs
    - boards
    - drivers
    - subsystems
    - projects
    - samples
    - A listing of all modules included and at which SHAs (or release tags)
    - A collection of module-specific documentation
- Module repo documentation contains at least the following:
  - overview
  - maintenance organization
  - drivers
  - subsystems
  - boards
  - guides for in-module subsystems/stacks, samples

Getting there will take some restructuring of the `zephyr` repository documentation and identifying what module documentation looks like and how it gets aggregated into manifest-repo generated documentation.

Thoughts?

manoj153, 2021-02-04:
My request was to support out-of-tree doc,
My application code does not influence zephyr tree (they can be separate)

I start my project with a new directory and git tree with the following structure.
```
/PROJECTA
    /src
        main.c
        main.h
CMakeLists.txt
```

The idea is to document API in main.h

gregshue, 2021-02-04:
The code tree will fit the Zephyr documentation process more easily if:

- All project-specific logic has been refactored into subsystems, leaving an empty PROJECTA/src/main.c::main().
- Project documentation follows the location and pattern of sample documentation.  (After all, samples and tests are simply special purpose projects!)
- The project repository is recognized as a Zephyr module (i.e. `zephyr/module.yml` exists), where
  - Paths to your project tree(s) are part of the module samples configuration
  - Your CMakeLists.txt adds the subsystem interface `include` folder to the zephyr include path.  :-)

carlescufi, 2021-02-09:
> @carlescufi Does this issue touch on features similar to those needed for Zephyr to integrate stand-alone `west` documentation?

Not really, those are quite different issues. Zephyr doesn't really integrated standalone west doc, but rather contains the whole of the west documentation in its main tree (`doc/guides/west`).

closed 2025-06-19 by kartben