zephyrproject-rtos/zephyr #31912: STM32: Configure MCO pin through device tree pinctrl

erwango, 2021-02-03:
**Is your enhancement proposal related to a problem? Please describe.**
When enabling MCO1 or MCO2 (CONFIG_CLOCK_STM32_MCO1/2_SRC_FOO), a specific pin need to be configured using  pinctrl.
For instance, today, besides the CONFIG setting, the following should be added in a board dts:
1- Provide the pin configuration:
```
+&pinctrl{
+	mco1_pc9: mco1_pc9 {
+		pinmux = <STM32_PINMUX('C', 9, AF0)>;
+		slew-rate = "very-high-speed";
+	};
+};
```
2- Add this pin to a pinctrl list, so this setting applied:
```
&usart6 {
-	pinctrl-0 = <&usart6_tx_pg14 &usart6_rx_pg9>;
+	pinctrl-0 = <&usart6_tx_pg14 &usart6_rx_pg9 &mco_pc9>;
    ...
```

**Describe the solution you'd like**
Providing pins configuration (point 1 above) is easy and could be done in hal_stm32 w/o trouble.
Getting the pin configuration applied (point 2), requires a node defining it in a pinctrl-0 prop and having a driver extracting this info and using it in a pinctrl set call. This part is more tricky: 
- the matching driver for this functionality is `clock_control`, but due to driver init sequence, `clock_control` driver can't use calls to other drivers (indirectly `gpio` in current case, which is initialized after `clock_control`).
- there isn't any other driver that would match this functionality. So we'd need an empty node + driver which purpose would only be to extract pin control configuration and inject in a `pinctrl_set` call. Lots of code for mere operation.
- An alternative (my preference) would be to set up a dummy node for this kind of settings, but that wouldn't be specific to MCO.

GeorgeCGV, 2022-11-22:
> there isn't any other driver that would match this functionality. So we'd need an empty node + driver which purpose would only be to extract pin control configuration and inject in a pinctrl_set call. Lots of code for mere operation.

@erwango considering power management (_PM_) and possible requirements to control the MCO at runtime (_i.e. enabled/disabled_)  it might not be the worst case. Can you elaborate more on the dummy node approach?

What comes to mind as a quick solution is to use the `chosen` with a vendor-specific prefix like `st`.

Something like:
 
```
chosen {
    st,mco = &mco;
};

mco: mco {
    pinctrl-0 = <&mco1_pa8>;
};
```
where `mco1_pa8` is generated in `hal_stm32`.

Then the GPIO driver can get chosen `st,mco` and initialize it if present during initialization. However, I am not sure how PM and other requirements can be handled there without extending the GPIO which is definitely not the right way.

erwango, 2022-11-23:
> Can you elaborate more on the dummy node approach?

Tbh, I'm not sure today of what I meant that time.
Anyway, all things considered, I think a dedicated device driver would make sense, even if not backed by a generic API.

GeorgeCGV, 2022-12-01:
It might not be so straightforward to split that into a standalone driver.

The RM0468 states the following about `MCO<1/2>` and `MCO<1/2>PRE` (_`RCC_CFGR` register_):

> It is highly recommended to configure these bits only after reset, before enabling the external
> oscillators and the PLLs.

It would be great to comply with the recommendation. However, that implies configuration of the MCO upon RCC initialization (_as it is done for F1 and F4_).

Children nodes can be added to the RCC bindings to provide prescaler and clock source for the `MCO<1/2>`.

It doesn't look that there is any implication of pin configuration being done after MCO configuration. Therefore, providing pin via peripheral `pinctrl` works. 

The drawback is a bit complicated control of the MCO when it has to be controlled without the periphery. Additional custom application specific node has to be creates or something like that to define and control `pinctrl` states.

erwango, 2023-09-01:
See discussion here for an implementation proposal: https://github.com/zephyrproject-rtos/zephyr/discussions/58443#discussioncomment-6863657

erwango, 2024-08-01:
See https://github.com/zephyrproject-rtos/zephyr/pull/76556/ as a the latest implementation attempt

erwango, 2024-09-18:
Fixed by https://github.com/zephyrproject-rtos/zephyr/pull/76556

closed 2024-09-18 by erwango