zephyrproject-rtos/zephyr #31922: hci_usb: HCI ACL packet with size divisible by 64 not sent

mkarhumaa, 2021-02-03:
**Describe the bug**
I am using Zephyr's hci_usb application as Bluetooth LE dongle. I noticed that when trying to send HCI ACL packet which size is divisible by 64 (`size % 64 == 0`), packet is not actually never passed from USB stack to Bluetooth device class driver and therefore the packet won't be send over the air.

In `hci_usb` application Zephyr does USB read transfer to BULK endpoint to read HCI ACL data from host. HCI ACL packets may be fragmented on USB if they are bigger than BULK endpoint's maximum packet size (in this case 64).    `usb_transfer_work()` expects that if packet size is exactly same as maximum packet size, there should be more data coming. In practise transfer is not marked as completed if the packet size is 64 but instead more data or ZLP is expected. Linux's btusb does not use ZLP for indicating all the data has been sent.

```c
		/* ZLP, short-pkt or buffer full */
		if (!bytes || (bytes % usb_dc_ep_mps(ep)) || !trans->bsize) {
			/* transfer complete */
			trans->status = 0;
			goto done;
		}

```

According to [USB in a NutShell](https://www.beyondlogic.org/usbnutshell/usb4.shtml#Bulk), "a bulk transfer is considered complete when it has transferred the *exact amount of data requested*, transferred a packet less than the maximum endpoint size, or transferred a zero-length packet."

To my understanding this should be fixed by checking from upper layer, in this case from HCI ACL layer, if all the data was already received. To make this generic we should probably add some callback to device class driver where upper layer may judge whether enough data was received for example by checking that ACL Data total length matches to the data already received.

Adding this hack before the snippet shown above makes it work with hci_usb. Obviously this doesn't work for anything else and is not acceptable solution for upstream but it nicely shows the w
```c
                /* Check from upper layer if enough data was already
		 * transferred. This is specific to ACL data endpoint.*/
		u16_t acl_len = sys_get_le16(trans->buffer - trans->tsize + 2);
		LOG_DBG("Buffer: %p, Tsize: %d, ACL len: %d",
			trans->buffer, trans->tsize, acl_len);
		/* ACL header size 4 bytes must be added to acl_len. */
		if (acl_len + 4 == trans->tsize) {
			trans->status = 0;
			goto done;
		}

```

As I mentioned before, some kind of callback could be used for checking from device class driver if enough data has been read. I'm afraid this would mean API breaking changes to `usb_transfer.c`. I'm also unsure if this problem is specific to BULK endpoint

**To Reproduce**
Steps to reproduce the behavior using nRF52840 DK and nRF Connect:
1. cd samples/bluetooth/hci_usb
2. Add `CONFIG_BT_CTLR_TX_BUFFER_SIZE=251` to prj.conf
2. `west build -b nrf52840dk_nrf52840 && west flash`
3. On nRF Connect app make the phone advertising and add custom GATT service with one writable characteristic
4. Start btmon
5. On `bluetoothctl`:
    *  `scan on` and wait until you see the phone on the list
    *  `scan off` and then `connect <address-of-the-phone>`
    * Once connected BlueZ does GATT service discovery. Move to GATT menu by `menu gatt`
    * Select the writable characteristic for example: `select-attribute /org/bluez/hci1/dev_43_4F_C9_1C_4C_2D/service0028`
    * Try to do GATT Write to characteristic with exactly 53 bytes long value: `write "0x00 0xaa 0xaa 0xaa 0xaa 0xaa 0xaa 0xaa 0xaa 0xaa 0xaa 0xaa 0xaa 0xaa 0xaa 0x0f 0x10 0xaa 0xaa 0xaa 0xaa 0xaa 0xaa 0xaa 0xaa 0xaa 0xaa 0xaa 0xaa 0xaa 0xaa 0x1f 0x20 0xaa 0xaa 0xaa 0xaa 0xaa 0xaa 0xaa 0xaa 0xaa 0xaa 0xaa 0xaa 0xaa 0xaa 0x2f 0x30 0xaa 0xaa 0xaa 0xaa"`
6. You should see from btmon trace that `Number of Completed Packets` event is missing after the write. Also sniffer can be used for verifying that packet is not sent over the air.

**Expected behavior**
USB stack is expected to pass also the 64 bytes long HCI ACL packet to controller.

**Impact**
Causes packet loss. In addition host and controller go out of sync because lost packet causes missing `Number of Completed Packets` event. In practise this usually leads to disconnection.

**Logs and console output**
Full btmon trace: [zephyr-usb-transfer-not-completed-btmon.log](https://github.com/zephyrproject-rtos/zephyr/files/5918561/zephyr-usb-transfer-not-completed-btmon.log)

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04, kernel 5.8.0-41-generic.
 - Toolchain: Zephyr SDK
 - Affects 2.2.1, 2.4.0 and [master](https://github.com/zephyrproject-rtos/zephyr/commit/51c34bb60986d5db0286bf59eb10303ab10ef39c)

jfischer-no, 2021-02-03:
This is not a big problem and will not appear as long as BT_CTLR_TX_BUFFER_SIZE is not changed and BT_BUF_ACL_SIZE(32) is smaller than endpoint's MPS - 1 (63).
Error in the code is to start the transfer over BT_BUF_ACL_SIZE instead of USB_MAX_FS_BULK_MPS (`static uint8_t data[USB_MAX_FS_BULK_MPS];`) and not to use USB_TRANS_NO_ZLP flag.

>     2\. Add `CONFIG_BT_CTLR_TX_BUFFER_SIZE=251` to prj.conf

IMHO BT_CTLR_TX_BUFFER_SIZE greater than 59 has no advantages in this application.
BT_CTLR_TX_BUFFER_SIZE = **58** can be used as workaround until this bug is fixed. 58 comes from 63 - 1 - 4 + 4:

```
#define BT_L2CAP_HDR_SIZE                4

#define BT_L2CAP_MTU                     (CONFIG_BT_CTLR_TX_BUFFER_SIZE - BT_L2CAP_HDR_SIZE)

#define BT_BUF_ACL_SIZE                  BT_L2CAP_BUF_SIZE(BT_L2CAP_MTU)

...
#define BT_HCI_ACL_HDR_SIZE              4
...
#define BT_BUF_RESERVE                   MAX(CONFIG_BT_HCI_RESERVE, CONFIG_BT_HCI_RAW_RESERVE)
...

#define BT_L2CAP_BUF_SIZE(mtu)           (BT_BUF_RESERVE + \                                            
                                          BT_HCI_ACL_HDR_SIZE + BT_L2CAP_HDR_SIZE + \                   
                                          (mtu))
...
```

mkarhumaa, 2021-02-04:
@jfischer-no Thanks for taking a look into this!

My understanding is that to be able to use Bluetooth LE feature Data Length Extension which makes it possible to improve throughput by sending Link Layer packets up to 251 byte, `CONFIG_BT_CTLR_TX_BUFFER_SIZE=251` and `CONFIG_BT_DATA_LEN_UPDATE=y` are needed. Data Length Extensions is not enabled by default in `hci_usb` application but I would think it is quite commonly used.

I agree that using smaller value for `CONFIG_BT_CTLR_TX_BUFFER_SIZE` can be used as workaround if Data Length Extension is not needed.

cvinayak, 2021-02-04:
- USB device HCI_RAW to check hci->len  = 60, 124 and 188, and mark the HCI packet as complete for the bt_raw to consume it as a unique HCI packet.
- USB device HCI_RAW_H4 to check hci->len = 59, 123 and 187, and mark the HCI packet as complete for the bt_raw to consume it as a unique HCI packet.

The offset to the hci->len will be 2 bytes for standard HCI_USB and 3 bytes for HCI_USB_H4.

commit 7ab50b66 (2021-02-14): usb: bluetooth: fix OUT buffer handling

acl_read_cb does not handle incoming (ACL) data
if BT_CTLR_TX_BUFFER_SIZE is greater than
USB_MAX_FS_BULK_MPS - BT_HCI_ACL_HDR_SIZE.

Since the host adjusts the data according to
the BT_CTLR_TX_BUFFER_SIZE and does not use ZLP
we cannot start usb_transfer over the possible length of
the whole packet, with or without USB_TRANS_NO_ZLP flag.
But we can read the packet length from the header and
call net_buf_put() when the whole packet is received.

Fixes: #31922

Reported-by: Matias Karhumaa <matias.karhumaa@gmail.com>
commit af67c16b (2021-02-17): usb: bluetooth: fix OUT buffer handling

acl_read_cb does not handle incoming (ACL) data
if BT_CTLR_TX_BUFFER_SIZE is greater than
USB_MAX_FS_BULK_MPS - BT_HCI_ACL_HDR_SIZE.

Since the host adjusts the data according to
the BT_CTLR_TX_BUFFER_SIZE and does not use ZLP
we cannot start usb_transfer over the possible length of
the whole packet, with or without USB_TRANS_NO_ZLP flag.
But we can read the packet length from the header and
call net_buf_put() when the whole packet is received.

Fixes: #31922

Reported-by: Matias Karhumaa <matias.karhumaa@gmail.com>
closed 2021-02-14 by nashif