zephyrproject-rtos/zephyr #31924: IVSHMEM with ACRN not working

hidrol, 2021-02-03:
We are trying to use the IVSHMEM Device with following setting:

**Hardware:**
EHL A0 Board

**Software:**
Logical Partition Mode with Yocto

**VM0:**
Linux RT

**VM1:**
Zephyr

The xml scenario file regarding ivshmem looks like this:

```
<IVSHMEM desc="IVSHMEM configuration">
	    <IVSHMEM_ENABLED desc="Enable Share Memory between VMs by IVSHMEM.">y</IVSHMEM_ENABLED>
            <IVSHMEM_REGION desc="the name, size, colon separated IDs of communication VMs of share memory region, separated by comma like this: hv:/shm_region_0, 2, 0:2">hv:/shm_prl1_prl2, 2, 0:1</IVSHMEM_REGION>
            <IVSHMEM_REGION desc="the name, size, colon separated IDs of communication VMs of share memory region, separated by comma like this: hv:/shm_region_0, 2, 0:2">hv:/shm_prl2_prl1, 2, 0:1</IVSHMEM_REGION>
	    </IVSHMEM>
```

**Zephyr prj.conf file:**

```
CONFIG_PCIE=y
CONFIG_PCIE_MSI=y
CONFIG_PCIE_MSI_X=y
CONFIG_PCIE_MSI_MULTI_VECTOR=y
CONFIG_VIRTUALIZATION=y
CONFIG_IVSHMEM=y
```

**Zephyr Kconf in drivers/virtualization:**

```
# Virtualization drivers configuration options

# Copyright (c) 2015-2020 Intel Corporation
# SPDX-License-Identifier: Apache-2.0

menuconfig VIRTUALIZATION
	bool "Virtualization guests drivers"
	help
	  This contains various drivers meant to support and expose features
	  when Zephyr is running as a guest in a virtualized or emulated
	  environment.

if VIRTUALIZATION

config IVSHMEM
	bool "Inter-VM shared memory device (ivshmem)"
	depends on PCIE
	help
	  This will enable support of qemu's ivshmem device, which is also
	  present in ACRN hypervizor, and lets VM sharing memory with each
	  other.

if IVSHMEM

config IVSHMEM_DEV_NAME
       string
       default "IVSHMEM"

module = IVSHMEM
module-str = ivshmem
source "subsys/logging/Kconfig.template.log_config"

config IVSHMEM_DOORBELL
	bool "Support interrupt based ivshmem (doorbell version)"
	depends on PCIE_MSI_X && PCIE_MSI_MULTI_VECTOR
	help
	  This will enable support of ivshmem-doorbell, i.e. the interrupt
	  based ivshmem.

config IVSHMEM_MSI_X_VECTORS
	int "How many notification vectors should be pre-allocated?"
	default 2
	depends on IVSHMEM_DOORBELL
	help
	  MSI-X vector holders must be pre-allocated. One can pre-allocate
	  more or less than necessary. Depends on how many VMs will connect
	  with each other. These are know to be the notification vectors in
	  ivshmem.

config IVSHMEM_INT_PRIORITY
	int "Interrupt priority"
	default 2
	depends on IVSHMEM_DOORBELL
	help
	  Interrupt priority used for the MSI-X generated interrupts.

config IVSHMEM_SHELL
	bool "Enable IVshmem shell module"
	depends on SHELL
	help
	  This is mostly a module to help getting info the ivshmem and/or
	  quickly testing your ivshmem client application (from another VM).

endif # IVSHMEM

endif # VIRTUALIZATION
```

**Zephyr user application c code:**

```
#include <zephyr.h>
#include <drivers/virtualization/ivshmem.h>
#include <sys/printk.h>
#include <string.h>
#include <device.h>

int main()
{
	printk("IVSHMEM_TEST\n");
	struct device *ivshmem;
	printk("1\n");
	uintptr_t mem;
	printk("2\n");
	size_t size;
	printk("3\n");
	char message[] = "Hello from Zephyr!";	
	printk("4\n");
	ivshmem = device_get_binding(CONFIG_IVSHMEM_DEV_NAME);
	printk("5\n");
	size = ivshmem_get_mem(ivshmem, &mem);
	printk("6\n");
	if (size < 0){ 
		printk("Could not get Shared Memory Adress %s\n", CONFIG_BOARD);
		return -1;
	}
	else{
		strcpy((char*)mem, message);
		printk("Memory written, Message: %s \n", message);
	}

	return 0;
}
```

When we run this application on qemu it works without problems, on acrn instead it gets stuck in 
`size = ivshmem_get_mem(ivshmem, &mem);`

We are not quite sure how we must edit the xml or Kconfig files in ACRN or Zephyr.

hidrol, 2021-02-04:
With the following code we checked if the device is being found. Turns out Zephyr can not find the ivshmem device.

if (!ivshmem) {
                printk("IVshmem device cannot be found");
                return -1;
}

tbursztyka, 2021-02-05:
missed that issue. But #32020 is fixing it.

nashif, 2021-02-05:
fixed by  #32020

closed 2021-02-05 by nashif