zephyrproject-rtos/zephyr #31937: sample.bluetooth.peripheral_hr_rv32m1_vega_ri5cy does not build

pabigot, 2021-02-03:
Showed up in https://buildkite.com/zephyr/zephyr-daily/builds/229

Local build shows additional failures:
```
[97/204] Building C object zephyr/subsys/bluetooth/controller/CMakeFiles/subsys__bluetooth__controller.dir/util/memq.c.obj
/mnt/nordic/zp/zephyr/subsys/bluetooth/controller/util/memq.c: In function 'memq_enqueue':
/mnt/nordic/zp/zephyr/subsys/bluetooth/controller/util/memq.c:103:2: warning: implicit declaration of function 'cpu_dmb' [-Wimplicit-function-declaration]
  103 |  cpu_dmb(); /* Ensure data accesses are synchronized */
      |  ^~~~~~~
[121/204] Building C object zephyr/subsys/bluetooth/controller/CMakeFiles/subsys__bluetooth__controller.dir/ll_sw/openisa/hal/RV32M1/radio/radio.c.obj
FAILED: zephyr/subsys/bluetooth/controller/CMakeFiles/subsys__bluetooth__controller.dir/ll_sw/openisa/hal/RV32M1/radio/radio.c.obj 
/usr/local/zephyr-sdk/riscv64-zephyr-elf/bin/riscv64-zephyr-elf-gcc -DBUILD_VERSION=v2.5.0-rc2-5-g5b8d79fe9a16 -DCPU_RV32M1_ri5cy -DKERNEL -DXCVR_GENFSK_ENABLED -D_FORTIFY_SOURCE=2 -D__ZEPHYR__=1 -I../../../../subsys/bluetooth/controller/ll_sw/openisa/lll -I../../../../subsys/bluetooth/controller/ll_sw/openisa -I../../../../subsys/bluetooth/controller/hci/openisa -I../../../../subsys/bluetooth/controller/. -I../../../../subsys/bluetooth/controller/include -I../../../../subsys/bluetooth/controller/ll_sw -I../../../../include -Izephyr/include/generated -I../../../../soc/riscv/openisa_rv32m1 -I../../../../subsys/bluetooth -I/mnt/nordic/zp/modules/hal/openisa/vega_sdk_riscv/RISCV -I/mnt/nordic/zp/modules/hal/openisa/vega_sdk_riscv/devices/RV32M1 -I/mnt/nordic/zp/modules/hal/openisa/vega_sdk_riscv/devices/RV32M1/drivers/. -I/mnt/nordic/zp/modules/hal/openisa/vega_sdk_riscv/middleware/wireless/framework/XCVR/RV32M1 -I/mnt/nordic/zp/modules/crypto/tinycrypt/lib/include -isystem ../../../../lib/libc/minimal/include -isystem /usr/local/zephyr-sdk-0.12.2/riscv64-zephyr-elf/bin/../lib/gcc/riscv64-zephyr-elf/10.2.0/include -isystem /usr/local/zephyr-sdk-0.12.2/riscv64-zephyr-elf/bin/../lib/gcc/riscv64-zephyr-elf/10.2.0/include-fixed -Os -imacros /mnt/nordic/zp/zephyr/samples/bluetooth/peripheral_hr/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -mabi=ilp32 -march=rv32ima -imacros /mnt/nordic/zp/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wno-address-of-packed-member -Wno-unused-but-set-variable -Werror=implicit-int -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=/mnt/nordic/zp/zephyr/samples/bluetooth/peripheral_hr=CMAKE_SOURCE_DIR -fmacro-prefix-map=/mnt/nordic/zp/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/mnt/nordic/zp=WEST_TOPDIR -ffunction-sections -fdata-sections -march=rv32imc -std=c99 -nostdinc -O2 -MD -MT zephyr/subsys/bluetooth/controller/CMakeFiles/subsys__bluetooth__controller.dir/ll_sw/openisa/hal/RV32M1/radio/radio.c.obj -MF zephyr/subsys/bluetooth/controller/CMakeFiles/subsys__bluetooth__controller.dir/ll_sw/openisa/hal/RV32M1/radio/radio.c.obj.d -o zephyr/subsys/bluetooth/controller/CMakeFiles/subsys__bluetooth__controller.dir/ll_sw/openisa/hal/RV32M1/radio/radio.c.obj   -c /mnt/nordic/zp/zephyr/subsys/bluetooth/controller/ll_sw/openisa/hal/RV32M1/radio/radio.c
/mnt/nordic/zp/zephyr/subsys/bluetooth/controller/ll_sw/openisa/hal/RV32M1/radio/radio.c: In function 'radio_sleep':
/mnt/nordic/zp/zephyr/subsys/bluetooth/controller/ll_sw/openisa/hal/RV32M1/radio/radio.c:1535:2: error: #error "Missing atomic operation in radio_sleep()"
 1535 | #error "Missing atomic operation in radio_sleep()"
      |  ^~~~~
/mnt/nordic/zp/zephyr/subsys/bluetooth/controller/ll_sw/openisa/hal/RV32M1/radio/radio.c: In function 'radio_wake':
/mnt/nordic/zp/zephyr/subsys/bluetooth/controller/ll_sw/openisa/hal/RV32M1/radio/radio.c:1584:2: error: #error "Missing atomic operation in radio_wake()"
 1584 | #error "Missing atomic operation in radio_wake()"
      |  ^~~~~
[135/204] Building C object zephyr/subsys/bluetooth/controller/CMakeFiles/subsys__bluetooth__controller.dir/ll_sw/openisa/lll/lll_conn.c.obj
In file included from /mnt/nordic/zp/zephyr/subsys/bluetooth/controller/ll_sw/openisa/lll/lll_conn.c:16:
../../../../subsys/bluetooth/controller/./util/mfifo.h: In function 'mfifo_by_idx_enqueue':
../../../../subsys/bluetooth/controller/./util/mfifo.h:126:2: warning: implicit declaration of function 'cpu_dmb' [-Wimplicit-function-declaration]
  126 |  cpu_dmb(); /* Ensure data accesses are synchronized */
      |  ^~~~~~~
[141/204] Building C object zephyr/subsys/bluetooth/controller/CMakeFiles/subsys__bluetooth__controller.dir/ll_sw/openisa/lll/lll_adv.c.obj
In file included from /mnt/nordic/zp/zephyr/subsys/bluetooth/controller/ll_sw/openisa/lll/lll_adv.c:23:
../../../../subsys/bluetooth/controller/./util/mfifo.h: In function 'mfifo_by_idx_enqueue':
../../../../subsys/bluetooth/controller/./util/mfifo.h:126:2: warning: implicit declaration of function 'cpu_dmb' [-Wimplicit-function-declaration]
  126 |  cpu_dmb(); /* Ensure data accesses are synchronized */
      |  ^~~~~~~
[144/204] Building C object zephyr/subsys/bluetooth/controller/CMakeFiles/subsys__bluetooth__controller.dir/ll_sw/ull.c.obj
In file included from /mnt/nordic/zp/zephyr/subsys/bluetooth/controller/ll_sw/ull.c:24:
../../../../subsys/bluetooth/controller/./util/mfifo.h: In function 'mfifo_by_idx_enqueue':
../../../../subsys/bluetooth/controller/./util/mfifo.h:126:2: warning: implicit declaration of function 'cpu_dmb' [-Wimplicit-function-declaration]
  126 |  cpu_dmb(); /* Ensure data accesses are synchronized */
      |  ^~~~~~~
[153/204] Building C object zephyr/subsys/bluetooth/controller/CMakeFiles/subsys__bluetooth__controller.dir/ll_sw/ull_conn.c.obj
In file included from /mnt/nordic/zp/zephyr/subsys/bluetooth/controller/ll_sw/ull_conn.c:22:
../../../../subsys/bluetooth/controller/./util/mfifo.h: In function 'mfifo_by_idx_enqueue':
../../../../subsys/bluetooth/controller/./util/mfifo.h:126:2: warning: implicit declaration of function 'cpu_dmb' [-Wimplicit-function-declaration]
  126 |  cpu_dmb(); /* Ensure data accesses are synchronized */
      |  ^~~~~~~
[154/204] Building C object zephyr/subsys/bluetooth/host/CMakeFiles/subsys__bluetooth__host.dir/hci_core.c.obj
ninja: build stopped: subcommand failed.
```

carlescufi, 2021-02-03:
@cvinayak I took a quick look at this but I am not sure how you want to fix it.

George-Stefan, 2021-02-03:
We can define cpu_dmb() in subsys/bluetooth/controller/ll_sw/openisa/hal/cpu.h. That's easy.

I don't see how this condition (https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/bluetooth/controller/ll_sw/openisa/hal/RV32M1/radio/radio.c#L1526)
#if (CONFIG_BT_CTLR_LLL_PRIO == CONFIG_BT_CTLR_ULL_HIGH_PRIO) && \
	(CONFIG_BT_CTLR_LLL_PRIO == CONFIG_BT_CTLR_ULL_LOW_PRIO)

fails since it's not related to cpu_dmb()

pabigot, 2021-02-03:
@George-Stefan I believe it fails because those Kconfig properties are only provided with default values for Nordic platforms.

cvinayak, 2021-02-03:
@pabigot @George-Stefan I am looking into this right now. If I dont post a PR, I will leave a comment here.

George-Stefan, 2021-02-03:
> @George-Stefan I believe it fails because the `BT_CTLR_ULL_{HIGH,LOW}_PRIO` properties are only provided with default values for Nordic platforms.

It must be recent change as well of the config/project files?

PR #31941 (2021-02-03): Bluetooth: controller: openisa: Fix build failures
Fix OpenISA port to include compiler memory barrier as in
commit 1a14f8b ("Bluetooth: controller: Use DMB instead
of DSB").

Relates to commit 1af2b91 ("Bluetooth: controller: Fix
Tx Buffer Overflow") and to commit ef2ece8 ("Bluetooth:
controller: openisa: Fix sanitycheck failures") and to
commit c2fc629 ("Bluetooth: controller: 255 byte AD
payload support").

Fix BT_CTLR_ULL_HIGH_PRIO default value to equal
BT_CTLR_LLL_PRIO value if BT_CTLR_LOW_LAT is enabled, else
default ULL_HIGH to lower IRQ priority level in comparison
to LLL IRQ priority [... cut]
commit 59616c0b (2021-02-03): Bluetooth: controller: openisa: Fix unsupported ULL LLL Prio config

Fix for OpenISA rv32m1_vega_ri5cy board not supporting
different IRQ Priority levels for LLL, ULL_HIGH and ULL_LOW
execution contexts.

Fixes #31937.
commit b7acb6bf (2021-02-03): Bluetooth: controller: openisa: Fix to use compiler memory barrier

Fix OpenISA port to include compiler memory barrier as in
commit 1a14f8b3a690 ("Bluetooth: controller: Use DMB instead
of DSB").

Relates to commit 1af2b91c2362 ("Bluetooth: controller: Fix
Tx Buffer Overflow") and to commit ef2ece82c054 ("Bluetooth:
controller: openisa: Fix sanitycheck failures") and to
commit c2fc629dd21b ("Bluetooth: controller: 255 byte AD
payload support").

Fixes #31937.
closed 2021-02-03 by nashif