zephyrproject-rtos/zephyr #31948: tests: drivers: spi: spi_loopback: became skipped whereas it used to be passed

ABOSTM, 2021-02-03:
**Describe the bug**
tests/drivers/spi/spi_loopback used to be passed on some STM32 boards.
It requires some harness/fixture which where implemented on automatic bench (map.yaml).

Some time ago, this test became systematically skipped for all boards
When removing fixture requirement in the test, it is passed.
So it looks like fixture mechanism has changed.
Is it bugging or something to change in harness/fixture declaration ?

Note: it seems this test is also skipped for other manufacturer (but I don't know whether it uses to be passed):
https://testing.zephyrproject.org/daily_tests/v2.5.0-rc1-190-g0ab1372b7c87/report/index.html

**To Reproduce**
Steps to reproduce the behavior:
0. be sure that map.yaml contains fixture spi_llopback for the board used
1. sanitycheck -N --device-testing --hardware-map ../map.yaml -c -p nucleo_f429zi -T tests/drivers/spi/spi_loopback/
2. see twister logs

**Expected behavior**
Test Passed

**Logs and console output**
```
$ cat twister-out/twister_discard.csv 
test,arch,platform,reason
tests/drivers/spi/spi_loopback/drivers.spi.loopback,arm,nucleo_f429zi,Not runnable on device
```

**Environment (please complete the following information):**
 - OS:  Linux
 - Toolchain Zephyr SDK
 - Commit SHA: 0ab1372b7c87

galak, 2021-02-04:
@ABOSTM what does the contents of your ../map.yaml look like?

galak, 2021-02-04:
If I do `scripts/twister --device-testing  -c -p nucleo_f429zi -T tests/drivers/spi/spi_loopback/ -X spi_loopback -v -v`

That seems to build and try and run `drivers.spi.loopback`.  So maybe something w/regards to the map.yaml and specifying the FIXTURE.

ABOSTM, 2021-02-04:
@galak 
here is an extract of my map.yaml
```
- available: true
  connected: true
  fixtures:
  - gpio_loopback
  - spi_loopback
  id: 067AFF323535474B43094111
  platform: nucleo_f429zi
  product: STM32 STLink
  runner: openocd
  serial: /dev/ttyACM12
```
I guess it is good as it used to work some time ago.

I agree with you, when adding fixture directly on the command line, it is working fine
`scripts/twister --device-testing -c -p nucleo_f429zi -T tests/drivers/spi/spi_loopback/ -X spi_loopback `
but not with map.yaml

PR #31992 (2021-02-05): twister: fix possible race and  hardware map parsing fix
Fix possible race when determining available devices. Find available
device and return it in the same operation to avoid the race.

Also fix issue where we did not parse fixtures from hardware map.

Fixes #31769
Fixes #31948
commit ce954d47 (2021-02-05): twister: fix HW map fixtures parsing

We were not parsing fixtures from hardware map, so some tests are being
skipped.

Fixes #31948
closed 2021-02-05 by nashif