zephyrproject-rtos/zephyr #31952: Linking fails with latest master on ARM64 platform

igallib86, 2021-02-03:
Working with ARM64 platform, after rebase over latest mainline, I get the following linker error:

zephyr/kernel/libkernel.a(mmu.c.obj): in function `z_mem_manage_init':
~/zephyrproject/zephyr/kernel/mmu.c:527:(.text.z_mem_manage_init+0x9c): relocation truncated to fit: R_AARCH64_ADR_PREL_PG_HI21 against symbol `z_mapped_size' defined in *ABS* section in zephyr/zephyr_prebuilt.elf
collect2: error: ld returned 1 exit status

Basing on the following commits:
1454730faa (mainline/master) Bluetooth: samples: hci_uart: Add overlays for nRF5340

galak, 2021-02-03:
What are you building that you see this error with?

galak, 2021-02-03:
What platform, test/application, and which toolchain?

carlocaione, 2021-02-04:
The problem arises when for example compiling with an SRAM base address very high in memory, i.e. `CONFIG_SRAM_BASE_ADDRESS=0x2000880000`. One possible solution here is to compile with `-fPIC`. I'll prepare a PR for that.

nashif, 2021-02-09:
this should have been fixed already, please confirm.

igallib86, 2021-02-09:
Hi @nashif , it works, thanks.

closed 2021-02-10 by carlocaione

TheClogger666, 2023-11-06:
Okay compiling with "-fpic" doesn't fix shit you fucking retards. Find a fucking solution that works.

kartben, 2023-11-06:
Comment added November 6, 2023 removed due to absolute violation of Zephyr's Code of Conduct // https://github.com/zephyrproject-rtos/zephyr/blob/main/CODE_OF_CONDUCT.md. User may apply for reinstatement by emailing [conduct@zephyrproject.org](mailto:conduct@zephyrproject.org), for which the Code of Conduct committee will discuss with the user and make a determination regarding the reinstatement