zephyrproject-rtos/zephyr #31959: BLE firmware update STM32WB stuck in loop waiting for CPU2

rickconrey, 2021-02-03:
When using the Zephyr master branch, I am unable to update the firmware over BLE and McuMgr-Android. Zephyr seems to be getting stuck in a while loop in flash_stm32wbx.c at:
while (LL_FLASH_IsActiveFlag_OperationSuspended());  line 93

I am running on custom hardware and have been upgrading the RTOS from Zephyr 2.2.1 to Zephyr Master when I ran into this issue.  This previously worked on Zephyr 2.2.1. I haven't noticed any other issues with the upgrade.

I unfortunately don't remember which version of mcuboot we are using, but I'd guess 1.3 or 1.4. Is there a way to check? Does this matter? This is what it looks like when it boots:
** Booting Zephyr OS build zephyr-v2.1.0-1271-gad5c07a7d0ee  ***
[00:00:00.005,000] <inf> mcuboot: Starting bootloader
[00:00:00.013,000] <inf> mcuboot: Primary image: magic=good, swap_type=0x4, copy_done=0x1, image_ok=0x1
[00:00:00.023,000] <inf> mcuboot: Scratch: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
[00:00:00.032,000] <inf> mcuboot: Boot source: none
[00:00:00.038,000] <inf> mcuboot: Swap type: none
[00:00:00.666,000] <inf> mcuboot: Bootloader chainload address offset: 0xc000
[00:00:00.673,000] <inf> mcuboot: Jumping to the first image slot

@erwango suggested on slack to try reverting flash_stm32wbx.c to before [https://github.com/zephyrproject-rtos/zephyr/pull/26929](https://github.com/zephyrproject-rtos/zephyr/pull/26929)
I only reverted the file, flash_stm32wbx.c, not the whole commit. It appears to be working closer to how it used to on Zephyr 2.2.1. It takes two attempts to upgrade the firmware. The first attempt seems to erase the firmware in slot1 and then disconnects BLE, then once I connect again, the second attempt completes successfully.

I tried upgrading the BLE stack from ST. I was previously running 1.5.0 and am now running 1.10.0. The same issue happens on both. It didn't seem to change anything.

Please let me know if you need more information or would like me to try anything else.

ABOSTM, 2021-02-04:
Hi @rickconrey ,
Are you able, with debugger for example (or printf), to read following registers when you get stuck at line 93
FLASH -> ACR
FLASH -> C2ACR

erwango, 2021-02-04:
> Hi @rickconrey ,
> Are you able, with debugger for example (or printf), to read following registers when you get stuck at line 93
> FLASH -> ACR
> FLASH -> C2ACR

@rickconrey point is to know C2 state when running MCUboot and if the related flash protection procedure should apply.
Typically, if `CONFIG_BT=n` (which should be the case in MCUBoot) these protections may not apply. Implementation in #26929, took that into account, but this was made under some assumptions which might have to be reviewed.

So besides the state of the registers, it would help to have more details on the procedure steps vs CPU2 state.

ABOSTM, 2021-02-04:
@rickconrey, @erwango and myself have different hypothesis about your issue.
So first of all, we need to determine whether your problem occurs 
* during new firmware download (which means while regular zephyr firmware is running)

or
* during firmware swap (which means while mcuboot is running)

Then we are still interested in FLASH -> ACR and FLASH -> C2ACR value when issu occurs.

rickconrey, 2021-02-04:
@erwango, @ABOSTM This happens when the regular zephyr firmware is running and I'm trying to send a new firmware over BLE. I don't seem to have any issues with the firmware swap while mcuboot is running.

FLASH->ACR is 0x601
FLASH->C2ACR is 0x8700

So it looks like CPU2 is not clearing the PES bit.

erwango, 2021-02-04:
@rickconrey Thanks for these details. Have you tried to wait for long duration to see if the binary write eventually happens ?

rickconrey, 2021-02-04:
I just updated the message above..

Yes, I have tried waiting and it does not ever happen.

erwango, 2021-02-04:
CPU2 is supposed to hold PES bit while RF is on and reset it once done. Mechanism is rather complex (There is an application note about it), but we should not end up in this situation.

rickconrey, 2021-02-04:
Hmmm, oh. I thought they both had to be cleared to access the flash?

Do you guys need any other information or want me to try anything else?

erwango, 2021-02-04:
> Do you guys need any other information or want me to try anything else?

Well, everything related to CPU2 like FLASH_C2SR, C2CR. IPCC_C2 registers as well.. 
Unfortunately the possibilities for CPU2 debugging are limited.  We'll give more thinking at it tomorrow

rickconrey, 2021-02-05:
FLASH->SR = 0x80000
FLASH->C2SR = 0x80000
FLASH->C2CR = 0
IPCC->C1TOC2SR = 0
IPCC->C2CR = 0x10001
IPCC->C2MR = 0xFFFFFFD4
IPCC->C2SCR = 0
IPCC->C2TOC1SR = 0

For HSEM, the only semaphores that are locked are 2, 4, and 5. All of the others are unlocked.

erwango, 2021-02-05:
@rickconrey Thaks for the inputs. We're investigating possible scenarii that can explain current status.

ABOSTM, 2021-02-05:
Hi @rickconrey,
Thanks for your feedback.
Our BLE expert said there is a case where CPU2 PES may remain stuck to 1, but it is during erase, whereas line 93 where your are stuck correspond to write operation.
Knowing that erase and write have very similar implementation, it is possible that due to code optimization, with debugger you think you are on line 93 but you are in fact at line 222.
So could you provide the call stack, so we can check ?

Also this case described by our expert appends when BLE connect interval is smaller than 25ms . see Application AN5289 (chapter 4.7.3 Conflict between RF activity and Flash memory management).
Could you get this information (BLE connect interval) ? 
zephyr doc may help for that https://docs.zephyrproject.org/latest/guides/bluetooth/bluetooth-dev.html#embedded-hci-tracing
(or any other means you are aware of)

rickconrey, 2021-02-05:
You are correct, it is actually line 222. It happens during the erase, not the write. When I wrote up the description, I looked at the wrong spot in the code since they are almost identical.

Is the connection interval configurable? Is there somewhere I can change it?
Hmm, I'm not seeing anything over btmon after changing those setting as described in the document. I can see that something is being passed over the uart, but nothing shows up in btmon. I'll keep trying to get it to work.

rickconrey, 2021-02-05:
Ok, I'm getting data over btmon now. I had to remove all the printk calls in my code.

Am I looking for the lines that say "connection interval"?

I see some that say 7.5 msec and some that say 45msec

ABOSTM, 2021-02-08:
I @rickconrey ,
Thanks for the btmon.log.
Yes connection interval may change over time.
For sure 7,5ms or 15ms are not good if it set when starting flash erase.
This parameter is given by BLE master (your smartphone), but it is possible for the slave to request a change
 (but master is free to accept reject or propose/force alternative).

It seem the Zephyr API to do so is: bt_conn_le_param_update()
You need to set at least parameters:
	.interval_min
	.interval_max 
I suggest to use BT_GAP_INIT_CONN_INT_MIN  (= 30ms)  and BT_GAP_INIT_CONN_INT_MAX (= 50ms)

Difficulty is to do it at the right place.
For sure after connection is established ... and before flash erase.

It would the be interesting to get btmon.log to see if request has been efficient or not (master may reject it).
Also if it is possible for you to make a correlation between connection interval seen in btmon.log and activities like BT connection, Flash erase.
Finally see whether firmware is still stuck during erase operation.

rickconrey, 2021-02-08:
@ABOSTM 
Using menuconfig, I have the minimum (BT_PERIPHERAL_PREF_MIN_INT) and maximum (BT_PERIPHERAL_PREF_MAX_INT) connection interval already set to 24 * 1.25 and 40 * 1.25, so 30ms and 50ms. I also tried to manually update them using bt_conn_le_param_update() as you suggested. I do this after the device has established a connection in the "connected" callback.

There is no change, it is still behaving as it was.

Is this something that could be configured on the smartphone app side? or is it something that Android itself handles?

rickconrey, 2021-02-08:
I figured out how to update the connection interval on the smartphone app. Before the update it was set to the fastest setting (11.25 to 15ms) now it matches the device at 30ms to 50ms. This definitely seems to improve the success of the upload. I still have some failures, but it seems to have a lot more successes than before.  I can also see that the device requests that the connection interval be set to 30 - 50 ms when looking at btmon.

Thank you for pointing me in the right direction!  I will keep debugging to see if I can get it to work all the time.

ABOSTM, 2021-02-09:
@rickconrey,
Great. As you noticed, those parameter can be set by application (depending on application),
but I have been told that Operating System can also override this. For example on Apple device it is needed to request minimum interval higher or equal to 30ms (25ms will not work properly as apple valid range is [30ms ..])

With btmon, you could check and compare connection interval in case of success  and case of failure.

erwango, 2021-03-31:
@rickconrey Did you make some progress on your issue?

rickconrey, 2021-04-01:
@erwango Yes, I think I have figured this issue out. It appears to be a hardware issue related to an unstable LSE. Until we can tune the LSE, the fix for us currently is to run the BLE from the HSE. I found that Zephyr is not currently setup to support running BLE from the HSE, so maybe this has turned into a feature request instead of a bug report.

Besides not being able to run the BLE clock from HSE with Zephyr, I am also having trouble running it with LSI. I'm not sure that will work with how the clock source is currently chosen in the code?

There are two lines that need to be changed in ipm_stm32wb.c to have the option of running the BLE from HSE:

1. In the start_ble_rf function, set the RF wakeup source to HSE with: LL_RCC_SetRFWKPClockSource(LL_RCC_RFWKP_CLKSOURCE_HSE_DIV1024)
2. Change the value of CFG_BLE_LSE_SOURCE from 0 to 1.

I think doing the change for step 2 will make the LSI work, but I have not verified that.

The other thing I noticed while looking through ipm_stm32wb.c is that the RTC clock source is set while choosing the BLE clock wakeup source. Is this intentional?

erwango, 2021-04-02:
@rickconrey Thanks for this feedback

> Yes, I think I have figured this issue out. It appears to be a hardware issue related to an unstable LSE. Until we can tune the LSE, the fix for us currently is to run the BLE from the HSE.

Unexpected but great. 

> Besides not being able to run the BLE clock from HSE with Zephyr, I am also having trouble running it with LSI. I'm not sure that will work with how the clock source is currently chosen in the code?

```
/**
 *  Source for the 32 kHz slow speed clock
 *  1 : internal RO
 *  0 : external crystal ( no calibration )
 */
#define CFG_BLE_LSE_SOURCE  0
```
Setting CFG_BLE_LSE_SOURCE to 1 will then make it use LSI, as I understand it.
Though, maybe `LL_RCC_SetRFWKPClockSource(LL_RCC_RFWKP_CLKSOURCE_LSI);` would also be required.
To be clear, do you confirm that setting CFG_BLE_LSE_SOURCE from 0 to 1 is required for both LSI and HSE ?

> The other thing I noticed while looking through ipm_stm32wb.c is that the RTC clock source is set while choosing the BLE clock wakeup source. Is this intentional?

Not sure if intentional or a consequence of a broken design for selecting the BLE clock,
but it doesn't seem right indeed.
 
Thanks for these inputs, globally, it seems we need to review the BLE clock selection part:
- Enable selection of LSE/HSE/LSI
- Make it independent from clock selection of other subsystem

rickconrey, 2021-04-02:
>Setting CFG_BLE_LSE_SOURCE to 1 will then make it use LSI, as I understand it.
Though, maybe LL_RCC_SetRFWKPClockSource(LL_RCC_RFWKP_CLKSOURCE_LSI); would also be required.
To be clear, do you confirm that setting CFG_BLE_LSE_SOURCE from 0 to 1 is required for both LSI and HSE ?

I can confirm that setting CFG_BLE_LSE_SOURCE to 1 is necessary for running with HSE, and I would assume that it is also the case with LSI, but I am unable to confirm that.  When I turn off LSE using CONFIG_CLOCK_STM32_LSE, my device is very slow to boot and then resets itself after a fault.  Am I missing something else for running on LSI?

Here is the output that I capture when trying to run off of LSI:
[image]

The console output is a little backwards in the picture. It shows the fault before the slow boot, but that was the easiest way to capture both.  After the boot, the device hangs at the syscmd_status_not: line for roughly 25 seconds.  Then it boots the system and eventually runs into the "assertion fail". The whole process from boot to failure is around a minute.  This happens only when I turn off the LSE and try to run it off of LSI.  Is this just an issue on my side again?

erwango, 2021-04-16:
>  Then it boots the system and eventually runs into the "assertion fail". The whole process from boot to failure is around a minute. This happens only when I turn off the LSE and try to run it off of LSI. Is this just an issue on my side again?

For specific questions on controller behavior with clocks variations, please ask directly to WB support teams ([here](https://github.com/STMicroelectronics/STM32CubeWB) for instance).

For the ability to select clocks in Zephyr, we'll work on enhancing the current code in current release.
I'm moving this from 'bug' to 'enhancement' to reflect new status.

carlescufi, 2022-06-07:
Since clock selection is now available, closing this issue. Please reopen if you think it's not fixed.

erwango, 2022-06-07:
Reopenning as it remains to be implemented specifically for this use case.

closed 2023-01-08 by carlescufi