zephyrproject-rtos/zephyr #31964: up_squared: tests/kernel/timer/timer_api failed.

chen-png, 2021-02-04:
**Describe the bug**
this testcase was failed occasionally.

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b up_squared -p auto tests/kernel/timer/timer_api/ -DCONF_FILE="prj_tickless.conf"
2. west flash ~/up_squared.sh
3. see error

**Logs and console output**
Booting Zephyr OS build v2.5.0-rc2-42-gcce5ff15107b
Running test suite timer_api

START - test_time_conversions
 PASS - test_time_conversions

START - test_timer_duration_period
 PASS - test_timer_duration_period

START - test_timer_restart
 PASS - test_timer_restart

START - test_timer_period_0
 PASS - test_timer_period_0

START - test_timer_expirefn_null
 PASS - test_timer_expirefn_null

**START - test_timer_periodicity**

    **Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/timer/timer_api/src/main.c:354: test_timer_periodicity: (((delta) >= (period_ms)) && ((delta) <= (period_ms) + (1))) || (((10000 % 1000U) != 0) && (delta == period_ms - 1)) is false**

 **FAIL - test_timer_periodicity**

START - test_timer_status_get
 PASS - test_timer_status_get
...

START - test_timeout_abs
 PASS - test_timeout_abs

Test suite timer_api failed.

PROJECT EXECUTION FAILED

**Environment (please complete the following information):**
 - OS: Ubuntu 18.04.4
 - Toolchain: zephyr-sdk-0.12.2
 - Commit ID: cce5ff15107

andyross, 2021-02-04:
I just ran this 12 times without trouble.  How often is an "occasional" failure exactly?

nashif, 2021-02-08:
unable to reproduce, lowering priority

chen-png, 2021-02-09:
this test is passed on up_squared with the latest commit "v2.5.0-rc3", i will track it again tomorrow whether it has been fixed.

andyross, 2021-02-09:
For clarity: there's nothing wrong with reporting a spurious failure.  We've had them before and they're real.  It's just that part of the reporting process for spurious bugs is coming up with a good estimate for the frequency, because otherwise the poor developer has no guidance for how to reproduce.

andyross, 2021-02-12:
The same test was failing reliable on ehl_crb, which is otherwise a very similar configuration.  I strongly suspect that was the same issue (basically the test worked poorly on systems with 10 kHz ticks), even though I still haven't reproduced this on up2.

closed 2021-02-16 by nashif