zephyrproject-rtos/zephyr #31969: test_tcp_fn:net2 mximxrt1060/1064/1050 fails on test_client_invalid_rst with semaphore timed out

hakehuang, 2021-02-04:
**Describe the bug**
when do the test_tcp_fn:net2  tests/net/tcp2, there are issues meet on test_client_invalid_rst 

this issue happens on rt1060/1064/1050 but ok on rt1020.

**To Reproduce**
Steps to reproduce the behavior:
tests/net/tcp2
1. mkdir build; cd build
2. cmake -DBOARD=mximxrt1060_evk .. -- -DCONFIG_NET_TCP_RECV_QUEUE_TIMEOUT=1000
and 
 cmake -DBOARD=mximxrt1060_evk .. -- -DCONFIG_NET_TCP_RECV_QUEUE_TIMEOUT=0
3. make flash --runner=pyocd
4. See error

**Expected behavior**
test pass

**Impact**
net2 stack

**Logs and console output**

with -DCONFIG_NET_TCP_RECV_QUEUE_TIMEOUT=0
```
*** Booting Zephyr OS build v2.5.0-rc2  ***

Running test suite test_tcp_fn

===================================================================

START - test_presetup

 PASS - test_presetup

===================================================================

START - test_client_ipv4

 PASS - test_client_ipv4

===================================================================

START - test_client_ipv6

 PASS - test_client_ipv6

===================================================================

START - test_server_ipv4

 PASS - test_server_ipv4

===================================================================

START - test_server_with_options_ipv4

 PASS - test_server_with_options_ipv4

===================================================================

START - test_server_ipv6

 PASS - test_server_ipv6

===================================================================

START - test_client_syn_resend

 PASS - test_client_syn_resend

===================================================================

START - test_client_fin_wait_2_ipv4

 PASS - test_client_fin_wait_2_ipv4

===================================================================

START - test_client_closing_ipv6

 PASS - test_client_closing_ipv6

===================================================================

START - test_client_invalid_rst

    Assertion failed at WEST_TOPDIR/zephyr/tests/net/tcp2/src/main.c:199: test_sem_take: (false is false)

semaphore timed out (line 1243)

 FAIL - test_client_invalid_rst

===================================================================

START - test_server_recv_out_of_order_data

 PASS - test_server_recv_out_of_order_data

===================================================================

START - test_server_timeout_out_of_order_data

 PASS - test_server_timeout_out_of_order_data

===================================================================

Test suite test_tcp_fn failed.

===================================================================

PROJECT EXECUTION FAILED

```

with -DCONFIG_NET_TCP_RECV_QUEUE_TIMEOUT=1000

```
   *** Booting Zephyr OS build v2.5.0-rc2  ***

    Running test suite test_tcp_fn

    ===================================================================

    START - test_presetup

     PASS - test_presetup

    ===================================================================

    [******** Truncated '7' Lines Log ********]

     PASS - test_server_ipv4

    ===================================================================

    START - test_server_with_options_ipv4

     PASS - test_server_with_options_ipv4

    ===================================================================

    START - test_server_ipv6

     PASS - test_server_ipv6

    ===================================================================

    START - test_client_syn_resend

     PASS - test_client_syn_resend

    ===================================================================

    START - test_client_fin_wait_2_ipv4

     PASS - test_client_fin_wait_2_ipv4

    ===================================================================

    START - test_client_closing_ipv6

     PASS - test_client_closing_ipv6

    ===================================================================

    START - test_client_invalid_rst

    

        Assertion failed at WEST_TOPDIR/zephyr/tests/net/tcp2/src/main.c:199: test_sem_take: (false is false)

    semaphore timed out (line 1243)

     FAIL - test_client_invalid_rst

    ===================================================================

    START - test_server_recv_out_of_order_data

    E: Identical connection handler 0x80003500 already found.

    

        Assertion failed at WEST_TOPDIR/zephyr/tests/net/tcp2/src/main.c:1237: create_server_socket: (false is false)

    Failed to set accept on net_context

     FAIL - test_server_recv_out_of_order_data

    ===================================================================

    START - test_server_timeout_out_of_order_data

    I: ETH_0 enabled 100M full-duplex mode.

    

        Assertion failed at WEST_TOPDIR/zephyr/tests/net/tcp2/src/main.c:1476: test_server_timeout_out_of_order_data: (ret not equal to -EAGAIN)

    semaphore did not time out (0)

     FAIL - test_server_timeout_out_of_order_data

    ===================================================================

    Test suite test_tcp_fn failed.

    ===================================================================

    PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit: v-2.5.0-RC1/RC2 all has such issue

MaureenHelm, 2021-02-04:
git bisect landed on commit 0ed6a3cd6969f3a44a86dbf98f73e509698e065c from @jukkar :
```
commit 0ed6a3cd6969f3a44a86dbf98f73e509698e065c (HEAD, refs/bisect/bad)
Author: Jukka Rissanen <jukka.rissanen@linux.intel.com>
Date:   Thu Nov 19 19:33:20 2020 +0200

    tests: net: tcp2: Add RST packet handling tests
    
    Verify that TCP2 will drop invalid RST packets and accept
    valid ones.
    
    Signed-off-by: Jukka Rissanen <jukka.rissanen@linux.intel.com>
```

It looks like this new RST test never worked on these boards.

hakehuang, 2021-02-05:
> git bisect landed on commit [0ed6a3c](https://github.com/zephyrproject-rtos/zephyr/commit/0ed6a3cd6969f3a44a86dbf98f73e509698e065c) from @jukkar :
> 
> ```
> commit 0ed6a3cd6969f3a44a86dbf98f73e509698e065c (HEAD, refs/bisect/bad)
> Author: Jukka Rissanen <jukka.rissanen@linux.intel.com>
> Date:   Thu Nov 19 19:33:20 2020 +0200
> 
>     tests: net: tcp2: Add RST packet handling tests
>     
>     Verify that TCP2 will drop invalid RST packets and accept
>     valid ones.
>     
>     Signed-off-by: Jukka Rissanen <jukka.rissanen@linux.intel.com>
> ```
> 
> It looks like this new RST test never worked on these boards.

there is another issue found in former release #28602

dleach02, 2021-02-08:
I'm curious about what it is with the board that would keep the test from working? Also, do we know if there is a platform the tests work on for comparison?

MaureenHelm, 2021-02-08:
> Also, do we know if there is a platform the tests work on for comparison?

The test passes on `frdm_k64f`, which uses the same ethernet driver, but different PHY and lower core clock frequency.

pabigot, 2021-02-09:
The tests pass for me on mimxrt1060_evk at v2.5.0-rc3-29-g1b5f134477

hakehuang, 2021-02-10:
> The tests pass for me on mimxrt1060_evk at v2.5.0-rc3-29-g1b5f134477
@pabigot do you know which change related this I can still see failure on v2.5.0-rc3-21-g0b8fd867844c

```
    *** Booting Zephyr OS build v2.5.0-rc3-21-g0b8fd867844c  ***

    Running test suite test_tcp_fn

    ===================================================================

    START - test_presetup

     PASS - test_presetup

    ===================================================================

    START - test_client_ipv4

     PASS - test_client_ipv4

    ===================================================================

    START - test_client_ipv6

     PASS - test_client_ipv6

    ===================================================================

    START - test_server_ipv4

     PASS - test_server_ipv4

    ===================================================================

    START - test_server_with_options_ipv4

     PASS - test_server_with_options_ipv4

    ===================================================================

    START - test_server_ipv6

     PASS - test_server_ipv6

    ===================================================================

    START - test_client_syn_resend

     PASS - test_client_syn_resend

    ===================================================================

    START - test_client_fin_wait_2_ipv4

     PASS - test_client_fin_wait_2_ipv4

    ===================================================================

    START - test_client_closing_ipv6

     PASS - test_client_closing_ipv6

    ===================================================================

    START - test_client_invalid_rst

    

        Assertion failed at WEST_TOPDIR/zephyr/tests/net/tcp2/src/main.c:199: test_sem_take: (false is false)

    semaphore timed out (line 1243)

     FAIL - test_client_invalid_rst

    ===================================================================

    START - test_server_recv_out_of_order_data

     PASS - test_server_recv_out_of_order_data

    ===================================================================

    START - test_server_timeout_out_of_order_data

     PASS - test_server_timeout_out_of_order_data

    ===================================================================

    Test suite test_tcp_fn failed.

    ===================================================================

    PROJECT EXECUTION FAILED
```

pabigot, 2021-02-10:
OK, yes, I *can* confirm the failure when a live ethernet cable is connected, on imxrt1060.  The failure does not occur on k64f.

PR #34580 (2021-04-27): tests: net: tcp2: Perform tests on DUMMY interface
Tests are designed to run over the DUMMY interface but when run on
hardware, net_if_get_default() can return the actual hardware
interface.

Fixes: #31969
closed 2021-04-27 by jukkar