zephyrproject-rtos/zephyr #31982: tests: kernel: queue: regression introduced by FPU sharing PR #31772

ABOSTM, 2021-02-04:
**Describe the bug**
Following tests:
* tests/kernel/queue/kernel.queue
* tests/arch/arm/arm_interrupt/arch.interrupt.arm
* tests/arch/arm/arm_interrupt/arch.interrupt.extra_exception_info
* tests/arch/arm/arm_interrupt/arch.interrupt.no_optimizations

are now failed on stm32f3_disco.
This is a regression introduced by PR #31772, test is passed before this PR and failed after.
Error message is not the same depending on test,
but in one case I got stack overflow. So I tried to enlarge MAIN_STACK_SIZE to 2048. It doesn't help.
I also increase IDLE_STACK_SIZE, ISR_STACK_SIZE and SYSTEM_WORKQUEUE_STACK_SIZE
but test is still failed.
Note: I could not reproduce issue on windows, only reproduced on Linux (different toolchain)

**To Reproduce**
Steps to reproduce the behavior:
1. twister -N --device-testing --hardware-map ../map.yaml -c  -p stm32f3_disco -T tests/kernel/queue/
2. See error

**Expected behavior**
test passed

**Logs and console output**
```
*** Booting Zephyr OS build v2.5.0-rc1-261-gfb2b77c9952f  ***
Running test suite queue_api
===================================================================
START - test_queue_supv_to_user
PASS - test_queue_supv_to_user
===================================================================
START - test_queue_alloc_prepend_user
PASS - test_queue_alloc_prepend_user
===================================================================
START - test_queue_alloc_append_user
PASS - test_queue_alloc_append_user
===================================================================
START - test_auto_free
PASS - test_auto_free
===================================================================
START - test_queue_thread2thread
PASS - test_queue_thread2thread
===================================================================
START - test_queue_thread2isr
PASS - test_queue_thread2isr
===================================================================
START - test_queue_isr2thread
PASS - test_queue_isr2thread
===================================================================
START - test_queue_get_2threads
PASS - test_queue_get_2threads
===================================================================
START - test_queue_get_fail
PASS - test_queue_get_fail
===================================================================
START - test_queue_loop
* Pass data by queue in loop 0
main queue append ---> isr queue find and remove
isr queue get
isr queue append ---> thread queue find and remove
thread queue get
thread queue append ---> main queue find and remove
main queue get
* Pass data by queue in loop 1
main queue append ---> isr queue find and remove
isr queue get
isr queue append ---> thread queue find and remove
thread queue get
thread queue append ---> main queue find and remove
main queue get
* Pass data by queue in loop 2
main queue append ---> isr queue find and remove
isr queue get
isr queue append ---> thread queue find and remove
thread queue get
thread queue append ---> main queue find and remove
main queue get
* Pass data by queue in loop 3
main queue append ---> isr queue find and remove
isr queue get
isr queue append ---> thread queue find and remove
thread queue get
thread queue append ---> main queue find and remove
main queue get
* Pass data by queue in loop 4
main queue append ---> isr queue find and remove
isr queue get
isr queue append ---> thread queue find and remove
thread queue get
thread queue append ---> main queue find and remove
main queue get
* Pass data by queue in loop 5
main queue append ---> isr queue find and remove
isr queue get
isr queue append ---> thread queue find and remove
thread queue get
thread queue append ---> main queue find and remove
main queue get
* Pass data by queue in loop 6
main queue append ---> isr queue find and remove
isr queue get
isr queue append ---> thread queue find and remove
thread queue get
thread queue append ---> main queue find and remove
main queue get
* Pass data by queue in loop 7
main queue append ---> isr queue find and remove
isr queue get
isr queue append ---> thread queue find and remove
thread queue get
thread queue append ---> main queue find and remove
main queue get
* Pass data by queue in loop 8
main queue append ---> isr queue find and remove
isr queue get
isr queue append ---> thread queue find and remove
thread queue get
thread queue append ---> main queue find and remove
main queue get
* Pass data by queue in loop 9
main queue append ---> isr queue find and remove
isr queue get
isr queue append ---> thread queue find and remove
thread queue get
thread queue append ---> main queue find and remove
main queue get
* Pass data by queue in loop 10
main queue append ---> isr queue find and remove
isr queue get
isr queue append ---> thread queue find and remove
thread queue get
thread queueE: ***** MPU FAULT *****
E:   Stacking error (context area might be not valid)
E: r0/a1:  0x00000000  r1/a2:  0x00000000  r2/a3:  0x00000000
E: r3/a4:  0x00000000 r12/ip:  0x00000000 r14/lr:  0x00000000
E:  xpsr:  0x01000000
E: s[ 0]:  0x00000000  s[ 1]:  0x00000000  s[ 2]:  0x00000000  s[ 3]:  0x00000000
E: s[ 4]:  0x00000000  s[ 5]:  0x00000000  s[ 6]:  0x00000000  s[ 7]:  0x00000000
E: s[ 8]:  0x00000000  s[ 9]:  0x00000000  s[10]:  0x00000000  s[11]:  0x00000000
E: s[12]:  0x00000000  s[13]:  0x00000000  s[14]:  0x00000000  s[15]:  0x00000000
E: fpscr:  0x00000000
E: Faulting instruction address (r15/pc): 0x0800d432
E: >>> ZEPHYR FATAL ERROR 2: Stack overflow on CPU 0
E: Current thread: 0x20000718 (unknown)
Caught system error -- reason 2 0
Fatal error was unexpected, aborting...
```

**Environment (please complete the following information):**
 - OS:  Linux
 - Toolchain Zephyr SDK
 - Commit SHA fb2b77c9952f3435b1a264becb32a1043ac2e899

ioannisg, 2021-02-04:
There's probably some other thread that overflows, e.g. the ZTEST thread. I can take a look.

ioannisg, 2021-02-05:
@ABOSTM I've tried the test on other cortex-m platforms (building with FPU_SHARING=y), I don't see stack overflows, so I can't help much with debugging, since i don't have this ST board.
You could find out the thread that overflows - if you build with THREAD_NAME support.
Then, as I mentioned, might be the ZTEST thread that overflows, you can increase
CONFIG_ZTEST_STACKSIZE and CONFIG_TEST_EXTRA_STACKSIZE to see if the overflows disappear.

ABOSTM, 2021-02-05:
@ioannisg,
Ok I will do so. Thanks for the hints.

ioannisg, 2021-02-05:
This is a list of commonly used stacks, and the corresponding Kconfig options: 

CONFIG_ZTEST_STACKSIZE
CONFIG_MAIN_STACK_SIZE
CONFIG_ISR_STACK_SIZE
CONFIG_ZTEST_STACKSIZE
CONFIG_IDLE_STACK_SIZE
CONFIG_SYSTEM_WORKQUEUE_STACK_SIZE
CONFIG_TEST_EXTRA_STACKSIZE
CONFIG_PRIVILEGED_STACK_SIZE

ABOSTM, 2021-02-05:
tests/kernel/queue/kernel.queue is fixed thanks to #32034. Thanks @ioannisg.

As far as tests/arch/arm/arm_interrupt/xxx are concerned,
increasing stack sizes above doesn't help.
See log below.
Please note that oops ans panics a normal (see README.txt): 

```
*** Booting Zephyr OS build v2.5.0-rc2-83-g92c93b1b7fba  ***
Running test suite arm_interrupt
===================================================================
START - test_arm_interrupt
Available IRQ line: 81
E: >>> ZEPHYR FATAL ERROR 1: Unhandled interrupt on CPU 0
E: Current thread: 0x20000188 (unknown)
Caught system error -- reason 1
E: r0/a1:  0x00000003  r1/a2:  0x20003704  r2/a3:  0x00000003
E: r3/a4:  0x20003400 r12/ip:  0xa0000000 r14/lr:  0x08004c0f
E:  xpsr:  0x61000061
E: s[ 0]:  0x00000000  s[ 1]:  0x00000000  s[ 2]:  0x00000000  s[ 3]:  0x00000000
E: s[ 4]:  0x00000000  s[ 5]:  0x00000000  s[ 6]:  0x00000000  s[ 7]:  0x00000000
E: s[ 8]:  0x00000000  s[ 9]:  0x00000000  s[10]:  0x00000000  s[11]:  0x00000000
E: s[12]:  0x00000000  s[13]:  0x00000000  s[14]:  0x00000000  s[15]:  0xffffffff
E: fpscr:  0x6b5f7966
E: Faulting instruction address (r15/pc): 0x08000a6c
E: >>> ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0
E: Fault during interrupt handling
E: Current thread: 0x20000188 (unknown)
Caught system error -- reason 3
E: r0/a1:  0x00000004  r1/a2:  0x20003704  r2/a3:  0x00000004
E: r3/a4:  0x20003400 r12/ip:  0xa0000000 r14/lr:  0x08004c0f
E:  xpsr:  0x61000061
E: s[ 0]:  0x00000000  s[ 1]:  0x00000000  s[ 2]:  0x00000000  s[ 3]:  0x00000000
E: s[ 4]:  0x00000000  s[ 5]:  0x00000000  s[ 6]:  0x00000000  s[ 7]:  0x00000000
E: s[ 8]:  0x00000000  s[ 9]:  0x00000000  s[10]:  0x00000000  s[11]:  0x00000000
E: s[12]:  0x00000000  s[13]:  0x00000000  s[14]:  0x00000000  s[15]:  0xffffffff
E: fpscr:  0x6b5f7966
E: Faulting instruction address (r15/pc): 0x08000a88
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Fault during interrupt handling
E: Current thread: 0x20000188 (unknown)
Caught system error -- reason 4
ASSERTION FAIL [0] @ WEST_TOPDIR/zephyr/tests/arch/arm/arm_interrupt/src/arm_interrupt.c:207
Intentional assert
E: r0/a1:  0x00000004  r1/a2:  0x000000cf  r2/a3:  0x0000000a
E: r3/a4:  0x00000000 r12/ip:  0xa0000000 r14/lr:  0x08016d6d
E:  xpsr:  0x61000061
E: s[ 0]:  0x00000000  s[ 1]:  0x00000000  s[ 2]:  0x00000000  s[ 3]:  0x00000000
E: s[ 4]:  0x00000000  s[ 5]:  0x00000000  s[ 6]:  0x00000000  s[ 7]:  0x00000000
E: s[ 8]:  0x00000000  s[ 9]:  0x00000000  s[10]:  0x00000000  s[11]:  0x00000000
E: s[12]:  0x00000000  s[13]:  0x00000000  s[14]:  0x00000000  s[15]:  0x00000000
E: fpscr:  0x00000000
E: Faulting instruction address (r15/pc): 0x08016d9e
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Fault during interrupt handling
E: Current thread: 0x20000188 (unknown)
Caught system error -- reason 4
  
```

ioannisg, 2021-02-09:
@ABOSTM  could you update the log of the error you get when running arm_interrupt.arm,
 now that #32034 was merged?

ABOSTM, 2021-02-09:
@ioannisg ,
With merge of  #32034 I have 2 slightly different logs depending on test configuration vriant:

1) arch.interrupt.no_optimizations

*** Booting Zephyr OS build v2.5.0-rc3-25-g467f9168606c  ***
Running test suite arm_interrupt
===================================================================
START - test_arm_interrupt
Available IRQ line: 81
E: >>> ZEPHYR FATAL ERROR 1: Unhandled interrupt on CPU 0
E: Current thread: 0x20000188 (unknown)
Caught system error -- reason 1
E: r0/a1:  0x00000003  r1/a2:  0x20002b04  r2/a3:  0x00000003
E: r3/a4:  0x20002800 r12/ip:  0xa0000000 r14/lr:  0x08004c0f
E:  xpsr:  0x61000061
E: s[ 0]:  0x00000000  s[ 1]:  0x00000000  s[ 2]:  0x00000000  s[ 3]:  0x00000000
E: s[ 4]:  0x00000000  s[ 5]:  0x00000000  s[ 6]:  0x00000000  s[ 7]:  0x00000000
E: s[ 8]:  0x00000000  s[ 9]:  0x00000000  s[10]:  0x00000000  s[11]:  0x00000000
E: s[12]:  0x00000000  s[13]:  0x00000000  s[14]:  0x00000000  s[15]:  0xffffffff
E: fpscr:  0x6b5f7966
E: Faulting instruction address (r15/pc): 0x08000a6c
E: >>> ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0
E: Fault during interrupt handling
E: Current thread: 0x20000188 (unknown)
Caught system error -- reason 3
E: r0/a1:  0x00000004  r1/a2:  0x20002b04  r2/a3:  0x00000004
E: r3/a4:  0x20002800 r12/ip:  0xa0000000 r14/lr:  0x08004c0f
E:  xpsr:  0x61000061
E: s[ 0]:  0x00000000  s[ 1]:  0x00000000  s[ 2]:  0x00000000  s[ 3]:  0x00000000
E: s[ 4]:  0x00000000  s[ 5]:  0x00000000  s[ 6]:  0x00000000  s[ 7]:  0x00000000
E: s[ 8]:  0x00000000  s[ 9]:  0x00000000  s[10]:  0x00000000  s[11]:  0x00000000
E: s[12]:  0x00000000  s[13]:  0x00000000  s[14]:  0x00000000  s[15]:  0xffffffff
E: fpscr:  0x6b5f7966
E: Faulting instruction address (r15/pc): 0x08000a88
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Fault during interrupt handling
E: Current thread: 0x20000188 (unknown)
Caught system error -- reason 4
ASSERTION FAIL [0] @ WEST_TOPDIR/zephyr/tests/arch/arm/arm_interrupt/src/arm_interrupt.c:207
Intentional assert
E: r0/a1:  0x00000004  r1/a2:  0x000000cf  r2/a3:  0x0000000a
E: r3/a4:  0x00000000 r12/ip:  0xa0000000 r14/lr:  0x08016d6d
E:  xpsr:  0x61000061
E: s[ 0]:  0x00000000  s[ 1]:  0x00000000  s[ 2]:  0x00000000  s[ 3]:  0x00000000
E: s[ 4]:  0x00000000  s[ 5]:  0x00000000  s[ 6]:  0x00000000  s[ 7]:  0x00000000
E: s[ 8]:  0x00000000  s[ 9]:  0x00000000  s[10]:  0x00000000  s[11]:  0x00000000
E: s[12]:  0x00000000  s[13]:  0x00000000  s[14]:  0x00000000  s[15]:  0x00000000
E: fpscr:  0x00000000
E: Faulting instruction address (r15/pc): 0x08016d9e
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Fault during interrupt handling
E: Current thread: 0x20000188 (unknown)
Caught system error -- reason 4

ABOSTM, 2021-02-09:
2) arch.interrupt.extra_exception_info  and  arch.interrupt.arm 
logs are very similar: 

*** Booting Zephyr OS build v2.5.0-rc3-25-g467f9168606c  ***
Running test suite arm_interrupt
===================================================================
START - test_arm_interrupt
Available IRQ line: 81
E: >>> ZEPHYR FATAL ERROR 1: Unhandled interrupt on CPU 0
E: Current thread: 0x20000188 (unknown)
Caught system error -- reason 1
E: r0/a1:  0x00000003  r1/a2:  0x20002af0  r2/a3:  0x00000003
E: r3/a4:  0x20002800 r12/ip:  0xa0000000 r14/lr:  0x08002e6b
E:  xpsr:  0x61000061
E: s[ 0]:  0x00000000  s[ 1]:  0x00000000  s[ 2]:  0x00000000  s[ 3]:  0x00000000
E: s[ 4]:  0x00000000  s[ 5]:  0x00000000  s[ 6]:  0x00000000  s[ 7]:  0x00000000
E: s[ 8]:  0x00000000  s[ 9]:  0x00000000  s[10]:  0x00000000  s[11]:  0x00000000
E: s[12]:  0x00000000  s[13]:  0x00000000  s[14]:  0x00000000  s[15]:  0x00000000
E: fpscr:  0x00000000
E: EXC_RETURN: 0x0
E: Faulting instruction address (r15/pc): 0x080014f2
E: >>> ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0
E: Fault during interrupt handling
E: Current thread: 0x20000188 (unknown)
Caught system error -- reason 3
E: r0/a1:  0x00000004  r1/a2:  0x20002af0  r2/a3:  0x00000004
E: r3/a4:  0x20002800 r12/ip:  0xa0000000 r14/lr:  0x08002e6b
E:  xpsr:  0x61000061
E: s[ 0]:  0x00000000  s[ 1]:  0x00000000  s[ 2]:  0x00000000  s[ 3]:  0x00000000
E: s[ 4]:  0x00000000  s[ 5]:  0x00000000  s[ 6]:  0x00000000  s[ 7]:  0x00000000
E: s[ 8]:  0x00000000  s[ 9]:  0x00000000  s[10]:  0x00000000  s[11]:  0x00000000
E: s[12]:  0x00000000  s[13]:  0x00000000  s[14]:  0x00000000  s[15]:  0x00000000
E: fpscr:  0x00000000
E: EXC_RETURN: 0x0
E: Faulting instruction address (r15/pc): 0x08001510
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Fault during interrupt handling
E: Current thread: 0x20000188 (unknown)
Caught system error -- reason 4
ASSERTION FAIL [0] @ WEST_TOPDIR/zephyr/tests/arch/arm/arm_interrupt/src/arm_interrupt.c:207
Intentional assert
E: r0/a1:  0x00000004  r1/a2:  0x000000cf  r2/a3:  0x00000040
E: r3/a4:  0x00000061 r12/ip:  0xa0000000 r14/lr:  0x08002e6b
E:  xpsr:  0x41000061
E: s[ 0]:  0x00000000  s[ 1]:  0x00000000  s[ 2]:  0x00000000  s[ 3]:  0x00000000
E: s[ 4]:  0x00000000  s[ 5]:  0x00000000  s[ 6]:  0x00000000  s[ 7]:  0x00000000
E: s[ 8]:  0x00000000  s[ 9]:  0x00000000  s[10]:  0x00000000  s[11]:  0x00000000
E: s[12]:  0x00000000  s[13]:  0x00000000  s[14]:  0x00000000  s[15]:  0x00000000
E: fpscr:  0x00000000
E: EXC_RETURN: 0x0
E: Faulting instruction address (r15/pc): 0x0800bc38
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Fault during interrupt handling
E: Current thread: 0x20000188 (unknown)
Caught system error -- reason 4
Assertion failed at WEST_TOPDIR/zephyr/tests/arch/arm/arm_interrupt/src/arm_interrupt.c:322: test_arm_interrupt: (post_flag == j is false)
Test flag not set by ISR
FAIL - test_arm_interrupt
===================================================================
START - test_arm_esf_collection
Testing ESF Reporting
E: ***** USAGE FAULT *****
E:   Attempt to execute undefined instruction
E: r0/a1:  0x00000000  r1/a2:  0x00000001  r2/a3:  0x00000002
E: r3/a4:  0x00000003 r12/ip:  0x0000000c r14/lr:  0x0000000f
E:  xpsr:  0x01000000
E: s[ 0]:  0x00000000  s[ 1]:  0x00000000  s[ 2]:  0x00000000  s[ 3]:  0x00000000
E: s[ 4]:  0x00000000  s[ 5]:  0xffffffff  s[ 6]:  0x00000000  s[ 7]:  0x00000000
E: s[ 8]:  0x00000000  s[ 9]:  0x00000000  s[10]:  0x00000000  s[11]:  0x00000000
E: s[12]:  0x00000000  s[13]:  0x00000000  s[14]:  0x00001d46  s[15]:  0xffffffff
E: fpscr:  0xaac7b3df
E: r4/v1:  0x00000004  r5/v2:  0x00000005  r6/v3:  0x00000006
E: r7/v4:  0x00000007  r8/v5:  0x00000008  r9/v6:  0x00000009
E: r10/v7: 0x0000000a  r11/v8: 0x0000000b    psp:  0x200023d0
E: EXC_RETURN: 0xfffffffd
E: Faulting instruction address (r15/pc): 0x0800b0dc
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x20000080 (unknown)
Caught system error -- reason 0

ioannisg, 2021-02-09:
Thanks @ABOSTM , let's ignore the no_optimizations for now (there's an open ticket for this), and focus on the other test-variants.
Could you post the log you get if you increase both ZTEST and ISR stack sizes?

ABOSTM, 2021-02-09:
With ZTEST and ISZ stack double size,
I have the same logs.

PR #32132 (2021-02-14): test: arch: arm: arm_interrupt: clear FPSCR register in ISR
test: arch: arm: arm_interrupt: clear FPSCR register in ISR

Clear Floating Point Status and Control Register (FPSCR),
in case FPU IRQ is selected by the test as "Available IRQ line"

Fixes #31982
commit a276603c (2021-02-14): test: arch: arm: arm_interrupt: clear FPSCR register in ISR

Clear Floating Point Status and Control Register (FPSCR),
to prevent from having the interrupt line set to pending again,
in case FPU IRQ is selected by the test as "Available IRQ line"

Fixes #31982
commit 205c16af (2021-02-19): test: arch: arm: arm_interrupt: clear FPSCR register in ISR

Clear Floating Point Status and Control Register (FPSCR),
to prevent from having the interrupt line set to pending again,
in case FPU IRQ is selected by the test as "Available IRQ line"

Fixes #31982
closed 2021-02-14 by nashif