zephyrproject-rtos/zephyr #31985: riscv: Long execution time when TICKLESS_KERNEL=y

martin-aberg, 2021-02-04:
**Describe the bug**
Running a CPU bound test on RISC-V appears to take almost 30 times
longer when `TICKLESS_KERNEL=y` compared to `TICKLESS_KERNEL=n`.

`crypto.tinycrypt` test case on `qemu_riscv32` takes 142 s or 5 s depending
on `TICKLESS_KERNEL`. Similar for `tests/lib/heap`.

**To Reproduce**
```
$ twister --platform qemu_riscv32 -v -T tests/crypto/tinycrypt -x=CONFIG_TICKLESS_KERNEL=n
INFO    - 1/1 qemu_riscv32              tests/crypto/tinycrypt/crypto.tinycrypt            PASSED (qemu 5.061s)

$ twister --platform qemu_riscv32 -v -T tests/crypto/tinycrypt -x=CONFIG_TICKLESS_KERNEL=y
INFO    - 1/1 qemu_riscv32              tests/crypto/tinycrypt/crypto.tinycrypt            PASSED (qemu 142.436s)
```

Both tests use the same QEMU command line.

**Expected behavior**
It was expected that the two runs above have comparable execution times. Since
this is an emulation the host computer time may not be directly related to the
target system time. However, the difference is rather large.

For reference, doing the same with the `--platform qemu_cortex_m3` gives 6 s
and 5 s.

**Impact**
`TICKLESS_KERNEL` is enabled by default on RISC-V boards using the Machine Timer.

**Environment (please complete the following information):**
- Toolchain: Zephyr SDK 0.12.1
- Zephyr version: v2.5.0-rc2-46-g4d4a636555cd

**Additional context**
The behavior has also been observed when running on RISC-V hardware
(board not in tree).

Would be interesting to know if it reproduces on other RISC-V boards.

nashif, 2021-02-05:
@andyross  any idea what is going on?

katsuster, 2021-02-26:
This is quick report.

- I can reproduce on QEMU on my environment.
- But on my board (SiFive HiFive1 rev.B), I cannot reproduce.
(Both configs take about 30s "test_aes_key_chain"..."test_ccm_vector_8")

So, I guess it's QEMU depended problem... Anyway I'll check more details.

katsuster, 2021-02-26:
I remember that I faced similar problem when I created initial SMP support for RISC-V arch.
So I try to apply this commit and it seems works fine for this issue. Please check it.

Tickless kernel sometimes tries to set MAX timeout into machine timer. Many interrupts from machine timer occurred and Zephyr was going to freeze (or slow down) because current code sometimes calculate wrong next timeout by signed int overflow.

https://github.com/zephyrproject-rtos/zephyr/pull/29105/commits/9769a363856f19ee4015230dcb33f7435850199a

katsuster, 2021-02-26:
FYI: this is a log after applying my patch.

```
twister --platform qemu_riscv32 -v -T ../tests/crypto/tinycrypt -x=CONFIG_TICKLESS_KERNEL=n
INFO    - 1/1 qemu_riscv32              tests/crypto/tinycrypt/crypto.tinycrypt            PASSED (qemu 4.641s)

twister --platform qemu_riscv32 -v -T ../tests/crypto/tinycrypt -x=CONFIG_TICKLESS_KERNEL=y
INFO    - 1/1 qemu_riscv32              tests/crypto/tinycrypt/crypto.tinycrypt            PASSED (qemu 4.085s)
```

andyross, 2021-02-26:
Sorry, I didn't notice this in time.  But it sounds like you have a handle on the root cause (indeed, this test does no timer work whatsoever, and the resulting INT_MAX timeout exposed bugs in the driver).  Please submit as a proper pull request and add me as a reviewer?

We should be able to find the broken math without cutting the tick clamp in half -- my guess is it's something simpler.

katsuster, 2021-02-27:
@andyross
Sure. I created PR #32711.

closed 2021-03-02 by nashif