zephyrproject-rtos/zephyr #31993: Add west extension to parse yml file

carlocaione, 2021-02-04:
My usual workflow is
* hack hack hack
* `west build -b ${board} ${test_dir} --pristine`
* `west build -t debugserver`
* debug debug debug

The `debugserver` option is particularly handy because I do not have to remember all the parameters that are needed for QEMU when starting the gdb server.

Let's say that at a certain point I want to debug on issue on one specific test, i.e. `tests/posix/common/portability.posix.common.tls.newlib`.

With `west` neither I can build that specific test nor I can use `debugserver` to debug it. Instead I have to use `twister` to compile it and found another way to open a gdb server on it. This is really inconvenient.

mbolivar-nordic, 2021-02-04:
I don't understand why west needs to handle the twister specific bits of this.

Can you explain why something like this workflow isn't what you are asking for?

```
$ ./scripts/twister -T tests/posix/common --test tests/posix/common/portability.posix.common.tls.newlib --build-only --platform qemu_cortex_m3
$ west build -t debugserver -d twister-out/qemu_cortex_m3/tests/posix/common/portability.posix.common.tls.newlib/
```

carlocaione, 2021-02-04:
> I don't understand why west needs to handle the twister specific bits of this.

Because it would be handy to have one single tool to do the whole job, without considering that `twister` is way slower than `west`. 

Also it doesn't make sense to me that `west` can be used to compile a sample / test but only one without considering all the remaining tests in the yaml file.

> Can you explain why something like this workflow isn't what you are asking for?

Yes, this works, but it is clearly more cumbersome than just using one single tool `west` with literally 70% less typing.

mbolivar-nordic, 2021-02-04:
OK, so when you wrote:

> With west neither I can build that specific test

It's true that only twister can understand sample.yaml, but...

> nor I can use debugserver to debug it. Instead I have to use twister to compile it and found another way to open a gdb server on it.

 the build directory twister generates is an ordinary one which can be debugged using west, so this part is not a problem, at least.

So this issue is really about just building a test configuration from sample.yaml with west, and not debugging, right?

carlocaione, 2021-02-05:
> So this issue is really about just building a test configuration from sample.yaml or testcase.yaml with west, and not debugging, right?

My expectation is that once I can build a test configuration from sample.yaml or testcase.yaml with west I then can open a debug server on it as easily as before doing `west build -t debugserver` without having to specify the output twister directory and all the rest.

mbolivar-nordic, 2021-02-05:
I will take that as a "yes", I guess.

carlocaione, 2021-02-06:
That is a yes.

mbolivar-nordic, 2021-02-22:
@PerMac what do you think about this one?

mbolivar-nordic, 2021-02-22:
@nashif I was told by @PerMac that you were doing a twister overhaul that was touching the YAML files. Curious what you think about this issue. My first question would by "why is twister so slow" and "why can't twister do this".

I'm not sure sharing the code for parsing this YAML format between west build and twister makes the most sense, but would like your perspective.

nashif, 2021-05-28:
i think this request makes sense, I am often face with this with complex tests or samples that have 10s of options. 

> "why is twister so slow"

slow is relative, slow when you build times and executes 1000s of times in qemu, or slow when you try to build one sample? for one sample/test, I guess the slowness is due to the fact that twister is a tool that parses, builds and runs 1000s of tests and not one, so there is some overhead for startup.

> "why can't twister do this"

quote from above: Because it would be handy to have one single tool to do the whole job.

> I'm not sure sharing the code for parsing this YAML format between west build and twister makes the most sense, but would like your perspective.

Do not think this needs to be shared, for west you only need to 1 or 2 items from the yaml file and that should be enough to satisfy this request. Basically just read the `extra_args` and `extra_configs`, so in west you just point it to the `test` in the sample.yaml or testcase.yaml and west passes those to the build command. Maybe there other options, but the above is a good start.

Prototyped really quickly:

```
diff --git a/scripts/west_commands/build.py b/scripts/west_commands/build.py
index 9fa5f81fd8..6f7f177824 100644
--- a/scripts/west_commands/build.py
+++ b/scripts/west_commands/build.py
@@ -112,6 +112,9 @@ class Build(Forceable):
         group.add_argument('-t', '--target',
                            help='''run build system target TARGET
                            (try "-t usage")''')
+        group.add_argument('-T', '--test-item',
+                           help='''Build based on test data in testcase.yaml
+                           or sample.yaml''')
         group.add_argument('-o', '--build-opt', default=[], action='append',
                            help='''options to pass to the build tool
                            (make or ninja); may be given more than once''')
@@ -207,6 +210,7 @@ class Build(Forceable):
     def _parse_remainder(self, remainder):
         self.args.source_dir = None
         self.args.cmake_opts = None
+
         try:
             # Only one source_dir is allowed, as the first positional arg
             if remainder[0] != _ARG_SEPARATOR:
@@ -219,7 +223,37 @@ class Build(Forceable):
             if remainder:
                 self.args.cmake_opts = remainder
         except IndexError:
-            return
+            pass
+
+        if self.args.test_item:
+            import yaml
+            yf = os.path.join(self.args.source_dir, "sample.yaml")
+            if os.path.exists(yf):
+                with open(yf, 'r') as stream:
+                    try:
+                        y = yaml.safe_load(stream)
+                    except yaml.YAMLError as exc:
+                        log.die(exc)
+                tests = y.get('tests')
+                if tests:
+                    item = tests.get(self.args.test_item)
+                    if item:
+                        extra_args = item.get('extra_args')
+                        extra_configs = item.get('extra_configs')
+                        if extra_args:
+                            a = extra_args.split(" ")
+                            args = ["-D{}".format(arg.replace('"', '')) for arg in a]
+                            if self.args.cmake_opts:
+                                self.args.cmake_opts.extend(args)
+                            else:
+                                self.args.cmake_opts = args
+                        if extra_configs:
+                            args = ["-D{}".format(arg.replace('"', '')) for arg in extra_configs]
+                            if self.args.cmake_opts:
+                                self.args.cmake_opts.extend(args)
+                            else:
+                                self.args.cmake_opts = args
+        return
```

then

```west build -b native_posix samples/drivers/display/ -T sample.display.dummy```

or 

```
 west build -b nrf52840dk_nrf52840 samples/drivers/display/ -T sample.display.shield.ssd1306_128x32
```

nashif, 2021-05-29:
cleaned up a bit and put in a PR... https://github.com/zephyrproject-rtos/zephyr/pull/35808

PR #35808 (2021-06-12): west: parse options from testcase.yaml/sample.yaml during build
Allow the use of build configurations from testcase/sample yaml files.
This addition makes it easy to build a sample or a test with the options
provided in the test file by pointing to the test section to be built on
the command line of `west build`.

Fixes #31993
commit 070cf6a7 (2021-06-12): west: parse options from testcase.yaml/sample.yaml during build

Allow the use of build configurations from testcase/sample yaml files.
This addition makes it easy to build a sample or a test with the options
provided in the test file by pointing to the test section to be built on
the command line of `west build`.

Fixes #31993
closed 2021-06-12 by nashif