zephyrproject-rtos/zephyr #31996: tests/bluetooth/init/bluetooth.init.test_ctlr_peripheral_iso fails to build on a few platforms

galak, 2021-02-04:
`scripts/twister -s tests/bluetooth/init/bluetooth.init.test_ctlr_peripheral_iso -l`

```
ERROR   - nrf51dk_nrf51422          tests/bluetooth/init/bluetooth.init.test_ctlr_peripheral_iso FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/nrf51dk_nrf51422/tests/bluetooth/init/bluetooth.init.test_ctlr_peripheral_iso/build.log
INFO    - Total complete:    1/   3  33%  skipped:  325, failed:    1
ERROR   - nrf52dk_nrf52832          tests/bluetooth/init/bluetooth.init.test_ctlr_peripheral_iso FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/nrf52dk_nrf52832/tests/bluetooth/init/bluetooth.init.test_ctlr_peripheral_iso/build.log
INFO    - Total complete:    2/   3  66%  skipped:  325, failed:    2
ERROR   - nrf52840dk_nrf52840       tests/bluetooth/init/bluetooth.init.test_ctlr_peripheral_iso FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/twister-out/nrf52840dk_nrf52840/tests/bluetooth/init/bluetooth.init.test_ctlr_peripheral_iso/build.log
```

```
In file included from /home/galak/git/zephyr/subsys/bluetooth/controller/ll_sw/ull_iso.c:16:
/home/galak/git/zephyr/subsys/bluetooth/controller/./util/mfifo.h: In function 'mfifo_by_idx_enqueue':
/home/galak/git/zephyr/subsys/bluetooth/controller/./util/mfifo.h:126:2: error: implicit declaration of function 'cpu_dmb' [-Werror=implicit-function-declaration]
  126 |  cpu_dmb(); /* Ensure data accesses are synchronized */
      |  ^~~~~~~
cc1: all warnings being treated as errors
gmake[2]: *** [zephyr/subsys/bluetooth/controller/CMakeFiles/subsys__bluetooth__controller.dir/build.make:290: zephyr/subsys/bluetooth/controller/CMakeFiles/subsys__bluetooth__controller.dir/ll_sw/ull_iso.c.obj] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:3266: zephyr/subsys/bluetooth/controller/CMakeFiles/subsys__bluetooth__controller.dir/all] Error 2
```

galak, 2021-02-04:
@carlescufi please reassign to proper person.  This showed up the most recent daily run: https://buildkite.com/zephyr/zephyr-daily/builds/230

carlescufi, 2021-02-04:
@cvinayak this is similar to the RISC-V one.

PR #32013 (2021-02-05): Bluetooth: controller: Fix build error with ISO support
Fix build error due to missing include file with ISO support
enabled.

Fixes #31996.
commit a980012a (2021-02-05): Bluetooth: controller: Fix build error with ISO support

Fix build error due to missing include file with ISO support
enabled.

Fixes #31996.
closed 2021-02-05 by nashif