zephyrproject-rtos/zephyr #32000: 2021 GSoC Call for Project Ideas - deadline Feb.14, 2021 12:00PM PST

jocelyn-li, 2021-02-04:
About GSoC

Google Summer of Code is a global program focused on introducing students to open source software development. Students work on a 10 week programming project with an open source organization during their break from a post secondary academic program.

Since its inception in 2005, the program has brought together 16,000+ student participants and 13,000 mentors from over 118 countries worldwide. Google Summer of Code has produced 38,000,000+ lines of code for 715 open source organizations 

As a part of Google Summer of Code, student participants are paired with a mentor from the participating organizations, gaining exposure to real-world software development and techniques. Students have the opportunity to spend the break between their school semesters earning a stipend while working in areas related to their interests.

In turn, the participating organizations are able to identify and bring in new developers who implement new features and hopefully continue to contribute to open source even after the program is over. Most importantly, more code is created and released for the use and benefit of all.

More details can be found [here](https://summerofcode.withgoogle.com/about/).

Each project on the Ideas list should include following information
- a project title/description 
- more detailed description of the project (2-5 sentences) 
- expected outcomes 
- skills required/preferred 
- possible mentors
- if possible, an easy, medium or hard rating of each project
- Projects should take ~175 hours for students to complete

Please submit a pull request for your project idea by Feb.14, 2021 12:00PM PST.
- Titled "2021 GSoC Project Idea: project title"
- Labeled "GSoC"

Reference Information
- Project idea defining guideline link is [here](https://google.github.io/gsocguides/mentor/defining-a-project-ideas-list)
- Mentor guide link is [here](https://google.github.io/gsocguides/mentor/what-makes-a-good-mentor)
- Program Timeline link is [here](https://summerofcode.withgoogle.com/how-it-works/#timeline)
- Project idea examples link is [here](https://01.org/linuxmedia/gsoc/gsoc-2020-ideas)

beriberikix, 2021-02-22:
Ah, are submissions for the GSoC CFP closed? I was planning on formally submit #22247.

jocelyn-li, 2021-02-24:
> Ah, are submissions for the GSoC CFP closed? I was planning on formally submit #22247.

Official GSoC CFP was closed on Feb.20.As all the project ideas are put at wiki page https://github.com/zephyrproject-rtos/zephyr/wiki/2021-GSoC-Project-Ideas#project-ideas where GSoC committee will go and review. It still have chance. Please go ahead and submit your project idea. Thanks!

closed 2021-05-11 by github-actions[bot]