zephyrproject-rtos/zephyr #32005: stm32: async uart tests fail

shlomow, 2021-02-04:
**Describe the bug**
When running the uart async api on nucleo-l4r5zi they fails with the following:
```
*** Booting Zephyr OS build v2.5.0-rc2-67-g59365f576e03  ***
Running test suite uart_async_test
===================================================================
START - test_single_read_setup
 PASS - test_single_read_setup
===================================================================
START - test_single_read
 PASS - test_single_read
===================================================================
START - test_chained_read_setup
 PASS - test_chained_read_setup
===================================================================
START - test_chained_read
 PASS - test_chained_read
===================================================================
START - test_double_buffer_setup
 PASS - test_double_buffer_setup
===================================================================
START - test_double_buffer
 PASS - test_double_buffer
===================================================================
START - test_read_abort_setup
 PASS - test_read_abort_setup
===================================================================
START - test_read_abort
 PASS - test_read_abort
===================================================================
START - test_chained_write_setup
 PASS - test_chained_write_setup
===================================================================
START - test_chained_write
E: dma stream 5 is busy.
E: cannot configure the dmamux.
E: dma tx config error!

    Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/uart/uart_async_api/src/test_uart_async.c:512: test_chained_write: (memcmp(chained_write_tx_bufs[1], rx_buf + 10, 10) not equal to 0)
Buffers not equal
 FAIL - test_chained_write
```

This can be a bug in the dma driver or in the uart driver.

erwango, 2021-02-04:
Please check if fixed by #31977

shlomow, 2021-02-04:
fixed, thanks!

closed 2021-02-04 by shlomow