zephyrproject-rtos/zephyr #32018: iotdk: running tests/kernel/mem_pool/ failed sometimes for v1.14-branch

Zhaoningx, 2021-02-05:
To Reproduce

I used a script to run this case a lot of times(total:20 , failed: 4) on github 1.14 branch, it is reproduced.

Steps to reproduce the behavior:

export ZEPHYR_SDK_INSTALL_DIR=~/zephyr-sdk-0.10.3

sanitycheck -p iotdk --device-testing --device-serial /dev/ttyUSB0 -T tests/kernel/mem_pool/mem_pool_api/

see error

Booting Zephyr OS zephyr-v1.14.2-intel-rc3-ww02.2 ***** Running test suite mpool_api
===================================================================
starting test - test_mpool_alloc_free_thread
Exception vector: 0x6, cause code: 0x2, parameter 0x4
Address 0x644
Current thread ID = 0x8000047c
Faulting instruction address = 0x20000bd8
Fatal fault in thread 0x8000047c! Aborting.
 
Environment
OS: Fedora33
Toolchain:zephyr-sdk-0.10.3
Commit ID: a980762f70d704

closed 2021-05-30 by github-actions[bot]

ruuddw, 2021-05-31:
@Zhaoningx : I see this issue was automatically closed (stale bot). Can you please confirm this is no longer an issue (I don't believe the ARC team looked at it - I added the 'ARC' label to make it show up in our queries.

Zhaoningx, 2021-06-01:
Hi @ruuddw , I double checked this issue, it can't be reproduced, I ran it 15 times manually and all passed. So I think we can close this first, and I will re-open this if anything wrong with this.

ruuddw, 2021-06-01:
Thanks for double checking!