zephyrproject-rtos/zephyr #32033: Bluetooth mesh : LPN doesn't receive messages from Friend

Ripitchip, 2021-02-05:
**Describe the bug**
I'm trying to set up a Friend and LPN with nRF52840 boards. The code is based on [onoff-app](https://github.com/zephyrproject-rtos/zephyr/tree/master/samples/boards/nrf/mesh/onoff-app) sample.

Each time the LPN polls a message from Friend, which isn't linked to provisionment or node configuration, the message is dropped with a warning saying it's too short.

**To Reproduce**
Use onoff-app sample. 

The Kconfig is the following one for the **LPN** :
```
CONFIG_BT_DEVICE_NAME="Zephyr LPN"
CONFIG_BT_MESH=y
CONFIG_BT_MESH_RELAY=n
CONFIG_BT_MESH_RELAY_ENABLED=n
CONFIG_BT_MESH_BEACON_ENABLED=n
CONFIG_BT_MESH_FRIEND=n
CONFIG_BT_MESH_LOW_POWER=y
CONFIG_BT_MESH_DEBUG=y
CONFIG_BT_MESH_DEBUG_LOW_POWER=y
# Other have default value
``` 

The Kconfig is the following one for the **Friend** :
```
CONFIG_BT_DEVICE_NAME="Zephyr Friend"
CONFIG_BT_MESH=y
CONFIG_BT_MESH_RELAY=n
CONFIG_BT_MESH_RELAY_ENABLED=n
CONFIG_BT_MESH_BEACON_ENABLED=n
CONFIG_BT_MESH_LOW_POWER=n
CONFIG_BT_MESH_FRIEND=y
CONFIG_BT_MESH_FRIEND_ENABLED=y
CONFIG_BT_MESH_FRIEND_QUEUE_SIZE=16
CONFIG_BT_MESH_DEBUG=y
CONFIG_BT_MESH_DEBUG_FRIEND=y
# Other have default value
``` 
The provisionment is done with the nRF mesh app. They are configured as follow (first LPN, next Friend) :
**LPN** : 
- Generic On Off server of first element : subscription address == group 0xC000
- Generic On Off client of first element : publication address == group 0xC000

**Friend**
- Generic On Off server of first element : subscription address == group 0xC000
- Generic On Off client of first element : publication address == group 0xC000

Other parameters are let as is (default value).

**Logs and console output**

- **First scenario** : pressing button from the LPN. The LED of Friend turns on, then he enqueues a message. At next poll, LPN is unable to read paquet (warning that the packet is too short, hence dropped). The friendship is momentarily broken and established again after a while.
<details>
  <summary> Log of LPN </summary>

button_press_cnt 0x00
button_press_cnt 0x01 onoff_state 0x01
publish to 0xc000 onoff 0x0001 sw_idx 0x0000
gen_onoff_set
addr 0x08 state 0x01
addr 0x0008 onoff 0x01
Node 0x0008 OnOff status from 0x0008 with state 0x01
gen_onoff_set
addr 0x08 state 0x01
addr 0x0008 onoff 0x01
Node 0x0008 OnOff status from 0x0008 with state 0x01
[00:09:13.146,270] <dbg> bt_mesh_lpn.lpn_timeout: state: established
[00:09:13.146,270] <dbg> bt_mesh_lpn.send_friend_poll: lpn->sent_req 0x00
[00:09:13.147,003] <dbg> bt_mesh_lpn.req_sent: req 0x01 duration 90 err 0 state established
[00:09:13.147,003] <dbg> bt_mesh_lpn.lpn_set_state: established -> recv delay
[00:09:13.237,091] <dbg> bt_mesh_lpn.lpn_timeout: state: recv delay
[... 637 lines cut ...]
[00:02:55.738,433] <dbg> bt_mesh_friend.bt_mesh_friend_find: net_idx 0x0000 lpn_addr 0x0008
**[00:02:55.738,433] <wrn> bt_mesh_friend: No matching LPN addr 0x0008**
[00:02:56.184,509] <dbg> bt_mesh_friend.bt_mesh_friend_match: No matching LPN for address 0x000c
[00:02:56.184,539] <dbg> bt_mesh_friend.bt_mesh_friend_find: net_idx 0x0000 lpn_addr 0x0008
**[00:02:56.184,570] <wrn> bt_mesh_friend: No matching LPN addr 0x0008**
[00:02:56.629,913] <dbg> bt_mesh_friend.bt_mesh_friend_match: No matching LPN for address 0x000c
[00:02:56.629,943] <dbg> bt_mesh_friend.bt_mesh_friend_find: net_idx 0x0000 lpn_addr 0x0008
**[00:02:56.629,943] <wrn> bt_mesh_friend: No matching LPN addr 0x0008**

</details>

After both devices reboot, they are able to establish the friendship. But same problem would repeat if I press the Friend's button again.

**Expected behavior**
I expect the LPN to be able to receive messages from Friend. Well, at least it suprises me that the friendship breaks.

**Impact**
I need the Friend-LPN feature for a project, so it blocks me a bit.

**Environment (please complete the following information):**
- Linux Ubuntu 20.04.1 LTS
- Toolchain : Zephyr SDK
- Version 2.5.0 rc2 as per version file. The command `git rev-parse HEAD` gives me : `92c93b1b7fbae1b7623c6423b203d2504c0157aa` (which is the latest commit on master at 5th February).

Note : I'm on latest commit on master because before I was at version 2.4.99 (with commit `52ad9a468c1a3ae9a00d1e5aac389c5bfcf087ec`) and the friendship couldn't even be set up. Then, as I saw commit changes about friendship, I tried commit `c262239680507049a5a8d3c3cb9e18405413cdb3` but I had the same problem as described above. Then I tried most recent commit.

trond-snekvik, 2021-02-08:
Hi, thanks for reporting, this uncovered a couple of regressions. 

The main issue here comes from the re-encryption mechanism in the friend module, that decrypts and re-encrypts unsegmented messages from the friend node to the LPN. This mechanism has seen two regressions: 
1. The app_idx and net_idx in the message context structure passed to the app_key module is not initialized before being used, leading to invalid appkey index input parameters (see the line `[00:09:33.158,599] bt_mesh_friend: No matching LPN addr 0x0008` in the first friend log).
2. The sequence number passed to the re-encrypt function is no longer being reset to match the current sequence number, making subsequent messages from the friend to the lpn hit the LPNs message cache.

I have opened #32080 to resolve the issue you're seeing. If you can, please verify that this PR fixes your issue.

Ripitchip, 2021-02-08:
Hi, 

thanks very much for the PR. At first glance, it seems to solve the issue ! :)

Have a nice day.

trond-snekvik, 2021-02-08:
Reopening until PR goes in

PR #32080 (2021-02-10): Bluetooth: Mesh: Friend re-encrypt regressions
#28511 introduced a regression in the mechanism for re-encrypting unsegmented app messages from a Friend to one of its LPNs.
This PR fixes two issues related to this:
-  Bluetooth: Mesh: Initialize msg_ctx when re-encrypting friend msg
   Set app_idx and net_idx in the msg_ctx before calling bt_mesh_keys_resolve when re-encrypting friend messages, as they'll be referenced inside the function.
-  Bluetooth: Mesh: Update seqnum when re-encrypting for friend
   Sets the sequence number when re-encrypting messages from the friend to the lpn.

Fixes #3 [... cut]
closed 2021-02-10 by nashif