zephyrproject-rtos/zephyr #32035: Bluetooth: application notification when MTU updated

JordanYates, 2021-02-05:
**Is your enhancement proposal related to a problem? Please describe.**

The Bluetooth ATT MTU controls the maximum size of write, read_rsp, notification and indication packets.
Attempting to send more data than the current MTU results in warnings/errors such as:
```
[00:01:30.406,005] <wrn> bt_att: No ATT channel for MTU 48
[00:01:30.406,005] <wrn> bt_gatt: No buffer available to send indication
```

The MTU update process is as follows:
1. The client issues a request for a new MTU value
2. The server responds with the minimum of the requested MTU and the servers maximum MTU (`CONFIG_BT_L2CAP_TX_MTU`/`CONFIG_BT_L2CAP_RX_MTU`)

Therefore the amount of data that the server can transmit can vary over time, even on the same connection.
Currently the connection MTU can only be queried (`bt_conn_get_mtu`), with no option for the application to be notified when the value changes.

The current options are therefore to query the MTU before each transmit, or to re-query the MTU if a transmission fails due to insufficient space. Neither of these options are great for proactively constructing packets to send based on an expected data size.

**Describe the solution you'd like**

A callback added to `struct bt_conn_cb` that notifies the application when the MTU changes.

Vudentz, 2021-04-12:
Having a callback doesn't fix this problem, in fact it would likely result in even more code because it can still cause the above errors while the callback notification is pending. Also the spec is pretty clear regarding not to attempt to exchange the MTU multiple times so it is normally expected the client do the exchange MTU in the very beginning to avoid this sort of problem, or in case of custom protocol on top use L2CAP COC instead of using a GATT attribute as PSM, which is likely much faster and does support segmentation.

Vudentz, 2021-04-12:
```
4.3.1 Exchange MTU
This sub-procedure is used by the client to set the ATT_MTU to the maximum
possible value that can be supported by both devices when the client supports
a value greater than the default ATT_MTU for the Attribute protocol. This sub-
procedure **shall** only be initiated once during a connection.
```

PR #32039 (2021-06-07): Bluetooth: gatt: notify application of MTU updates
Add a new callback to `struct bt_conn_cb` to allow applications to track
changes to the ATT MTU. This callback is only called when the remote
side changes the MTU. The application is already notified of a changed
MTU if it initiated the update via the `struct bt_gatt_exchange_params`
callback.

Fixes #32035.
commit 43b24006 (2021-06-07): Bluetooth: gatt: callbacks on ATT MTU update

Adds a new callback structure to `<gatt.h>` for receiving notifications
of ATT MTU updates. This callback is called regardless of whether the
MTU update was initiated locally or remotely.

Fixes #32035.
closed 2021-06-07 by carlescufi