zephyrproject-rtos/zephyr #32037: Implement an I2S / SAI (shim) driver for NXP IMX-RT1060 (mcux).

micnet49, 2021-02-05:
**Is your feature request related to a problem? Please describe.**
I would like to develop an application which can play / receive audio via the SAI1 peripheral (by means of eDMA) of the NXP IMX-RT1060 MCU using the board IMXRT1060-EVK (and its I2C controlled codec WM8960).

**Describe the solution you'd like**
I simply like to use the Zephyr I2S API for getting audio sending and audio receiving to work at the SAI1 interface. I guess a shim driver for the exisiting [NXP HAL `driver fsl_sai.c` and `fsl_sai_edma.c` ](https://github.com/NXPmicro/hal_nxp/tree/master/mcux/drivers/imx) has to be developped.

**Describe alternatives you've considered**
I tried to use the NXP HAL drivers functions from `driver fsl_sai.h` and `fsl_sai_edma.h` directly... still without success.

**Additional context**
Codec driver for WM8960 (also missing for Zephyr).

micnet49, 2021-02-22:
Is there a chance that there will be a contribution to this issue within the next three month?

MaureenHelm, 2021-03-24:
> Is there a chance that there will be a contribution to this issue within the next three month?

We're planning to work on this, but a little further out. In the meantime, #32054 may provide some inspiration if you'd like to tackle it sooner.

micnet49, 2021-05-26:
I would like to have the "stale" label removed. The issue is still important for my application.

JonathanVanWin, 2021-07-13:
If you are interested, I started developing the general API, and driver for ST https://github.com/zephyrproject-rtos/zephyr/issues/34631#issuecomment-879055848

PR #37335 (2021-12-02): i2s: add i2s support for NXP imxrt10xx platform
1. add i2s/audio codec/pinmux/anatip drivers for rt series
2. enable i2s on mimxrt1060_evk to demo
2. tested on tests/drivers/i2s/i2s_api and i2s_speed

Fixes #32037
closed 2021-12-02 by carlescufi