zephyrproject-rtos/zephyr #32059: Getting Started - Windows - Toolchain not found

sidprice, 2021-02-06:
Just trying to get Zephyr installed and tested using the Getting Started Guide for Windows. When I try 

`F:\DataRoot\Projects\Zephyr\zephyrproject\zephyr>echo %ZEPHYR_TOOLCHAIN_VARIANT%
gnuarmemb

F:\DataRoot\Projects\Zephyr\zephyrproject\zephyr>echo %GNUARMEMB_TOOLCHAIN_PATH%
C:\GNU_ARM_9

F:\DataRoot\Projects\Zephyr\zephyrproject\zephyr>west build -p auto -b nucleo_f401re samples\basic\blinky
-- west build: generating a build system
Including boilerplate (Zephyr base (cached)): F:/DataRoot/Projects/Zephyr/zephyrproject/zephyr/cmake/app/boilerplate.cmake
-- Application: F:/DataRoot/Projects/Zephyr/zephyrproject/zephyr/samples/basic/blinky
-- Zephyr version: 2.5.0-rc2 (F:/DataRoot/Projects/Zephyr/zephyrproject/zephyr)
-- Found west (found suitable version "0.9.0", minimum required is "0.7.1")
-- Board: nucleo_f401re
-- Cache files will be written to: F:/DataRoot/Projects/Zephyr/zephyrproject/zephyr/.cache
-- Found toolchain: gnuarmemb (C:/GNU_ARM_9/bin)
CMake Error at F:/DataRoot/Projects/Zephyr/zephyrproject/zephyr/cmake/compiler/gcc/generic.cmake:8 (message):
  Zephyr was unable to find the toolchain.  Is the environment misconfigured?

  User-configuration:

  ZEPHYR_TOOLCHAIN_VARIANT: gnuarmemb

  Internal variables:

  CROSS_COMPILE: C:/GNU_ARM_9/bin/bin/arm-none-eabi-

  TOOLCHAIN_HOME: C:/GNU_ARM_9/bin

Call Stack (most recent call first):
  F:/DataRoot/Projects/Zephyr/zephyrproject/zephyr/cmake/generic_toolchain.cmake:42 (include)
  F:/DataRoot/Projects/Zephyr/zephyrproject/zephyr/cmake/app/boilerplate.cmake:534 (include)
  F:/DataRoot/Projects/Zephyr/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:24 (include)
  F:/DataRoot/Projects/Zephyr/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:40 (include_boilerplate)
  CMakeLists.txt:4 (find_package)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: 'C:\Program Files\CMake\bin\cmake.EXE' '-DWEST_PYTHON=c:\python37\python.exe' '-BF:\DataRoot\Projects\Zephyr\zephyrproject\zephyr\build' '-SF:\DataRoot\Projects\Zephyr\zephyrproject\zephyr\samples\basic\blinky' -GNinja
`

What seems odd is the line:

`CROSS_COMPILE: C:/GNU_ARM_9/bin/bin/arm-none-eabi-`

'bin' is in the path twice.

Any advice on how tp resolve this?

I have restarted the PC and the results are the same.

GunZi200, 2021-02-06:
Could it be that your  `GNUARMEMB_TOOLCHAIN_PATH` is set as  `GNUARMEMB_TOOLCHAIN_PATH=C:/GNU_ARM_9/bin/` when it should be `C:/GNU_ARM_9/`? I had this problem earlier today also.

sidprice, 2021-02-06:
No, my environment variables are shown at the top of the above post:

`F:\DataRoot\Projects\Zephyr\zephyrproject\zephyr>echo %ZEPHYR_TOOLCHAIN_VARIANT%
gnuarmemb

F:\DataRoot\Projects\Zephyr\zephyrproject\zephyr>echo %GNUARMEMB_TOOLCHAIN_PATH%
C:\GNU_ARM_9`

carlescufi, 2021-02-07:
Can you makes sure you try in a clean repo, with no `build/` folder at all in it?

sidprice, 2021-02-07:
@carlescufi Thank you, deleting the "build" folder fixed the issue.

Seems the tools should be able to take care of this, a "clean" option perhaps?

GunZi200, 2021-02-08:
My issue went away when I deleted the `<CMake Project DIR>/build/CMakeCache.txt` manually.

tejlmand, 2021-02-08:
> Seems the tools should be able to take care of this, a "clean" option perhaps?

`pristine` ? 
https://docs.zephyrproject.org/latest/guides/west/build-flash-debug.html#pristine-builds

>  It seems environment variables are only read the first time the CMake project is built.

Correct, and the reason for this is that people can work with multiple toolchains / boards / samples simultaneously, like:
```
# Can be always defined.
export ZEPHYR_SDK_INSTALL_DIR=/<path>/zephyr-sdk-0.12.0/
export GNUARMEMB_TOOLCHAIN_PATH=/<path>/gcc-arm-none-eabi-10-2020-q4-major/
export LLVM_TOOLCHAIN_PATH=/<path>/clang+llvm-10.0.0-x86_64-linux-gnu-ubuntu-18.04/

# Build folder for sample Zephyr SDK
export ZEPHYR_TOOLCHAIN_VARIANT=zephyr
cmake -DBOARD=<board> -Bbuild_zephyr samples/<sample>

# Build folder for sample GNU Arm emb
export ZEPHYR_TOOLCHAIN_VARIANT=gnuarmemb
cmake -DBOARD=<board> -Bbuild_gnuarmemb samples/<sample>

# Build folder for sample LLVM
export ZEPHYR_TOOLCHAIN_VARIANT=llvm
cmake -DBOARD=<board> -Bbuild_llvm samples/<sample>

# Users can easily build a sample with multiple toolchains, and that
# particular build folder does not change behavior based on new env setting.
ninja -Cbuild_zephyr     # Still builds with Zephyr SDK
ninja -Cbuild_gnuarmemb  # Still builds with GNU Arm emb
ninja -Cbuild_llvm       # Still builds with LLVM
```

The build folder has been configured for a given toolchain, which allows having multiple build folders and not having to worry for the current setting in environment, as that setting is only used for initial CMake run.

Users can still do a `ninja / make pristine` or `west build --pristine` if they want a clean build from scratch.

This is also true for other variables that can be picked up from environment settings, like BOARD, SHIELD, ZEPHYR_BASE.
This allows users to jump between them and do rebuilds without the build system changes behavior from initial configuration.

All those settings may also be specified using `-D<var>=<val>` as CMake argument.

tejlmand, 2021-02-08:
> My issue went away when I deleted the <CMake Project DIR>/build/CMakeCache.txt manually. 

Note. A `cmake -D<var>=<val>` will update the CMake cache to the new value, and this has precedence over the environment setting.

carlescufi, 2021-02-08:
Closing since this is resolved now.

closed 2021-02-08 by carlescufi