zephyrproject-rtos/zephyr #32070: How to manage power consumption when working with peripheral_hr sample on NRF52832

selman-nus, 2021-02-08:
Hi, 
I am using a modified version of the peripheral_hr example to sample ADC values and send it to a mobile phone using BLE (100Hz). 
NRF52832 is using 3mA on average and I couldn't figure out how to reduce this. I checked several issue posts here and I guess current consumption should be less than 1 mA? 

Is there a power management function that I can use to reduce power consumption? 
(Maybe to reduce CPU frequency or to turn off unused blocks of the MCU?)

My modified code:
[https://pastebin.com/Z1KHcXwV](https://pastebin.com/Z1KHcXwV)

These are the issues that I have seen:
https://github.com/zephyrproject-rtos/zephyr/issues/30304#issue-753067877
https://github.com/zephyrproject-rtos/zephyr/issues/31989#issue-801211963

cvinayak, 2021-02-08:
Why do you use `NRFX_DELAY_US`  ? This will not put the CPU to sleep.

selman-nus, 2021-02-08:
Sampling rate is 100Hz, so delays between cycles are only 10ms. Do you think it would make a difference? I will definitely test! 

What else can I try?

pabigot, 2021-02-08:
`k_sleep(K_MSEC(10));`

selman-nus, 2021-02-09:
It turns out I had board related issues too. (EBSHSNZWZ - taiyo yuden)

1. I separated the power line of NRF52 from the FTDI chip already but I didn't know I also had to cut rx/tx etc. It reduced most of the consumption by itself already. 

2. Then I changed `NRFX_DELAY_US` to `k_sleep`. It also helped a lot. I didn't expect NRF could transmit the packet that fast and go to sleep. 

Below is the final current consumption while transmitting ADC data (1.8v).

Please advise me if there are other ways to reduce the consumption even more. One thing that comes to my mind is reducing the Tx power of BLE to the lowest dBm level. 
Any other suggestions? I am pushing the limits here :)

[image]

In case the image is not visible: 
average: 562uA
max: 9.3mA

cvinayak, 2021-02-10:
I see you idle current is higher, ~300 uA, ADC when in use should use 700uA, but otherwise when no other peripheral is in use your idle current should drop down < 10uA. Check for your idle current with your application running but with BLE and ADC code disabled.

pabigot, 2021-02-10:
Also confirm whether you're measuring just the MCU, or the power consumed by the entire board.  Unless the PCB is optimized for power consumption 300 uA as a minimum might be all you can expect.  See what you get on an application that does nothing but go to sleep.

selman-nus, 2021-02-11:
Thank you for your suggestions. I did the following:

1- Change the board to nRF52 dev kit since it seems more reliable for power measurements.
2- Test the same code (ADC and BLE). Exact same current consumption.
3- Open hello_world example and measure the consumption. Result was 300uA. Still not below 10uA.
4- Delete printk functions and add k_sleep to enable low power state. Still same, 300uA.
5- Enable several power management configs:
```
CONFIG_DEVICE_POWER_MANAGEMENT
CONFIG_SYS_POWER_MANAGEMENT
```
and result is still same. 300uA. 

My main file looks like this now:
```
#include <zephyr.h>
//#include <sys/printk.h>

void main(void)
{
	//printk("Hello World! %s\n", CONFIG_BOARD);
          k_sleep(K_SECONDS(5));
}
```

**So I don't know what else to try to get that 10uA as @cvinayak advised. I am eager to listen to your suggestions?**
.
.
.

Here is the result from the power profiler: 
(All measurements are done on nRF52 dev kit and I am using SEGGER Studio with nRF Connect SDK which is executed by nRF toolchain manager)

[image]

pabigot, 2021-02-11:
Set `CONFIG_SERIAL=n`.  The 300uA is probably the UART peripheral being enabled.

selman-nus, 2021-02-11:
FInally! Haha thank you. This solves the problem. 
The average is less than 3uA now. 

Here is the final measurement on nRF52 dk

[image]

pabigot, 2021-02-11:
Great! I'm intending to provide more information on power measurement and management in the `samples/boards/nrf/system_off` example.  That already notes the need to disable UART, but it could probably be made more clear.

closed 2021-02-11 by pabigot