zephyrproject-rtos/zephyr #32071: devicetree: `bus:` does not work in `child-binding`

kurddt, 2021-02-08:
**Describe the bug**
I'm trying to have a child binding declare a new bus controller. That does not seems to work as expected, the childs of this new controller does not refers to it in the generated defines.

It also seems that even when using a separate binding file for the controller, it must be on a bus (binding must contains `on-bus`) otherwise the childs also don't use it as the controller

**To Reproduce**
Steps to reproduce the behavior:
Run gen_defines with provided dts and .yaml files
The joined archive contains:
* zephyr.dts: small description containing a soc, a root i2c controller, an i2c mux, i2c devices
* bindings: binding of the i2c mux using `child-bindings`
* bindings_2: bindings of the i2c mux using two separate files
* debug_child.h: generated header using `bindings` folder
* debuf_sep.h: generated header using `bindings_2` folder

**Expected behavior**
Childs of the controller should refers to it in their `*_BUS` define

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Arm gnu
 - Zephyr v2.4

kurddt, 2021-02-08:
@mbolivar-nordic

mbolivar-nordic, 2021-02-08:
@kurddt Thank you for the detailed report. This is an interesting case; I've never run into the TCA9546A.

Let me confirm the problem.

In debug_sep.h, I see the following bus macros generated for `/soc/i2c@40004000/tca9546a@3B/mux_i2c@0/tmp116@49`:

```c
/* Bus info (controller: '/soc/i2c@40004000/tca9546a@3B/mux_i2c@0', type: 'i2c') */
#define DT_N_S_soc_S_i2c_40004000_S_tca9546a_3b_S_mux_i2c_0_S_tmp116_49_BUS_i2c 1
#define DT_N_S_soc_S_i2c_40004000_S_tca9546a_3b_S_mux_i2c_0_S_tmp116_49_BUS DT_N_S_soc_S_i2c_40004000_S_tca9546a_3b_S_mux_i2c_0
```

This is what you want. The `tmp1` device is an I2C device on the `mux1` bus.

In debug_child.h, I see these instead:

```c
/* Bus info (controller: '/soc/i2c@40004000', type: 'i2c') */
#define DT_N_S_soc_S_i2c_40004000_S_tca9546a_3b_S_mux_i2c_0_S_tmp116_49_BUS_i2c 1
#define DT_N_S_soc_S_i2c_40004000_S_tca9546a_3b_S_mux_i2c_0_S_tmp116_49_BUS DT_N_S_soc_S_i2c_40004000
```

This is not what you want. This binding points at the twim device, which is the wrong bus.

I don't see anything wrong with bindings_2/i2c-tca9546a.yaml and I agree this looks like a bug at this point. I'm investigating.

mbolivar-nordic, 2021-02-09:
@kurddt I believe https://github.com/zephyrproject-rtos/zephyr/pull/32105 is a fix. Please retest and confirm on your end if you can. Thanks.

kurddt, 2021-02-09:
@mbolivar-nordic I looked and tested your fix. Would it be possible to have the same behavior when the child node does not have any compatible?

In my use case this binding will not be used anywhere in the driver code, the parent device will create the child devices.

mbolivar-nordic, 2021-02-09:
Thanks for testing!

> Would it be possible to have the same behavior when the child node does not have any compatible?

Not easily with the current tooling, I'm afraid.

kurddt, 2021-02-11:
Ok, I think we can close this ticket when your fix is merged then. I will create a new enhancement request for the child compatible unless @mbolivar-nordic you prefer to do so yourself ?

mbolivar-nordic, 2021-02-23:
> I will create a new enhancement request for the child compatible unless @mbolivar-nordic you prefer to do so yourself ?

Please go ahead, thanks.

PR #32105 (2021-03-25): devicetree: fix child binding registry
Whenever a child-binding: dict has a compatible, we ought to make that available in EDT._compat2binding. If we don't, we can't look it up later.

This has adverse effects like missing child bindings which describe buses.

Fix it.

Fixes: #32071
commit 2dca9f4a (2021-03-25): dts: edtlib: fix child bindings which have compatibles

Whenever a child-binding: dict has a compatible, we ought to make
that available in EDT._compat2binding. If we don't, we can't look it
up later.

This has adverse effects like missing child bindings which describe
buses.

Fixes: #32071
closed 2021-03-25 by galak