zephyrproject-rtos/zephyr #32072: tests/kernel/common seems to fail on nrf52833dk_nrf52833

ioannisg, 2021-02-08:
**Describe the bug**
tests/kernel/common tests variants all fail on nrf52833dk_nrf52833

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=nrf52833dk_nrf52833
3. make
4. See error

**Expected behavior**
The test should not assert.

**Impact**
Should fix this before release.

**Logs and console output**

```
DEBUG   - DEVICE: ***** delaying boot 500ms (per build configuration) *****
DEBUG   - DEVICE: *** Booting Zephyr OS build v2.5.0-rc3-2-g89a909633434  (delayed boot 500ms) ***
DEBUG   - DEVICE: Running test suite common
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_bootdelay
DEBUG   - DEVICE: PASS - test_bootdelay
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_irq_offload
DEBUG   - DEVICE: PASS - test_irq_offload
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_byteorder_memcpy_swap
DEBUG   - DEVICE: PASS - test_byteorder_memcpy_swap
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_byteorder_mem_swap
DEBUG   - DEVICE: PASS - test_byteorder_mem_swap
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_sys_get_be64
DEBUG   - DEVICE: PASS - test_sys_get_be64
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_sys_put_be64
DEBUG   - DEVICE: PASS - test_sys_put_be64
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_sys_get_be48
DEBUG   - DEVICE: PASS - test_sys_get_be48
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_sys_put_be48
DEBUG   - DEVICE: PASS - test_sys_put_be48
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_sys_get_be32
DEBUG   - DEVICE: PASS - test_sys_get_be32
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_sys_put_be32
DEBUG   - DEVICE: PASS - test_sys_put_be32
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_sys_get_be24
DEBUG   - DEVICE: PASS - test_sys_get_be24
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_sys_put_be24
DEBUG   - DEVICE: PASS - test_sys_put_be24
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_sys_get_be16
DEBUG   - DEVICE: PASS - test_sys_get_be16
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_sys_put_be16
DEBUG   - DEVICE: PASS - test_sys_put_be16
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_sys_get_le16
DEBUG   - DEVICE: PASS - test_sys_get_le16
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_sys_put_le16
DEBUG   - DEVICE: PASS - test_sys_put_le16
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_sys_get_le24
DEBUG   - DEVICE: PASS - test_sys_get_le24
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_sys_put_le24
DEBUG   - DEVICE: PASS - test_sys_put_le24
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_sys_get_le32
DEBUG   - DEVICE: PASS - test_sys_get_le32
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_sys_put_le32
DEBUG   - DEVICE: PASS - test_sys_put_le32
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_sys_get_le48
DEBUG   - DEVICE: PASS - test_sys_get_le48
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_sys_put_le48
DEBUG   - DEVICE: PASS - test_sys_put_le48
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_sys_get_le64
DEBUG   - DEVICE: PASS - test_sys_get_le64
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_sys_put_le64
DEBUG   - DEVICE: PASS - test_sys_put_le64
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_atomic
DEBUG   - DEVICE: PASS - test_atomic
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_bitfield
DEBUG   - DEVICE: SKIP - test_bitfield
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_printk
DEBUG   - DEVICE: 22 113 10000 32768 40000 22
DEBUG   - DEVICE: p 112 -10000 -32768 -40000 -22
DEBUG   - DEVICE: 0x1 0x01 0x0001 0x00000001 0x0000000000000001
DEBUG   - DEVICE: 0x1 0x 1 0x   1 0x       1
DEBUG   - DEVICE: 42 42 0042 00000042
DEBUG   - DEVICE: -42 -42 -042 -0000042
DEBUG   - DEVICE: 42 42   42       42
DEBUG   - DEVICE: 42 42 0042 00000042
DEBUG   - DEVICE: 255     42    abcdef        42
DEBUG   - DEVICE: 68719476735 -1 18446744073709551615 ffffffffffffffff
DEBUG   - DEVICE: 0xcafebabe 0xbeef 0x2a
DEBUG   - DEVICE: PASS - test_printk
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_timeout_order
DEBUG   - DEVICE: thread 0 synced on timer 0
DEBUG   - DEVICE: thread 1 synced on timer 1
DEBUG   - DEVICE: thread 2 synced on timer 2
DEBUG   - DEVICE: PASS - test_timeout_order
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_clock_uptime
DEBUG   - DEVICE: PASS - test_clock_uptime
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_clock_cycle
DEBUG   - DEVICE: PASS - test_clock_cycle
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_version
DEBUG   - DEVICE: PASS - test_version
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_multilib
DEBUG   - DEVICE: PASS - test_multilib
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_thread_context
DEBUG   - DEVICE: PASS - test_thread_context
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_ms_time_duration
DEBUG   - DEVICE:
DEBUG   - DEVICE: Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/common/src/clock.c:211: test_ms_time_duration: (tdata.duration_count == 1 is false)
DEBUG   - DEVICE: duration 0 not 1
DEBUG   - DEVICE: FAIL - test_ms_time_duration
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_bounds_check_mitigation
DEBUG   - DEVICE: PASS - test_bounds_check_mitigation
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: Test suite common failed.
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows):  Ubuntu 20.04 LTS
 - Toolchain (e.g Zephyr SDK, ...):  Zephyr SDK
 - Commit SHA or Version used: 89a90963343437378a462aa64431b2283b7c397a

**Additional context**
- I tested nRF52dk, nrf5340dk and the test is passing. Might be only _this_ platform that fails.

ioannisg, 2021-02-08:
@carlescufi @anangl @pabigot I would appreciate if anyone else could test on this board and provide feedback.

PerMac, 2021-02-08:
I don't have nrf52833 in my setups. I've run this test on nrf52dk_nrf52832, nrf9160dk_nrf9160 and nrf52840dk_nrf52840 and all tests passed from this suite

jfischer-no, 2021-02-08:
@ioannisg Tested it on nrf52833dk_nrf52833 version 0.9.0 and 1.0.0, all test pass.

ioannisg, 2021-02-08:
> @ioannisg Tested it on nrf52833dk_nrf52833 version 0.9.0 and 1.0.0, all test pass.

Thanks @jfischer-no I think I've got board revision 0.7.1 - that might be explaining why this fails. Going to remove the blocking priority.

anangl, 2021-02-08:
@ioannisg FWIW I've just run the test on my board which is revision 0.7.0, and all test cases passed.

pabigot, 2021-02-08:
@ioannisg I also don't have this board.

You might try running `samples/boards/nrf/clock_skew` on your board, and compare the error to those of another Nordic board that passes.  This tests uses HFCLK to time LFCLK, and a high skew between those clocks is often the source of similar errors.  (May sure the clock config is the same as the application uses.)

ioannisg, 2021-02-08:
Will try that, thanks @pabigot

ioannisg, 2021-02-08:
@pabigot this is what I getting when i run the sample on my nrf52833 board:
```
Ty  Latest           Base             Span             Err
HF  00:00:20.270287  00:00:00.015638  00:00:20.254648
LF  00:00:20.533050  00:00:00.503265  00:00:20.029785 -00:00:00.224863
Skew 1.011226 ; err -11226415 ppb

```

And this is what i get on my nrf52832 board
```
Ty  Latest           Base             Span             Err
HF  00:00:20.270287  00:00:00.015638  00:00:20.254648
LF  00:00:20.533050  00:00:00.503265  00:00:20.029785 -00:00:00.224863
Skew 1.011226 ; err -11226415 ppb
```

Could that difference explain the error I see?
@anangl @jfischer-no could you compare this with your findings by running the sample on your boards?

pabigot, 2021-02-08:
@ioannisg You should let that run for a while; the first reading doesn't have enough duration to estimate accurately.  See the output in the sample README.  I'd give it at least five minutes.

But yes, that's unusual from my perspective, first in that the skews are negative (LFCLK runs slower than HFCLK) and in magnitude (your skews are at least twice what I see).  The sign is most likely to be a problem with the test.

ioannisg, 2021-02-08:
I'll get back to this with new data @pabigot thanks.

pabigot, 2021-02-08:
> The sign is most likely to be a problem with the test.

By which I mean that the test is more likely to be robust if LFCLK is *faster* than HFCLK, because the timer will fire and increase the count early.  If LFCLK is slower the spinning wait will complete before the timer fires.  Especially as your board is showing an error of 1.3% so the 1 ms slop as an upper bound is insufficient: you would have to wait for 1.12 additional ms to pass.

You could test that by changing:
```
        k_busy_wait((DURATION + 1) * 1000);
```
to
```
        k_busy_wait((DURATION + 2) * 1000);
```

jfischer-no, 2021-02-09:
nrf52833 0.9.0
```
Ty  Latest           Base             Span             Err
HF  00:10:23.455168  00:00:00.015667  00:10:23.439501
LF  00:10:21.659088  00:00:00.404632  00:10:21.254455 -00:00:02.185046
Skew 1.003517 ; err -3517150 ppb
```

nrf52833 1.0.0
```
Ty  Latest           Base             Span             Err                                    
HF  00:10:23.760501  00:00:00.015667  00:10:23.744834                                         
LF  00:10:21.651428  00:00:00.397583  00:10:21.253845 -00:00:02.490989                        
Skew 1.004010 ; err -4009604 ppb   
```