zephyrproject-rtos/zephyr #32084: Custom Log Backend breaks performance

caco3, 2021-02-08:
I try to implement my own log backend.
It is used additionally to the standard UART LOG Backend as enabled through
```
CONFIG_LOG=y
CONFIG_UART_CONSOLE=y
```

I did not find much documentation about custon backend implementation except https://lgl88911.gitee.io/2019/04/03/Zephyr-log%E7%B3%BB%E7%BB%9F%E5%8E%9F%E7%90%86%E4%B9%8Bbackend/ which itself refers to `zephyr/blob/master/subsys/logging/log_backend_bt.c`.

So I took the code from `log_backend_bt.c` and gave it a try:
```c
#include <logging/log_backend.h>
#include <logging/log_core.h>
#include <logging/log_msg.h>
#include <logging/log_output.h>
#include <logging/log_backend_std.h>

//#define USE_PRINTK // backend writes to also to standard UART for testing (not performant at all!

static int char_out(uint8_t *data, size_t length, void *ctx) {
    ARG_UNUSED(ctx);

#ifdef USE_PRINTK
    printk("  bt log:");
    for (size_t i = 0; i < length; i++) {
        printk("%c", data[i]);
    }
#endif

    return length;
}

static uint8_t bt_output_buf[250];

LOG_OUTPUT_DEFINE(log_output_bt, char_out, bt_output_buf, 250);

static void put(const struct log_backend *const backend, struct log_msg *msg) {
#ifdef USE_PRINTK
    uint32_t flags;
    flags = LOG_OUTPUT_FLAG_LEVEL | LOG_OUTPUT_FLAG_TIMESTAMP;
    if (IS_ENABLED(CONFIG_LOG_BACKEND_FORMAT_TIMESTAMP)) {
        flags |= LOG_OUTPUT_FLAG_FORMAT_TIMESTAMP;
    }

    log_output_msg_process(&log_output_bt, msg, flags);
#endif

    log_msg_put(msg); // Release msg, it will be released when no one is using msg
}

static void log_backend_bt_init(void) {

}

static void panic(struct log_backend const *const backend) {
    log_backend_std_panic(&log_output_bt);
}

static void dropped(const struct log_backend *const backend, uint32_t cnt) {
    ARG_UNUSED(backend);

    log_backend_std_dropped(&log_output_bt, cnt);
}

const struct log_backend_api log_backend_bt_api = {
    .put = put,
    .put_sync_string = NULL,
    .put_sync_hexdump = NULL,
    .panic = panic,
    .init = log_backend_bt_init,
    .dropped = dropped,
};

LOG_BACKEND_DEFINE(log_backend_bt, log_backend_bt_api, true);
```

It is running on a [nrf52840-Dongle](https://www.nordicsemi.com/Software-and-tools/Development-Kits/nRF52840-Dongle).

Without my own backend enabled, the system works fine and I see the logs on UART without any lag.

**Expected behavior**
The custom backend should not create an inacceptable lag. This is especially true when `USE_PRINTK` is commented out (disabled) as shown above.

**Impact**
Even when `USE_PRINTK` is commented out (disabled), the normal UART log is lagging to a state the system is not usable anymore (even the 60s watchdog fires occasionally).

**Environment (please complete the following information):**
 - OS: Linux, Ubuntu 20.04
 - Toolchain: Zephyr SDK 0.12.2
 - Commit SHA or Version used: 2.5.0-rc3

caco3, 2021-02-08:
For performance testing I now added a `printk("%c", data[i]);` directly into the system file `zephyr/subsys/logging/log_backend_uart.c` after https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/logging/log_backend_uart.c#L23

The performance is still ok, so the issue must somewhere with the instantiation of the custom log backend.

caco3, 2021-02-09:
I seem to have found the culprit:
If I do **not** call `log_msg_put();` , it works as expected.

I have that line from the mentioned blog. And looking on the [functions description](https://docs.zephyrproject.org/latest/reference/logging/index.html?highlight=log_msg_put#c.log_msg_put), I would expect that such message actually is needed.

How ever the blog is from 2019, maybe that got now internally solved and is no longer needed?

carlescufi, 2021-02-09:
> I seem to have found the culprit:
> If I do **not** call `log_msg_put();` , it works as expected.
>
> I have that line from the mentioned blog. And looking on the [functions [...]

@nordic-krch can you confirm whether a call to `log_msg_put()` is required?

nordic-krch, 2021-04-12:
Function requires getting and putting the message as described here: https://docs.zephyrproject.org/2.5.0/reference/logging/index.html#logger-backends

closed 2021-04-12 by nordic-krch