zephyrproject-rtos/zephyr #32098: Implement a driver for the   Microphone / audio sensor (MP23ABS1) in Sensotile.Box

jimakos96, 2021-02-08:
**Is your feature request related to a problem? Please describe.**
I would like to develop an application which can  do audio sampling from the audio sensor in the Sensotile Box 

**Describe the solution you'd like**
I simply like to use the Zephyr ADC API for audio receiving .

**Describe alternatives you've considered**
 I tried to use the STM32HAL drivers functions for the ADC but i  wasn't succesful

erwango, 2021-02-09:
@jimakos96 I think this is already available. @avisconti do you confirm ?

avisconti, 2021-02-09:
> @jimakos96 I think this is already available using I2S. @avisconti do you confirm ?

Unfortnately not. Th Sensortile.box board has a MP23ABS1 microphone with an analog output. All the work I have done was on STM digital microphone with pdm output (like MP34DT05). In the latter case the pdm output can be interfaced to I2S and there is a sample here: zephyr/samples/shields/x_nucleo_iks02a1/microphone

In case of analog microphone I think that the only way is infact to use the ADC to convert from analog to digital, but I do not have much experience on it. So, it would be an interesting exercise for @jimakos96

erwango, 2021-02-09:
@jimakos96 instead of the HAL driver, I'd suggest using the ADC zephyr driver

jimakos96, 2021-02-09:
ok I will try to implement it using the adc driver for getting audio out of the internal microphone .

jimakos96, 2021-03-31:
Hello 
I am trying to get some samples from the microphone using the zephyr adc driver using the following code 
`#include <zephyr.h>
#include <sys/printk.h>

#include <drivers/gpio.h>
#include <usb/usb_device.h>
#include <drivers/uart.h>
#include <stdio.h>
#include <stdint.h>
#include <stdbool.h>

#include <drivers/adc.h>
#include <string.h>
uint16_t temp_data[8];

static const struct adc_channel_cfg channel_cfg_0 = {
        .channel_id       = 8,
};

const struct adc_sequence sequence = {
        .channels    = BIT(0),
        .buffer      = temp_data,
        .buffer_size = sizeof(temp_data),
        .resolution  = 12,
};

void main(void)
{

	/* Application must enable USB by itself */
	if (usb_enable(NULL)) {
		return;
	}
        int ret;
        const struct device *adc_dev = device_get_binding(DT_LABEL(DT_ALIAS(mic)));
        adc_channel_setup(adc_dev, &channel_cfg_0);

        while (1) {
                ret = adc_read(adc_dev, &sequence);
                if (ret != 0) {
                        printk("adc_read() failed with code %d", ret);
                } else {
                        printk("adc_read() value %d\n", temp_data[0]);
                }
        }
}
`
But i am not getting anything .In the board folder i have added these :
`
aliases {
		mic= &adc1;
	};
&adc1 {
	 pinctrl-0 = <&adc1_in8_pa3>;
	 status = "okay";
};`

closed 2021-04-14 by jimakos96