zephyrproject-rtos/zephyr #32100: spi_nor flash multi-instance support

DarrylGamroth, 2021-02-08:
**Is your enhancement proposal related to a problem? Please describe.**
I would like to use multiple SPI connected serial NOR flash ICs but the existing spi_nor driver only supports a single device.

**Describe the solution you'd like**
Multi-instance support similar to spi_flash_at45.

**Describe alternatives you've considered**
Using AT45 DataFlash except its capacity is too small.

madscientist159, 2021-04-16:
Just ran into this rather odd limitation.  Looking into fixing it.

madscientist159, 2021-04-17:
Initial patch here:
https://gitlab.raptorengineering.com/kestrel-collaboration/kestrel-firmware/zephyr-rtos/-/commit/e93d84d2621b569bc5700e7a8e57fb6fcc6661c5

We're blocked on submitting a PR for this until https://github.com/zephyrproject-rtos/zephyr/pull/32844 has been merged.

dakejahl, 2021-06-23:
What's the status here? Looks like #32844 is merged. I am also interested in using multiple external flash devices.

Man0D, 2022-06-28:
Hi, is there any news on this issue ?

madscientist159, 2022-12-24:
Unfortunately commit fb76f47f864 (merged shortly after #32844) broke this support, and with Zephyr 3.2 dropping support for key features required by Kestrel (#46758) we probably won't be looking at this in the near term.  If someone else would like to pick it up we can test on existing hardware with multiple devices.

GeorgePriestner-Eaton, 2023-05-04:
Is there any update on this please?

I am attempting to use external NOR flash with STM32H743 and getting 'undefined reference to `__device_dts_ord_132' error.

I disabled all SPI controllers other than spi3 (the one I'm, using with the jedec,spi-nor) but still get the issue.

de-nordic, 2024-02-01:
> 
> I am attempting to use external NOR flash with STM32H743 and getting 'undefined reference to `__device_dts_ord_132' error.

That describes nothing.

de-nordic, 2024-10-11:
The request is quite vague but I guess that https://github.com/zephyrproject-rtos/zephyr/pull/70415 has already fixed it, closing. If not fixed, reopen and fill in with more details what is required.

closed 2024-10-11 by de-nordic