zephyrproject-rtos/zephyr #32112: intel_adsp_cavs15: a part of testcases run failed with same error

KangJianX, 2021-02-09:
**Describe the bug**
running kernel testcases failed of a part of testing

**To Reproduce**
Steps to reproduce the behavior:
1. twister -p intel_adsp_cavs15 --device-testing -T **tests/kernel/** --west-flash="/home/ztest/work/zephyrproject/zephyr/boards/xtensa/intel_adsp_cavs15/tools/flash.sh,/home/ztest/work/zephyrproject/modules/audio/sof/keys/otc_private_key.pem,/home/ztest/work/zephyrproject/modules/audio/sof/zephyr/ext/rimage/config,/home/ztest/work/zephyrproject/modules/audio/sof/zephyr/ext/rimage/build/rimage" --device-serial-pty="/home/ztest/work/zephyrproject/zephyr/boards/xtensa/intel_adsp_cavs15/tools/adsplog.py,--no-history"

**Logs and console output**
INFO    - Adding tasks to the queue...
INFO    - Added initial list of jobs to queue
INFO    - Total complete:    4/  77   5%  skipped:   35, failed:    0
ERROR   - intel_adsp_cavs15         tests/kernel/mem_slab/mslab_concept/kernel.memory_slabs.concept FAILED: Timeout
ERROR   - see: /home/ztest/work/zephyrproject/zephyr/twister-out/intel_adsp_cavs15/tests/kernel/mem_slab/mslab_concept/kernel.memory_slabs.concept/handler.log
INFO    - Total complete:    7/  77   9%  skipped:   36, failed:    1
ERROR   - intel_adsp_cavs15         tests/kernel/mbox/mbox_usage/kernel.mailbox.usage  FAILED: Timeout
ERROR   - see: /home/ztest/work/zephyrproject/zephyr/twister-out/intel_adsp_cavs15/tests/kernel/mbox/mbox_usage/kernel.mailbox.usage/handler.log
INFO    - Total complete:   14/  77  18%  skipped:   39, failed:    2
ERROR   - intel_adsp_cavs15         tests/kernel/profiling/profiling_api/kernel.common.profiling FAILED: Timeout
ERROR   - see: /home/ztest/work/zephyrproject/zephyr/twister-out/intel_adsp_cavs15/tests/kernel/profiling/profiling_api/kernel.common.profiling/handler.log
INFO    - Total complete:   15/  77  19%  skipped:   39, failed:    3
ERROR   - intel_adsp_cavs15         tests/kernel/common/kernel.common                  FAILED: Timeout
ERROR   - see: /home/ztest/work/zephyrproject/zephyr/twister-out/intel_adsp_cavs15/tests/kernel/common/kernel.common/handler.log
INFO    - Total complete:   17/  77  22%  skipped:   40, failed:    4
ERROR   - intel_adsp_cavs15         tests/kernel/common/kernel.common.misra            FAILED: Timeout
ERROR   - see: /home/ztest/work/zephyrproject/zephyr/twister-out/intel_adsp_cavs15/tests/kernel/common/kernel.common.misra/handler.log
INFO    - Total complete:   23/  77  29%  skipped:   44, failed:    5
ERROR   - intel_adsp_cavs15         tests/kernel/sleep/kernel.common.timing            FAILED: Failed
ERROR   - see: /home/ztest/work/zephyrproject/zephyr/twister-out/intel_adsp_cavs15/tests/kernel/sleep/kernel.common.timing/handler.log
INFO    - Total complete:   33/  77  42%  skipped:   51, failed:    6
ERROR   - intel_adsp_cavs15         tests/kernel/threads/thread_stack/kernel.threads.thread_stack FAILED: Timeout
ERROR   - see: /home/ztest/work/zephyrproject/zephyr/twister-out/intel_adsp_cavs15/tests/kernel/threads/thread_stack/kernel.threads.thread_stack/handler.log
INFO    - Total complete:   38/  77  49%  skipped:   55, failed:    7
ERROR   - intel_adsp_cavs15         tests/kernel/mem_protect/sys_sem/kernel.memory_protection.sys_sem.nouser FAILED: Timeout
ERROR   - see: /home/ztest/work/zephyrproject/zephyr/twister-out/intel_adsp_cavs15/tests/kernel/mem_protect/sys_sem/kernel.memory_protection.sys_sem.nouser/handler.log
INFO    - Total complete:   39/  77  50%  skipped:   55, failed:    8
ERROR   - intel_adsp_cavs15         tests/kernel/msgq/msgq_api/kernel.message_queue    FAILED: Timeout
ERROR   - see: /home/ztest/work/zephyrproject/zephyr/twister-out/intel_adsp_cavs15/tests/kernel/msgq/msgq_api/kernel.message_queue/handler.log
INFO    - Total complete:   42/  77  54%  skipped:   55, failed:    9
ERROR   - intel_adsp_cavs15         tests/kernel/mem_protect/sys_sem/kernel.memory_protection.sys_sem FAILED: Timeout
ERROR   - see: /home/ztest/work/zephyrproject/zephyr/twister-out/intel_adsp_cavs15/tests/kernel/mem_protect/sys_sem/kernel.memory_protection.sys_sem/handler.log
INFO    - Total complete:   49/  77  63%  skipped:   57, failed:   10
ERROR   - intel_adsp_cavs15         tests/kernel/mem_heap/mheap_api_concept/kernel.memory_heap FAILED: Timeout
ERROR   - see: /home/ztest/work/zephyrproject/zephyr/twister-out/intel_adsp_cavs15/tests/kernel/mem_heap/mheap_api_concept/kernel.memory_heap/handler.log
INFO    - Total complete:   55/  77  71%  skipped:   60, failed:   11
ERROR   - intel_adsp_cavs15         tests/kernel/mem_heap/k_heap_api/kernel.k_heap_api FAILED: Timeout
ERROR   - see: /home/ztest/work/zephyrproject/zephyr/twister-out/intel_adsp_cavs15/tests/kernel/mem_heap/k_heap_api/kernel.k_heap_api/handler.log
INFO    - Total complete:   70/  77  90%  skipped:   62, failed:   12
ERROR   - intel_adsp_cavs15         tests/kernel/queue/kernel.queue                    FAILED: Timeout
ERROR   - see: /home/ztest/work/zephyrproject/zephyr/twister-out/intel_adsp_cavs15/tests/kernel/queue/kernel.queue/handler.log
INFO    - Total complete:   71/  77  92%  skipped:   62, failed:   13
ERROR   - intel_adsp_cavs15         tests/kernel/fifo/fifo_api/kernel.fifo             FAILED: Timeout
ERROR   - see: /home/ztest/work/zephyrproject/zephyr/twister-out/intel_adsp_cavs15/tests/kernel/fifo/fifo_api/kernel.fifo/handler.log
INFO    - Total complete:   73/  77  94%  skipped:   62, failed:   14
ERROR   - intel_adsp_cavs15         tests/kernel/lifo/lifo_usage/kernel.lifo.usage     FAILED: Timeout
ERROR   - see: /home/ztest/work/zephyrproject/zephyr/twister-out/intel_adsp_cavs15/tests/kernel/lifo/lifo_usage/kernel.lifo.usage/handler.log
INFO    - Total complete:   75/  77  97%  skipped:   62, failed:   15
ERROR   - intel_adsp_cavs15         tests/kernel/lifo/lifo_api/kernel.lifo             FAILED: Timeout
ERROR   - see: /home/ztest/work/zephyrproject/zephyr/twister-out/intel_adsp_cavs15/tests/kernel/lifo/lifo_api/kernel.lifo/handler.log
INFO    - Total complete:   77/  77  100%  skipped:   62, failed:   16
INFO    - 34 of 50 test configurations passed (68.00%), 16 failed, 62 skipped with 0 warnings in 1621.99 seconds
INFO    - In total 513 test cases were executed, 555 skipped on 1 out of total 329 platforms (0.30%)
INFO    - 50 test configurations executed on platforms, 0 test configurations were only built.

**error log**:
```
ASSERTION FAIL [!arch_mem_coherent(stack)] @ WEST_TOPDIR/zephyr/kernel/thread.c:575
@ WEST_TOPDIR/zephyr/lib/os/assert.c:45
```

**Environment (please complete the following information):**
OS: Fedora28
Toolchain: Zephyr-sdk-0.12.2
Commit ID: e61dbe59b0c26432687d83

KangJianX, 2021-02-09:
**A important things**, this bug reproduce base on my pull request [#31902](https://github.com/zephyrproject-rtos/zephyr/pull/31902)

nashif, 2021-02-09:
this is related to cache coherence asserts.

PR #32144 (2021-02-11): intel_adsp coherence fixups
There was a recent fix to a typo that unmasked a bunch of assertions in the coherence framework designed to catch situations where (incoherent) stack memory is being shared between CPUs.  Address the inevitable bitrot that crept in while these errors were hidden, and a handful of other related issues.

Fixes #32112
commit 38195d5d (2021-02-11): kernel/include: Put kernel stack memory in the right memory

The CONFIG_KERNEL_COHERENCE framework merged with a typo that left its
validation asserts disabled.  But it was written before the "kernel
stacks" feature merged, and so missed the K_KERNEL_STACK_* macros,
which need to put their stacks into __stackmem and not merely
__noinit.

Turning the asserts on exposed the bug.

Fixes #32112
closed 2021-02-11 by nashif