zephyrproject-rtos/zephyr #32133: Current atomics are subtly broken on AArch64 due to memory ordering

jharris-intel, 2021-02-09:
**Describe the bug**

On AArch64 atomic_inc gets turned into a LDAXR and a STLXR. This results in a subtly broken spinlock, as a load half-barrier followed by a store half-barrier is not equivalent to a full barrier.

**To Reproduce**

Look at compiler output for atomic_inc on AArch64.

**Expected behavior**

Spinlocks have the documented semantics on AArch64.

**Impact**

Long-tail corruption and hangs on local WIP AArch64 SMP version. No impact that I know of on non-SMP. No impact so far as I can tell on x86. Likely only impacts SMP on architectures with weak memory models.

**Logs and console output**

n/a

**Environment (please complete the following information):**
 - OS: Ubuntu 18.04.5 LTS
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used: based off of /u/MrVan's AArch64 SMP branch (https://github.com/zephyrproject-rtos/zephyr/pull/30676) with some additional fixes that I'm in the process of upstreaming.
 
 **Additional context**

While looking into this and _why_ the compiler was compiling `__atomic_compare_exchange_n(target, &old_value, new_value,
					   0, __ATOMIC_SEQ_CST,
					   __ATOMIC_SEQ_CST);` ([link](https://github.com/zephyrproject-rtos/zephyr/blob/master/include/sys/atomic.h#L49)) with two half-barriers, I found the following two threads, both of which are informative:

1. The linux kernel changed from `LDAXR` / `STLXR` to a different implementation in 2014, and this patch goes over the precise issue with the ordering much better than I could: http://lists.infradead.org/pipermail/linux-arm-kernel/2014-February/229588.html
2. There is a subtle difference between the `__sync` and `__atomic` builtins, in that the former act as full barriers whereas the latter (only) comply to the C11 memory model: https://gcc.gnu.org/bugzilla/show_bug.cgi?id=65697

In particular, the C11 atomics model only requires that `SEQ_CST` be `SEQ_CST` _in a program without data races_ - but many "standard" usages of atomics in kernel code are for code that C11 considers to have data races. (Or rather, in C11 a barrier for an operation on an object is only considered to apply to _that object_... with the obvious impedance mismatch between this and "using this for a spinlock". In general the C11 memory model has an interesting combination of very relaxed ordering by default and no way in general to restore sequential consistency once it has been lost.)

One example of this (assume all variables are 0 to start and all spinlocks unlocked):

```c
// thread A
*a = 1;
k_spinlock_key_t key = k_spin_lock(&lock);
*b = 1;

// thread B
l_b = atomic_get(&b);
l_a = atomic_get(&a);
assert(l_a || !l_b); // can fail
```

(Or on godbolt: https://godbolt.org/z/7Trr51 )

Essentially, thread A's operations, which are essentially

```asm
str     w4, [x3, #:lo12:.LANCHOR0] ; *a = 1
ldaxr   w3, [x0] ; load from lock->locked
stlxr   w4, w1, [x0] ; store to lock->locked
str     w1, [x2, 12] ; *b = 1
```
Can be reordered (in terms of the memory model, not the actual physical ldxr/stxr) as follows:

```
read lock->locked
write a
write b
write lock->locked
```

and from there to

```
read lock->locked
write b
write a
write lock->locked
```

herd7 agrees with me here:

```c
AArch64 Weird_Lock
{
0:X1=first; 0:X2=lock; 0:X3=second;
1:X1=first; 1:X2=lock; 1:X3=second;
}
 P0                 | P1;
 MOV X4, #1         | NOP;
 STR X4, [X1]       | NOP;
 NOP                | NOP;
 again:             | NOP;
 LDAXR X6, [X2]     | NOP;
 CBNZ X6, again     | NOP;
 STLXR W7, X4, [X2] | NOP;
 CBNZ W7, again     | NOP;
 NOP                | NOP;
 STR X4, [X3]       | NOP;
 NOP                | LDAR X5, [X3];
 NOP                | DSB SY;
 NOP                | LDAR X4, [X1];
exists
(1:X5=1 /\ 1:X4=0)
```
(Yes, the barriers are rather overkill on P1. Mainly just to show that it's a write-side problem rather than anything P1 is doing.)

You can run this online [here](http://diy.inria.fr/www/).

Resulting output:

```
Test Weird_Lock Allowed
States 4
1:X4=0; 1:X5=0;
1:X4=0; 1:X5=1;
1:X4=1; 1:X5=0;
1:X4=1; 1:X5=1;
Ok
Witnesses
Positive: 3 Negative: 9
Condition exists (1:X5=1 /\ 1:X4=0)
Observation Weird_Lock Sometimes 3 9
Time Weird_Lock 1.41
Hash=f46e985e74d452cc02786dee8b7c898f
```
If you add `dmb`s instead of the two NOP/NOP lines, this result is forbidden instead.

All told, with the existing C11-semantics spinlock, any code that relies on the access to `a` happening before the access to `c` above is subtly broken. I am unsure what the desired resolution here is. Some possibilities:

1. Use `__sync` primitives instead, which are identical on most architectures but on architectures where it makes a difference (e.g. AArch64) have the desired barrier semantics.
2. Add explicit barriers into the atomics implementation instead of relying on barriers in the operations themselves.
3. Scrub through and change, if necessary, any code that relies on shared communication of any sort from a section in a spinlock to a section outside a spinlock, or vice versa, as well as any other code that relies on atomics being considered full barriers.

carlescufi, 2021-02-09:
> Long-tail corruption and hangs on local WIP AArch64 SMP version. No impact that I know of on non-SMP. No impact so far as I can tell on x86. Likely only impacts SMP on architectures with weak memory models.

@jharris-intel 
FYI: https://github.com/zephyrproject-rtos/zephyr/pull/30676

carlescufi, 2021-02-09:
CC @carlocaione @MrVan

jharris-intel, 2021-02-09:
@carlescufi

> Commit SHA or Version used: based off of /u/MrVan's AArch64 SMP branch (#30676) with some additional fixes that I'm in the process of upstreaming.

:-)

I'm about to make a few comments about other issues I found in that PR, actually. I think that's the best way to approach PR-specific issues. (Although I don't know that much about processes here. So if there's a better method I'm all ears.)

As for this issue: it's more general than just AArch64 SMP - it's just that AArch64 is the first architecture in practice where it matters.

nashif, 2021-02-09:
reassigning, after reading the bug more carefully.

carlocaione, 2021-02-09:
Uhm I'll take a look but to me it looks like a general problem, not really specific to AArch64.

andyross, 2021-02-09:
I see the references, and buy the theoretical problem.  I guess I'm failing to see how this actually impacts the use of the GCC atomics layer to implement a spinlock in Zephyr.   As long as

1. The instructions between the LDAXR/STLXR pair don't touch memory (and they don't, AFAICS)
2. Nothing but LDAXR/STLXR touches the value at the relevant address

Then no race exists and all other accesses in the system are guaranteed to see a strong ordering across the sequence.  Right?  Is the worry that something might have come along and memcmp()'d the range containing the spinlock or something?

jharris-intel, 2021-02-09:
>Then no race exists and all other accesses in the system are guaranteed to see a strong ordering across the sequence. Right?

Nope :-)

That's precisely why this bug is so subtle by the way - acquire/release semantics are close to, but subtly different from, the intuitive semantics.

In particular on ARM the previous store can be ordered past the LDAXR, and the next store can be ordered before the STLXR, and indeed before the previous store.

I suggest you look at the herd7 ordering I posted. It provides specific examples allowed within the ARM memory model that violate "all other accesses in the system are guaranteed to see a strong ordering across the sequence".

```asm
AArch64 Weird_Lock
{
0:X1=first; 0:X2=lock; 0:X3=second;
1:X1=first; 1:X2=lock; 1:X3=second;
}
 P0                 | P1;
 MOV X4, #1         | NOP;
 STR X4, [X1]       | NOP;
 NOP                | NOP;
 again:             | NOP;
 LDAXR X6, [X2]     | NOP;
 CBNZ X6, again     | NOP;
 STLXR W7, X4, [X2] | NOP;
 CBNZ W7, again     | NOP;
 NOP                | NOP;
 STR X4, [X3]       | NOP;
 NOP                | LDAR X5, [X3];
 NOP                | DSB SY;
 NOP                | LDAR X4, [X1];
exists
(1:X5=1 /\ 1:X4=0)
```

The hurd7 run shows that you can get x5=1, but x4=0.

Or, breaking this down a little. Assume that all memory is zero to start, and that X1-X3 contain unique well-aligned addresses.

Core 0 does the following:

```asm
 MOV X4, #1
 STR X4, [X1] 
```
a.k.a. "store 1 to *x1".

Then

```asm
again:
 LDAXR X6, [X2]
 CBNZ X6, again
 STLXR W7, X4, [X2]
 CBNZ W7, again
```
a.k.a. a LDAXR/STLXR spinlock, or effectively `while (!atomic_cas(&x2, 0, 1)) {}`

Then

```asm
 STR X4, [X3] 
```
a.k.a. "store 1 to x3"

Note that your constraints are adhered to. The instructions between the LDAXR/STLXR pair don't touch memory, and nothing but LDAXR/STLXR touches the value at the relevant address (x2).

Meanwhile, Core 1 does the following:

```asm
LDAR X5, [X3];
DSB SY;
LDAR X4, [X1];
```
...which is overkill for "read from x3, then a full barrier, then read from x1".

All told in C-ish:

```c
static uint64_t a = 0;
static uint64_t b = 0;
static uint64_t c = 0;

void on_core_0(void) {
    *x1 = 1;
    while (!atomic_cas(&x2, 0, 1)) {}
    *x3 = 1;
}
bool on_core_1(void) {
    uint64_t a = *x3;
    __sync_synchronize();
    uint64_t b = *x1;
    return b && !a;
}
```
...and on_core_1 can return `true`. Which is not a "strong ordering across the sequence".

> I see the references, and buy the theoretical problem

Unfortunately, this is a very real problem on AArch64 SMP. Fortunately, so far as I can tell this is only really a real problem on AArch64 SMP.

MrVan, 2021-02-10:
@jharris-intel Do you have any plan to fix the issues that you mentioned, or any TODO list?

carlocaione, 2021-02-10:
I'm still a bit lost on the details but for the sake of discussion I pushed #32170 where all the atomics are now implemented using the `__sync` builtins.

jharris-intel, 2021-02-10:
> @jharris-intel Do you have any plan to fix the issues that you mentioned, or any TODO list?

I'm working through the issues I found and making bug reports here or comments directly on your PR, as appropriate. For the most part your PR is solid however - I had to make surprisingly few changes to get it to work on our system. (You fixed one of the issues before I had a chance to report it, even!)

In general I'm impressed with the Zephyr project's responsiveness here. Kudos!

pabigot, 2021-02-16:
If the original example in the problem description is changed to:

```
// thread A
atomic_set(&a, 1);  // NOT *a = 1;
k_spinlock_key_t key = k_spin_lock(&lock);
atomic_set(&b, 1);  // NOT *b = 1;

// thread B
l_b = atomic_get(&b);
l_a = atomic_get(&a);
assert(l_a || !l_b); // can fail
```

does the particular reordering problem described there disappear?

jharris-intel, 2021-02-16:
@pabigot - that particular example no longer has the reordering problem, because `atomic_set` ends up as another LDAXR / STLXR loop. (I'm more familiar with that particular primitive being called `atomic_exchange`.)
 
```asm
AArch64 Weird_Lock
{
0:X1=first; 0:X2=lock; 0:X3=second;
1:X1=first; 1:X2=lock; 1:X3=second;
}
 P0                 | P1;
 MOV X4, #1         | ;
 againA:            | ;
 LDAXR X6, [X1]     | ;
 STLXR W7, X4, [X1] | ;
 CBNZ W7, againA    | ;
                    | ;
 againB:            | ;
 LDAXR X6, [X2]     | ;
 CBNZ X6, againB    | ;
 STLXR W7, X4, [X2] | ;
 CBNZ W7, againB    | ;
                    | ;
 againC:            | ;
 LDAXR X6, [X3]     | ;
 STLXR W7, X4, [X3] | ;
 CBNZ W7, againC    | ;
                    | LDAR X5, [X3];
                    | DSB SY;
                    | LDAR X4, [X1];
exists
(1:X5=1 /\ 1:X4=0)
```

which results in

```
States 3
1:X4=0; 1:X5=0;
1:X4=1; 1:X5=0;
1:X4=1; 1:X5=1;
Loop No
Witnesses
Positive: 0 Negative: 81
Observation Weird_Lock Never 0 81
```

The reordering I mentioned earlier only happens with "plain" loads/stores. Unfortunately, the current usage model for spinlocks is typically:

```c
k_spinlock_key_t key = k_spin_lock(&some_lock);
do_some_unsynchronized_operation();
k_spin_unlock(&some_lock, key);
do_some_other_operation();
```
which (almost?) never uses atomics for `do_some_unsynchronized_operation`. By itself this isn't a problem - except if something else in the system expects to see the side effects of `do_some_unsynchronized_operation` before any of the side effects of `do_some_other_operation`.

*****

This comes back to my previous statement:

> [Option 3:] Scrub through and change, if necessary, any code that relies on shared communication of any sort from a section in a spinlock to a section outside a spinlock, or vice versa, as well as any other code that relies on atomics being considered full barriers.

We could, conceivably, scrub through and replace any case that a variable might be impacted by this issue with `atomic_*` API calls instead of direct accesses. It would be a hefty effort. Some fallout of this would be:

1. A performance hit, because we now have many extraneous barriers. (If you do `lock(); *a = 1; *b = 2; unlock();`, you don't need a barrier between the write to `a` and the write to `b` in many cases, but if you're replacing things with `atomic_` calls you will end up with that.
2. Subtly-incorrect behavior if anyone introduces any additional behavior in future (_I'm_ certainly not confident that I can catch every case that would require additional atomics in this model.)
3. Wrinkles with user callbacks. It is within Zephyr's power to change the internals of Zephyr. Not so much user code - which in some cases includes callbacks from Zephyr code.

In general, the C11 memory model works reasonably well for application-level development. Kernel development, on the other hand... it fails to expose much of the functionality of the underlying hardware (e.g. `there is no equivalent to `dmb` in the C11 memory model - `atomic_thread_fence` comes close but fails to synchronize with non-atomics in quite the same manner), and assumes that all code is trusted (or rather, all code follows its own assumptions about the world).

Personally, I don't think that this approach is particularly feasible. There's a lot of very subtle behavior that has to be grok'd globally in order to use this approach.

jharris-intel, 2021-02-25:
By the way, I have been unable to actually hit this issue in practice on a Cortex-A53. Which means one of a couple of things:

1. The memory model allows this behavior, but the Cortex-A53 is stronger than the ARMv8 memory model. Entirely plausible - the A53 is a fairly little core all things considered (in-order dispatch, for one).
2. The memory model allows this behavior, as does the Cortex-A53, but I did not hit whatever edge case would be necessary to actually cause this behavior on the A53. Again, entirely plausible, although I made sure to test it with a fair bit of cache churn
3. The memory model does not allow this behavior, and both I, the kernel devs, and the gcc devs were mistaken. Possible, although I think this one is somewhat unlikely. (Hopefully this isn't too hubristic...)

I'm going to see if I can get an answer from ARM on this one I think. It affects atomics performance in a nontrivial manner.

closed 2021-08-19 by github-actions[bot]