zephyrproject-rtos/zephyr #32137: Provide test execution times per ztest testcase

nashif, 2021-02-09:
**Is your enhancement proposal related to a problem? Please describe.**
We do not have any information about how long each test runs when running tests using ztest.

**Describe the solution you'd like**
For each test, print our the execution time as part of the results

**Describe alternatives you've considered**
Provide test execution time per testcase.

For example:

now:
```
START - test_sem_multi_take_timeout_diff_sem
 PASS - test_sem_multi_take_timeout_diff_sem
```
with per test execution time:
```
START - test_sem_multi_take_timeout_diff_sem
 PASS - test_sem_multi_take_timeout_diff_sem in 2.54 seconds
```

nashif, 2021-02-09:
@jocelyn-li FYI

enjiamai, 2021-03-06:
Ok, got it, will add this.

PR #33014 (2021-03-11): tests: ztest: provide test execution times per ztest testcase
Add execution time for testing result of each ztest testcase as:

START - test_sem_multi_take_timeout_diff_sem
 PASS - test_sem_multi_take_timeout_diff_sem in 2.54 seconds

Fix #32137.
commit 2c5e2c23 (2021-03-11): tests: ztest: provide test execution times per ztest testcase

Add execution time for testing result of each ztest testcase as:

START - test_sem_multi_take_timeout_diff_sem
PASS - test_sem_multi_take_timeout_diff_sem in 2.54 seconds

Fix #32137.
closed 2021-03-11 by nashif